letscontrolit/ESPEasy #2854: RSSI unit is dB, not dBm

ohminy, 2020-01-15:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [o] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [O] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

As I know... I think... normaly WiFi RSSI's unit is "dBm",
But, in the main page of the web interface, there is "dB" ^^ 
that's all ^^;;

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: megal-20191208-normal

ESP Easy settings/screenshots: 

### Rules or log data

```

```

Grovkillen, 2020-01-15:
Yep, and it's fixed in the upcoming new GUI. Thanks for noticing.

TD-er, 2020-01-15:
That's a simple change for once :)
Maybe we also have it in other places too.

tonhuisman, 2022-04-18:
This issue has been fixed, so probably can be closed.

closed 2022-04-18 by TD-er