letscontrolit/ESPEasy #2857: SendToHTTP problem

piotrrekkk, 2020-01-17:
I noticed, that we I pointing to IP(or URL address) outside of my local network using SendToHTTP it doesn't work. Tested on couple networks using DHCP or static IP address. It's working well when IP is in the same network.
Tested on latest version _mega-20191208_ and _mega-20191028_
***Sample of my code below:***

```
on WiFi#Connected do
   timerSet,1,30
endon

On Rules#Timer=1 do
   logentry,pm25=[SDS011#PM10]&pm25=[SDS011#PM25]
   SendToHTTP 38.240.11.12,81,/kobylany?type=command&pm10=[SDS011#PM10]&pm25= [SDS011#PM25]
    timerSet,1,60
endon
```

Part of my log below:
```
422293: WD : Uptime 7 ConnectFailures 0 FreeMem 20256 WiFiStatus 3
427737: EVENT: Clock#Time=Fri,10:40
439737: EVENT: Rules#Timer=1
439782: ACT : logentry,pm25=22.60&pm25=16.90
439792: Command: logentry
439793: pm25=22.60&pm25=16.90
439803: ACT : SendToHTTP 38.240.11.12,81,/kobylany?type=command&pm10=22.60&pm25=16.90
439813: Command: sendtohttp
439923: ACT : timerSet,1,60
439933: Command: timerset
452293: WD : Uptime 8 ConnectFailures 0 FreeMem 20224 WiFiStatus 3
482293: WD : Uptime 8 ConnectFailures 0 FreeMem 20224 WiFiStatus 3
487737: EVENT: Clock#Time=Fri,10:41
500737: EVENT: Rules#Timer=1
500782: ACT : logentry,pm25=22.60&pm25=16.90
500792: Command: logentry
500794: pm25=22.60&pm25=16.90
500803: ACT : SendToHTTP 38.240.11.12,81,/kobylany?type=command&pm10=22.60&pm25=16.90
500814: Command: sendtohttp
500923: ACT : timerSet,1,60
500934: Command: timerset
512293: WD : Uptime 9 ConnectFailures 0 FreeMem 20160 WiFiStatus 3
```

If can be useful I can paste all my config data.

piotrrekkk, 2020-01-17:
Config here:
[config.txt](https://github.com/letscontrolit/ESPEasy/files/4076479/config.txt)

TD-er, 2020-01-18:
You can show exactly the same as in the linked config.txt by just copy/paste the web page content of that sysinfo page directly in the text editor here on Github.
That will create a nice looking table, which is better readable than a separate .txt file.

Could it be related to an external host demanding the client to wait for a confirmation?

piotrrekkk, 2020-01-19:
@TD-er , thanks, You're right, pasted:

Unit Number: | 0
-- | --
Uptime: | 0 days 0 hours 2 minutes
Load: | 16.50% (LC=3089)
CPU Eco Mode: | false
Free Mem: | 19312 (17896 - sendContentBlocking)
Free Stack: | 3568 (2052 - sendContentBlocking)
Heap Max Free Block: | 16360
Heap Fragmentation: | 16%
Boot: | Cold boot (0)
Reset Reason: | External System
Last Task: | Background Task
SW WD count: | 0
Network ❔
Wifi: | 802.11N (RSSI -54 dB)
IP Config: | DHCP
IP / Subnet: | 192.168.31.22 / 255.255.255.0
Gateway: | 192.168.31.1
Client IP: | 192.168.31.25
DNS: | 192.168.31.1 / (IP unset)
Allowed IP Range: | 192.168.31.0 - 192.168.31.255
STA MAC: | B4:E6:2D:53:D9:4F
AP MAC: | B6:E6:2D:53:D9:4F
SSID: | domek_2 (28:6C:07:CF:C0:33)
Channel: | 11
Connected: | 1 m 29 s
Last Disconnect Reason: | (1) Unspecified
Number Reconnects: | 0
WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Firmware
Build:⋄ | 20104 - Mega
System Libraries:⋄ | ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191028
Plugins:⋄ | 46 [Normal]
Build Md5: | 72e99c9ec2124433ea347675c5b76d3c
Md5 check: | passed.
Build Time:⋄ | Oct 28 2019 11:28:15
Binary Filename:⋄ | ESP_Easy_mega-20191028_normal_ESP8266_4M1M.bin
System Status
Syslog Log Level: | None
Serial Log Level: | Info
Web Log Level: | None
ESP Board
ESP Chip ID: | 5495119 (0x53D94F)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 847 kB (2224 kB free)
SPIFFS Size: | 934 kB (844 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 116
Maximum open files: | 5
Maximum path length: | 32

piotrrekkk, 2020-01-19:
Same thing via generic http is working well.

TD-er, 2020-01-19:
If you are connecting to this node, then make sure the IP subnet filter is disabled when you try to fetch it from a different subnet.

tonhuisman, 2023-09-02:
This should be fixed, so can be closed.

closed 2023-09-02 by TD-er