letscontrolit/ESPEasy #2859: Dew point value

michass, 2020-01-22:
Hi,

can I use dew point on straight way from logs, or need I create extra rules and dummy devices? 

[image]

TD-er, 2020-01-22:
See [docs example dew point](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#dew-point-for-temp-humidity-sensors-bme280-for-example)
In a future update I want to add it as a standard conversion for rules as the formula in this example is not as accurate as it can be, but it is a relative good approximation for humidity >= 50%.

Tom-Bom-badil, 2020-01-22:
I thought the %c_dew_th% had been introduced for this and is already more accurate than the mentioned example? I am using this on all my devices for writing into a dummy device - any knwon disadvantage doing it this way?

`on Sensor#Temperature do`
`  TaskValueSet,3,1,%c_dew_th%([Sensor#Temperature],[Sensor#Humidity])`
`endon`

/tom

TD-er, 2020-01-22:
Ah I was not 100% sure I already implemented it, couldn't find it in the docs :)
So if we already have it, then that's the proper formula.

Tom-Bom-badil, 2020-01-22:
Yes - if I remember right, there was a thread in the forums about this a year or even two ago. Took it over from there, it worked - so kept going ...
/tom

Tom-Bom-badil, 2020-01-22:
... which reminds me that it could be useful to replace addressing of DummyDevice with it's name instead of ID. So instead of

`on Sensor#Temperature do`
`TaskValueSet,3,1,%c_dew_th%([Sensor#Temperature],[Sensor#Humidity])`
`endon`

it would be nice to read

`on Sensor#Temperature do`
`TaskValueSet,DummyDevice,1,%c_dew_th%([Sensor#Temperature],[Sensor#Humidity])`
`endon`

But thats a 'LuxuryFeature' for reading written code months later, not a must-have ...

/tom

michass, 2020-01-23:
Ok, thanks,

it works, but if this parameter is present in logs, then exist. No need to be calculated once again. Can be added straight to BMP plugin as 4th parameter, or maybe optional parameter if not needed.

TD-er, 2020-01-23:
> Ok, thanks,
> 
> it works, but if this parameter is present in logs, then exist. No need to be calculated once again. Can be added straight to BMP plugin as 4th parameter, or maybe optional parameter if not needed.

Well no it can't.
For example the Domoticz controller is very picky on how many parameters are present and in what order.
If you look at the code, I already do some 'odd' things for Domoticz to support both BME280 and BMP280 in the same plugin.
So no, I cannot add a new value to the plugin without breaking stuff.
Also because of the troubles caused by the switching option of the sysinfo and dummy plugin, I am for now not adding a switch to select outputs on plugins until that's fixed with a proper fix (which may need a lot of code changes to get it right)

tonhuisman, 2023-09-02:
This has been resolved, so can be closed.

closed 2023-09-02 by TD-er