letscontrolit/ESPEasy #2860: Jump into the Next Line with a Switch

Alexxx2005, 2020-01-23:
Hello,

it is a very Good Project for the Wemos ! Thank You !

I want to use the example to sign up the Song Title from
my Music Band. I usw the OLED with 128x64 and the 12Lines
are full with tracks.

My Question : It is possible to jump with a Switch Button
from Line 1 to Line 2, when i press the Button again i jump
from Line 2 to Line 3 usw.

someone gave me a solution best regards Alex

tonhuisman, 2020-03-08:
@Alexxx2005 How would you mark the selected line, as we don't have a cursor. A possible solution could be to write a `<` character at the rightmost  column. Selecting the desired line can be done using a Dummy Device (I named it Variables with Selected as the 1st value, in Device 12). Then adding this rule (shortened, but you can copy/adapt the rest)
```
on Variables#Selected do
  if [Variables#Selected] = 1
    oled,1,16,"<"
  else
    oled,1,16," "
  endif
  if [Variables#Selected] = 2
    oled,2,16,"<"
  else
    oled,2,16," "
  endif
...
endon
```
Then some code for the Up/Down buttons:
```
on ButtonUp#State do
  if [Variables#Selected] > 1
    TaskValueSet,12,1,[Variables#Selected] - 1
  endif
endon
on ButtonDown#State do
  if [Variables#Selected] < 12
    TaskValueSet,12,1,[Variables#Selected] + 1
  endif
endon
```

tonhuisman, 2022-05-06:
This feature has been added to the OledFramed (P036) plugin, but but not to the Oled (P023) plugin.

closed 2022-05-06 by TD-er