letscontrolit/ESPEasy #2863: Display - OLED SSD1306/SH1106 Framed Bug: Can't send strings with spaces using http request

tonhuisman, 2020-01-25:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [X] filled out all applicable fields below.

### Steps already tried...
- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)
- [X] Tested on other node to make sure hardware isn't defective.
- [X] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

Currently latest build: ESP_Easy_mega-20191208_test_beta_ESP8266_4M1M

### Summarize of the problem/feature request

On EspEasy ESP_Easy_mega-20191208_test_beta_ESP8266_4M1M installed on a Wemos D1 mini clone I have a 128x64px OLED display
Using a bash script I try to send text to a line of the display using curl with an url like this:
 url=http://${ESPADDR2}/control?cmd=OLEDFRAMEDCMD,1,update%3a%25SP%25${now}
But the only text displaying is the "update:" part, anything after the first space is ignored (I also tried to send %20 for a space, but to no avail, so I tried the systemvariable %SP%)

The bug appears to be in the P036 code, in PLUGIN_WRITE section:

```
            // content functions
            success = true;
            String NewContent = parseStringKeepCase(string, 3);

```
It only reads 1 word from the url, as there is a yet unsupported parameter <mode> that needs to be parsed sometime in the future.
Proposed solution:
- Read the rest of the parameters, parse for the last comma, if found remove the part after it from NewContent and put that into the <mode> argument (though unused), if not found, keep the found text as NewContent.
- Proceed the current processing

### Expected behavior

I'd expect, assuming the command format should not change, that anything up until the last comma(or end of string if no comma is found) should be read from the parameters as NewContent

### Actual behavior

Only the first word is read (and that word can't be 'display'...)

### Steps to reproduce
1. use a browser or curl to send a command to an espeasy OLed Framed display con taining a text of multiple words
2. Only the first word sent will show
3. 

It's a software bug, so restarting isn't going to help

### System configuration
Hardware:

Wemos D1 mini clone

ESP Easy version: ESP_Easy_mega-20191208_test_beta_ESP8266_4M1M

### Rules or log data

TD-er, 2020-01-25:
Have you tested with the hints given here: https://github.com/letscontrolit/ESPEasy/issues/2724  ?

For example:
`http://192.168.88.234/tools?cmd=oledframedcmd%2C1%2C%22text+example+with+spaces%22`

tonhuisman, 2020-01-25:
Ok, adding quotes around the string and a + instead of a space does work from a browser (some more investigation to have curl not massage out the +...). But this does sound quite counter-intuitive, as the P023 OLED plugin does accept standard spaces (either a space or %20 in the url).
And the documentation is somewhat out of sync, as the oledframedcmd command isn't found there (AFAICS).
Someone should write the wiki page for P036 with all this info, maybe I'll volunteer for that...

I still think that adding the proper content-handling to the plugin is the right way to go, though this is a somewhat acceptable workaround.

tonhuisman, 2020-01-25:
For those interested, the url to pass to curl (on a RPi) turned out to this:
`url=http://${ESPADDR2}/control?cmd=OLEDFRAMEDCMD,1,\'update%3a+${now}\'`

TD-er, 2020-02-04:
As this is not really a bug, but maybe lack of links to documentation, I will close this one.

closed 2020-02-04 by TD-er