letscontrolit/ESPEasy #2865: Central command and event handling

TD-er, 2020-01-27:
Currently we have the possibility to receive commands from a number of sources.
For example:
- serial port
- p2p controller
- MQTT (Domoticz/OpenHAB)
- HTTP GET
- Rules

The problem is that there is no generic handler for receiving a command and executing it.
For example two MQTT controllers (C002 - Domoticz and C005 OpenHAB) do accept commands or events.
But both do have their own unique special handling for some plugins.
For some specific scenario's this can be a valid use case, but for most it is just code duplication and adding to the confusion why functionality differs among controllers.

So in short:
- Add (or choose) a generic command handler.
- Find a way to handle specific commands or events inside the plugin and not in the controllers.
- Proper detect whether something is an event or a command, so we don't have to use the "spaghetti method" to see what call may be successful in handling the incoming command or event.
- Evaluate demands for permission for some events or commands.  (factory reset via MQTT doesn't sound like a good idea)

giig1967g, 2020-01-27:
Ah ah ah!!!! I can't stop laughing.

>     * Proper detect whether something is an event or a command, so we don't have to use the "spaghetti method" to see what call may be successful in handling the incoming command or event.

What is a "spaghetti method" ? 
(Warning: I am italian!) 
:) :) :) 💯

TD-er, 2020-01-27:
"Spaghetti programming" is trying to random things and see what sticks.
Analogue to testing whether Spaghetti is well cooked. If you throw it to a wall and it sticks, then it is OK.  (well, at least that's what I have been told, never tried it)

There is also "banana programming", which is releasing it and let it ripe at the customer via lots and lots of bug fixes :)

giig1967g, 2020-01-27:
> Analogue to testing whether Spaghetti is well cooked. If you throw it to a wall and it sticks, then it is OK. (well, at least that's what I have been told, never tried it)
 
That's too much fun!

tonhuisman, 2023-09-02:
This seems no longer relevant, except for the jokes, but still can be closed. 😄

closed 2023-09-02 by TD-er