letscontrolit/ESPEasy #2866: MQTT topics problems in latest develop

crnjan, 2020-01-29:
Hi!

I'm working on a new plugin and noticed strange behaviour in latest develop (while being fine in last pre-release).

Long story short - when using CPLUGIN_005 (Home Assistant (openHAB) MQTT) to publish messages to broker, in latest develop the topic is not correct.

I.e. in my setup the topic should be 

"ESP_Easy/xxx/MitsubishiHeatPump" 

but its send as 

"ESP_Easy/xxx/MitsubishiHeatPump " 

(notice the space at the end). I checked what could be the issue and my guess is that change [here](https://github.com/letscontrolit/ESPEasy/commit/c0eb4ec28f5c04b499fae1937339cd63c6c92e58) is the problem.

So `parseTemplate` pads spaces at the end to match length:

```
  // padding spaces
  while (newString.length() < lineSize) {
    newString += ' ';
  }
```

but when we call the `parseTemplate` from `parseControllerVariables`, it still contains the placeholder `%sysname%`:

"%sysname%/xxx/%valname%"

which is replaced with "ESP_Easy" within the `parseTemplate` function and since  "ESP_Easy" has 8 chars (compared to "%sysname%" with 9 chars), `parseTemplate` will append a space to the end:

"ESP_Easy/xxx/%valname% "

(notice the space at the end). Publisher will use this topic (with extra space) to publish the message ...

Hopefully my explanation makes sense :)

giig1967g, 2020-01-29:
I never noticed it, probably because my broker trims the spaces, but it's indeed happenning.
One of my topic is: `"ESPT12/d15/State<+ 3 spaces>"`

giig1967g, 2020-01-29:
@crnjan good catch by the way

giig1967g, 2020-01-29:
this will fix it:

```
  // padding spaces
  while (newString.length() < lineSize) {
    newString += ' ';
  }
  STOP_TIMER(PARSE_TEMPLATE);
  checkRAM(F("parseTemplate3"));

//ADDING TRIM BEFORE RETURN
  newString.trim();
  return newString;
}

```

giig1967g, 2020-01-29:
see #2867

crnjan, 2020-01-30:
Hey! Thx for the prompt response. My only comment about the fix is that topic can actually contain spaces ... not sure if someone uses that in practice, but standard allows them ...

> Topic Names and Topic Filters can include the space character

Taken from [here](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/errata01/os/mqtt-v3.1.1-errata01-os-complete.html#_Toc442180919).

TD-er, 2020-01-30:
> this will fix it:
> 
> ```
>   // padding spaces
>   while (newString.length() < lineSize) {
>     newString += ' ';
>   }
>   STOP_TIMER(PARSE_TEMPLATE);
>   checkRAM(F("parseTemplate3"));
> 
> //ADDING TRIM BEFORE RETURN
>   newString.trim();
>   return newString;
> }
> ```

It shouldn't be fixed there, but in the place where it is called.
As you can see in the code you quoted, it is done intentional. It was once meant to provide alignment for displaying on a display.
The call to this function also somewhere has a number provided to set the (minimal) string length.

I will make a patch for it, as topics obviously don't need to have trailing spaces, so it makes sense to sanitize the topics in the MQTT related code.

TD-er, 2020-02-04:
> Hey! Thx for the prompt response. My only comment about the fix is that topic can actually contain spaces ... not sure if someone uses that in practice, but standard allows them ...
>
>
> Taken from [...]

When trimming the topic, it removes only the leading and trailing spaces.
I think it is not  too limiting to have a requirement not to have a leading or trailing space in a topic or filter, right?

closed 2020-02-04 by TD-er

crnjan, 2020-02-07:
I too think that topics with trailing/leading spaces are not really used in practice. Thx for the fix, will give it a spin over the weekend!

PR #2864 (2020-02-04): [Controllers] Restructure protocolIndex / controllerIndex and cplugin_ID
This was rather messy code and no checks.
It is now somewhat similar to the code for plugins defining the relation between:
- Plugin ID
- Task Index
- Device Index.

Also added the option to call CPLUGIN_EXIT and a call for 10/sec loops which we apparently didn't have until now for controllers.

Fixes: #2821
Fixes: #2866
commit 3f627787 (2020-02-04): Fix unexpected string padding in parseTemplate (#2866)

The extra padding to a minimal line length does only make sense for use on displays and some special formatting.
So the majority of the uses of parseTemplate now uses 0 as minimum line length.

Fixes: #2866
Closes: #2867 (superseeds PR)