letscontrolit/ESPEasy #2868: mega-20191208 generic pulsecounter dashboard does not refresh

GravityRZ, 2020-01-31:
i am using the latest build with a generic pulsecounter and also dummy device

the dummy device screen values(devices screen) updates it values in realtime
The generic pulsecounter scren however does not update it's values
it works but the numbers on screen do not refresh. eg counter does not advance, time does not change

[image]

TD-er, 2020-01-31:
What interval time is set to the active devices?
The update interval is determined by the lowest "interval" time of all tasks.

Have you seen it update in previous builds?

GravityRZ, 2020-01-31:
the update interval of the device for count/total/Time is 1 second.
2 other devices have update intervals of 60 seconds.

this is the first build i tested so can not tell if others have the same problem.
i expect they do because they use the same pulse counter plugin.

GravityRZ, 2020-03-10:
is this bug fixed in newer releases? can not seem to find that it is.

TD-er, 2020-03-10:
I don't remember I have had a specific look at the code for this.

So I guess you still experience issues with this?

Is this only a problem when you have only this plugin set as an active task, or do you have other tasks active as well?

GravityRZ, 2020-03-10:
i have 3 devices total
1 is the counter on gpio-14
2 is a temp sensor on gpio-13 which updates in realtime
3 is a dummy device which counts perfectly and updates realtime
so nothing special

[image]

GravityRZ, 2020-03-12:
i tested it all the way back to build mega 20181218 but the problem stays the same
the count dashboard does not refresh for the generic counter
even tried internet explorer but still not working.

this is easy to reproduce
take a fresh module with espeasy on it
put a generic pulsecounter in either line 1,2 or 3
generate a pulse and watch the screen(nothing happens)
update the screen F5 and watch it change

PR #3673 (2021-06-05): [Pulse counter] Fix values not being updated on Devices tab
This `PLUGIN_WEBFORM_SHOW_VALUES` is no longer needed for P003 pulse counter.
Removing it, fixes the (more than once) reported issue of not updating pulse counter values on the Devices tab.

Fixes: #2868
closed 2021-06-05 by TD-er