letscontrolit/ESPEasy #2869: P_68_Environment possible wrong temperature

Egyras, 2020-01-31:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Seems sht35 reads temperature incorrectly 

### Expected behavior

Temperature should be lower approximately about 2C. 

### Actual behavior

Reads temperature incorrectly 

### Steps to reproduce

1. Use sht35 sensor with wemos d1 mini
2. 
3. 

 

### System configuration

Hardware:

ESP Easy version: 
Latest from nightly builds

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2020-01-31:
Are you sure the sensor doesn't get heated by the other components in the setup?
How is the sensor positioned? (pics would be helpful here)

Egyras, 2020-01-31:
@TD-er its isolated from wemos d1 mini in small enclosure. Sensor placed according recommended desing documentation.

Egyras, 2020-01-31:
Seems small enclosure impact, use sleep to cooldown.

closed 2020-01-31 by Egyras