letscontrolit/ESPEasy #2871: ESPEasy 20191208 fails to compile on arduino-ide-1.8.11 with esp arduino core 2.3.0 on mint 19.1

m-anish, 2020-02-02:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I am trying to get a working setup of ESPEasy with arduino IDE using [this](https://www.letscontrolit.com/wiki/index.php?title=Tutorial_Arduino_Firmware_Upload) guide. 

Version details:
arduino-ide: 1.8.11
board version on ide: 2.3.0
esp-easy version: 20191208 release

Upon compilation, I get this error:
```
Multiple libraries were found for "Servo.h"
Networking:775:23: error: AddrList.h: No such file or directory
 Used: /home/anish/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/Servo
 # include <AddrList.h>
 Not used: /home/anish/Documents/arduino-ide/arduino-1.8.11/libraries/Servo
                       ^
compilation terminated.
exit status 1
AddrList.h: No such file or directory
```

I suspect it has something to do with version incompatibilities.

What i'd like as a response:
* What versions of espeasy, arduino-ide and arduino esp8266 core interoperate?
* Can the very helpful guide in the wiki be updated accordingly, if indeed there is a problem?

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

m-anish, 2020-02-02:
FWIW, espeasy builds with the esp8266 board core version 2.6.3, so probably a doc fix is needed :)

TD-er, 2020-02-02:
Do you have the "servo" library installed in the Arduino IDE?

m-anish, 2020-02-03:
yes.

```
anish@elrond:~/Documents/arduino-ide/arduino-1.8.11/libraries$ ls -1
Adafruit_Circuit_Playground
Adafruit_Motor_Shield_V2
Adafruit_MPR121
Adafruit_NeoPixel
Adafruit_Sensor
Adafruit_SGP30-1.0.0.13
Adafruit_TCS34725
Adafruit_TSL2591
AM2320
ArduinoJson-6.x
AS_BH1750
Blynk
Bridge
ccronexpr
ESP32_ping
esp8266-oled-ssd1306
ESPEasy_ESP8266Ping
ESPEasySerial
Esplora
Ethernet
Firmata
GSM
HeatpumpIR
HLW8012_1.1.1
HT16K33
I2Cdevlib
IRremoteESP8266
Keyboard
LiquidCrystal
LiquidCrystal_I2C
MechInputs
Mouse
NewPingESP8266
pubsubclient
RN2483-Arduino-Library
Robot_Control
RobotIRremote
Robot_Motor
SD
SDM_Energy_Meter
SerialDevices
Servo
SpacebrewYun
SparkFun_APDS-9960_Sensor_Arduino_Library
Stepper
Temboo
TFT
TinyGPSPlus-1.0.2
WiFi

```

TD-er, 2020-02-03:
Try uninstalling that one, as it is also not included in the libraries we have in the ZIP file. (see lib directory)

TD-er, 2020-02-25:
I assume it is now fixed, so I will close this.
If it is still an issue, let me know.

closed 2020-02-25 by TD-er