letscontrolit/ESPEasy #2875: old ESP config not properly activated

tparvais, 2020-02-05:
I have a running ESP with many sensors. backup config to config.dat.
I upgrade the firmware to last release compiled locally from source published yesterday
Everythinh seems ok after reboot, but no sensors are working (DS18B20): i need to open eahc of them and then submit then close to make the reading working fine

uzi18, 2020-02-05:
what was your last working version?
please attach here your backup file

TD-er, 2020-02-05:
That may have to do with recent added features to that plugin.
I guess the plugin is not using proper defaults for those new features and thus gets into some limbo state.
Not sure how we can fix that right now, as it should be that either "0" values should be the default, or the internal "build" number, which is actually a file format revision number should be incremented and a transition for it should have been made.
But now we have too many versions out there that were saved with the right parameters, so now we cannot patch those versions as we simply don't know what version introduced the changes and what settings were saved before and what after.

tparvais, 2020-02-05:
i as running end of year release (compiled locally) then upgraded to release disclosed yesterday (i need fix ip address fix, that works fine now)

@TD-er i understand your concern. btw, this is not a blocking bug.

config.data attached

uzi18, 2020-02-06:
In fact it is not blocking bug, because someone will find sensors not read properly and will submit settings.
We just want to be sure if we can tweak this situation for future, to not have such issues reported.
Will check it, thanks for report

TD-er, 2020-02-06:
> config.dat cannot be attached, .dat is not supported by github

You can maybe ZIP it, or share via dropbox and give us the link?

uzi18, 2020-02-06:
just zip it and drop here

tonhuisman, 2023-09-02:
This can be closed, no longer relevant.

closed 2023-09-02 by TD-er