letscontrolit/ESPEasy #2877: OLED SSD1306 laylout issue

dynamicdave1515, 2020-02-06:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

dynamicdave1515, 2020-02-06:
I'm using the ESP-Easy release 20200204 although this problem appears in previous releases.

I'm sending a series of commands (like the following) from Node-RED to a Wemos D1 Mini using MQTT.

msg.payload = "oledcmd,clear";
node.send(msg);

msg.payload = "oled,1,1,Weather Report"; // OLED line 1
node.send(msg);

msg.payload = "oled,2,1,Line 2"; // OLED line 2
node.send(msg);

msg.payload = "oled,3,1,Line 3"; // OLED line 1
node.send(msg);

They work fine, BUT if I alter the starting column on say line-3, it wipes-out row-1 off by the same amount. Couple of examples are attached.

[image]
[image]

TD-er, 2020-02-06:
I took a quick look at the code to see if there were some obvious mistakes, but couldn't find any.

Right now I cannot explain why it is even clearing the positions in front of "Line 3".
If this was meant to be cleared, then a quick fix could be to add spaces to the front of the line before writing it, so you never have to set the column as write position.

There is however another bug, or at least something I don't understand.
The code has some support for size optimized printing of characters, but that character width is not used to set the start of the string.
So writing "iiii1111" in 2 parts like writing "1111" from pos 5 on the line and "iiii" from the first position does show some space in between parts if using size_optimized.

dynamicdave1515, 2020-02-06:
My idea was to centralise the City and Country in the SSD1306 rather than having it left-aligned -thought it would look nicer!!!.
When I try to do that, it rubs out the same number of characters on row-1.
e.g. If I say start on col-3, it rubs-out the fisrt three chars or blanks them.

[image]

TD-er, 2020-02-06:
Just curious, what happens if you draw the lines in opposite order.
For example if you were writing on line 3 and then add something on line 2 with a column offset.
Will it then still clear the characters at the start of line 1?

dynamicdave1515, 2020-02-06:
Thanks for all your help and advice.

The simple workaround is to make sure the text on row-1 is the last item sent to the OLED panel.

TD-er, 2020-02-06:
So it is always part of line 1 that gets cleared?

dynamicdave1515, 2020-02-07:
I think so - I've not seen any other rows that get messed-up.

Also it doesn't seem to matter which row I choose to offset (they all seem to affect row-1 in the same way).

juggledad, 2020-02-09:
I took a look at this and I think I know what is happening. When you send something to the OLED, the code is always building a full  buffer 16 bytes (see _P023_OLED.ino)
`        else if (cmd.equalsIgnoreCase(F("OLED")))
        {
          success = true;
          String text = parseStringToEndKeepCase(arguments, 4);
          text = P023_parseTemplate(text, 16);
          Plugin_023_sendStrXY(OLED_Settings[index], text.c_str(), event->Par1 - 1, event->Par2 - 1);
        }
`
when the 16 bytes is sent with an offset, it overlays part of the next 'page' (line) on the OLED's memory buffer wiping out what is there.

TD-er, 2020-02-09:
Check, so we should limit the number of characters written to what's left on the line.

juggledad, 2020-02-09:
One way might be to calculate the lenth of the string sent - so if I was sending 'Weather Report' that's 14 characters. If I use a offset of 5 the length is now 19 so 3 characters need to be loped off the end of the text so you send 'Weather Rep' at an offset of 5.

That way it won't overlay what is in the next 'page' of the OLED's memory
 the offset should be added to the length of the data and  so. If I'm sending 'Weather Report' and you have a 128x64 device, it means you could have 16 characters on line

tonhuisman, 2020-03-08:
I took a look at the issue, and the original gripe, line 1 being partially wiped, seems to have been fixed somehow, but now, and maybe it did that before, the excess characters (actually pixels) overflow to the next line.
Assuming that isn't desirable as well, I fixed this by adding pixel-counting to Plugin_023_sendStrXY to prevent overflowing onto the next line. I tested with both Normal and Optimized Font Width settings, but as I have only 128 pixel wide displays, I couldn't test with a 64 pixel display, but I did take that width in account too.
I'll make a pull request with my fix.

PR #2935 (2020-03-11): [P023] Fix for text overflowing onto next line(s)
Added pixel-counting to Plugin_023_sendStrXY to prevent overflowing onto the next line.
I tested with both Normal and Optimized Font Width settings, but as I have only 128 pixel wide displays, I couldn't test with a 64 pixel display, but I did take that width in account too.

Fixes #2877
closed 2020-03-11 by TD-er