letscontrolit/ESPEasy #2878: Generic HTTP don't work since ESP_Easy_mega-20200204_normal_ESP8266_4M1M.bin

fly74, 2020-02-07:
The release before works fine.

Controller Publish: demo.php?name=%sysname%&task=%tskname%&valuename=%valname%&value=%value%

but the Apache receives:

`"GET /%2fdemo.php%3fname%3dESPBLK%26task%3dBLK%26valuename%3dTemperature%26value%3d4.24 HTTP/1.1" 404 985 "-" "ESP Easy/20104/Feb  4 2020 22:55:05"`

before withount %-coding:

`"GET /demo.php?name=ESPBLK&task=BLK&valuename=Humidity&value=58.95 HTTP/1.1" 200 31 "-" "ESP Easy/20104/Dec  8 2019 18:04:38"`

so it cant be processed with the apache.php

TD-er, 2020-02-07:
Hmm, so there is a step of URL encoding done at the wrong moment and variables not replaced?

Just to be sure, 20191208 was working fine?

fly74, 2020-02-07:
Yes I rolled back to this version.

TD-er, 2020-02-07:
Hmm looks like this commit: https://github.com/letscontrolit/ESPEasy/commit/c0eb4ec28f5c04b499fae1937339cd63c6c92e58

fly74, 2020-02-08:
What was the reason for the change ?

TD-er, 2020-02-08:
Not sure, I will have a look at it. Like I said in other topics also, I have been way too busy last weeks with moving to a new place and this was done in the preparation of that. (almost 2 months ago)

fly74, 2020-02-08:
Is it quick fixable?

TD-er, 2020-02-08:
Not sure, I will have a look now, as I just got home so I am now at a computer. :)

fly74, 2020-02-20:
Is there an release planned that fix it?

TD-er, 2020-02-20:
Planned is maybe to big a word, but I will for sure want to add an fix for it asap.
Last few days were a bit hectic again :(

TD-er, 2020-02-21:
OK, I have a fix ready and if I don't see anything else strange happening, I will merge it later this evening.

closed 2020-02-21 by TD-er

fly74, 2020-02-22:
Confimed it is fixed. THX.