letscontrolit/ESPEasy #2880: ESP32 goes into infinite look after configuring a serial server device.

cocooma, 2020-02-07:
Please see the serial log.
17549 : Info  :  Webserver args: 0: 'index' length: 1 1: 'page' length: 1
21797 : Info  :  Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'edit' length: 1 4: 'page' length: 1
21970 : Info  : SaveToFile: free stack: 7128
21996 : Info  : FILE : Saved /config.dat
21997 : Info  : SaveToFile: free stack after: 7128
22004 : Info  : SaveToFile: free stack: 7048
22400 : Info  : FILE : Saved /config.dat
22400 : Info  : SaveToFile: free stack after: 7048
31947 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 201072 WiFiStatus 3
38364 : Info  :  Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 4 4: 'TDE' length: 2 5: 'p020_port' length: 4 6: 'p020_baud' length: 4 7: 'p020_data' length: 1 8: 'p020_parity' length: 1 9: 'p020_stop' length: 1 10: 'taskdevicepin2' length: 2 11: 'taskdevicepin1' length: 2 12: 'p020_rxwait' length: 1 13: 'p020_events' length: 1 14: 'edit' length: 1 15: 'page' length: 1
38366 : Info  : SaveToFile: free stack: 6920
38463 : Info  : FILE : Saved /config.dat
38464 : Info  : SaveToFile: free stack after: 6920
38466 : Info  : SaveToFile: free stack: 7192
38551 : Info  : FILE : Saved /config.dat
38551 : Info  : SaveToFile: free stack after: 7192
38559 : Info  : SaveToFile: free stack: 7112
38982 : Info  : FILE : Saved /config.dat
38982 : Info  : SaveToFile: free stack after: 7112                                                                                                                                                                                                                                                                         ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
⸮U58 : Info  : 

INIT : Booting version: mega-20200204 (ESP32 SDK v3.2.3-14-gd3e562907)
58 : Info  : INIT : Free RAM:290716
59 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
68 : Info  : FS   : Mounting...
98 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
113 : Info  : CRC  : SecuritySettings CRC   ...OK 
176 : Info  : INIT : Free RAM:285124
177 : Info  : INIT : I2C
178 : Info  : INIT : SPI not enabled                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
⸮U58 : Info  :

TD-er, 2020-02-08:
It should at least disable the plugin/controller which is causing crashes.

What config do you use for the serial port?

cocooma, 2020-02-08:
The serial interface works until I restart the Esp.
After the restart the esp goes into the loop.
The config is baud 9600, 8 data, 1 stop bit evening else is left default.

TD-er, 2020-02-08:
What pins do you use?

cocooma, 2020-02-08:
I used 40 41 https://images.app.goo.gl/qdr2BvdGzUgxPaG5A

cocooma, 2020-02-08:
I've tried the firmware from the wiki https://www.letscontrolit.com/wiki/index.php/ESPEasy32.
It worked fine apart from this one was not restarting but for some reason the serial server stoped working after a restart. So after every restart had to reenable it

tonhuisman, 2023-09-02:
This is most likely fixed, so can be closed.

closed 2023-09-02 by TD-er