letscontrolit/ESPEasy #2881: update docs: GPIO trigger for rules

s0170071, 2020-02-09:
Hi all, 
Here: https://github.com/letscontrolit/ESPEasy/issues/1018
is a discussion about GPIO changes triggering rules. Nice. 

Sadly, I cannot find this new trigger neither here: 
https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference
nor here: 
https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules

TD-er, 2020-02-09:
See [Read The Docs](https://espeasy.readthedocs.io/en/latest/Reference/Events.html#internal-core-events)

s0170071, 2020-02-09:
Ah, thanks. Is it possible to link this from the other sites ? They are still the first results in google. 
Another question: is there a way to poll the GPIO pin states by means of a command ? Just like a digitalRead() ?

TD-er, 2020-02-09:
I think you could try [` [Plugin#GPIO#Pinstate#N]`](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#special-task-names), but I am not 100% sure about this.
I think you also must enable the monitoring on that pin.

s0170071, 2020-02-09:
Hmm. This works inside the rules. I need the MQTT or http command so I can integrate the I/O into homematic directly without configuring plugins. All the GPIO commands are for setting the state, i.e. pulse, LongPulse, GPIO, ... There is no command to read the state like digitalRead() would.

http://192.168.1.21/control?cmd=[Plugin#GPIO#Pinstate#N] 
does not work obviously.

TD-er, 2020-02-09:
Hmm it looks like we don't have a query command, but you could store the state in a dummy and write rules to update these dummy values on gpio state changes.

Maybe @giig1967g has another idea for this?

s0170071, 2020-02-09:
There is 
http://192.168.1.21/control?cmd=status,gpio,12
but this works only if there is a switch plugin configured and connected to GPIO12
If there is not, a JSON error is returned.

TD-er, 2020-02-09:
Well the problem here is that we can evaluate the value of the GPIO state but we don't have a http command to return the result of this "1" or "0" as a reply.

giig1967g, 2020-02-09:
there is the let command that works with http:
```
let,1,[Plugin#GPIO#Pinstate#N]
http://192.168.1.21/control?cmd=let,1,[Plugin#GPIO#Pinstate#N]
```

this command stores the pin state in %v1% variable.

giig1967g, 2020-02-09:
> 
> 
> Well the problem here is that we can evaluate the value of the GPIO state but we don't have a http command to return the result of this "1" or "0" as a reply.

I have some ideas, but we need to merge the GPIO core commands first.

giig1967g, 2020-02-09:
> 
> 
> There is
> http://192.168.1.21/control?cmd=status,gpio,12
> but this works only if there is a switch plugin configured and connected to GPIO12
> If there is not, a JSON error is returned.

as soon as the gpio core commands are merged, the status command will be available even if there is no plugin defined.

thomastech, 2020-03-12:
> Is it possible to link this from the other sites ? They are still the first results in google.

Links to the new readthedocs have been added. See the top line on these pages:
 https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference
 https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules

- Thomas

tonhuisman, 2023-09-02:
This seems outdated now, can be closed.

closed 2023-09-02 by TD-er