letscontrolit/ESPEasy #2882: Wifi looses connection after several Minutes

Coalado, 2020-02-09:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2020-02-09:
Hmm, that's not really informative and ignoring the issue template.

Coalado, 2020-02-09:
ups. I'm sorry. I didn't mean to "send" the issue. I will fill required information asap

Coalado, 2020-02-09:
Since a have 3 APs with the same ID (router, 2 repeater), it seems like the module disconnects due to 
(Disassociated because sending STA is leaving (or has left) BSS | Operating System moved the client to another access point using non-aggressive load balancing.) and fails to reconnect to (maybe another AP) due to "Auth expired" (Previous authentication no longer valid | Client has associated but is not authorised.)
Even if I reboot the device, I usually get the same "Auth expired" again for some time.

Coalado, 2020-02-10:
I could not reproduce the issue on 2 other MCUs.

I remember reading something about Wifi issues on devices that ran on an older version of espEasy.
The device that has problems was set up with EspEasy r120 before. May this be the problem?

TD-er, 2020-02-10:
Well when in doubt, you can alway fully erase the flash first and then program the firmware on it.
Older versions (or other firmware versions) could have saved WiFi settings in a part of the flash we don't erase or write on ourselves.
These can still be used when booting the ESP, so that's why it is often advised to clear the flash first using the blank images.

toast1234, 2020-02-11:
As I don't have that deep knowledge... But am curios to understand under what conditions a blank flash is relevant. Can you give an example? What type of hidden settings can be left when upgrading the SW? Using Reset to factory default will not clear those parameters? Is it only relevant when coming from other SW than Espeasy or very old builds?

Thanks

TD-er, 2020-02-11:
ESPEasy does try to use the WiFi with "persistent" disabled.  (for over 2 years now)
This is to make sure the flash will not wear out.

Some firmware versions and also self-built versions (with code of your own, or some examples) do not disable this persistent mode and thus the Arduino lib does write to flash every time the ESP makes a (successful) WiFi connection.
It does write the last used WiFi settings like SSID, BSSID, wifi key and channel (and some other settings also) to the same address in the flash, in a location we don't use for our settings, sketch or file system.
So after numerous writes (a few 1000 times), the flash will wear out on that spot and it will never be able to connect to WiFi again.
That's the reason we don't use this persistent mode, but that doesn't mean that if there is some working setting still in there, it may already have triggered a WiFi reconnect before our code is even started to execute.  Some boots it may have an effect and sometimes it doesn't, so that's why it may result in unpredictable WiFi stability issues and that's why completely erasing the flash may be of help.

closed 2022-11-09 by TD-er