letscontrolit/ESPEasy #2883: pulsecounter interval is used, even if send to controller is not selected

GravityRZ, 2020-02-09:
currently using latest build Mega20200204

i am using the pulsecounter plugin but do no want to send info to the controller so i deselected "send to controller"

as an alternative i am using a rule which triggers whenever the pulse counter(Watermeter#Counter) gets triggered. This works very well but i noticed the device interval is still used despite the fact that the Counter is being triggered.
e.g.

pulse counter interval 1 sec
On Watermeter#Count do
              if [Watermeter#Count] > 0
Publish domoticz/in,'{"idx":337,"nvalue":0,"svalue":"1"}'   // send 1 liter pulse to domoticz
this works when pulses are received and rule is executed

pulse counter interval 60 sec
On Watermeter#Count do
             if [Watermeter#Count] > 0
Publish domoticz/in,'{"idx":337,"nvalue":0,"svalue":"1"}'   // send 1 liter pulse to domoticz

this does not work. When pulses are received the rule does not get triggered before 60 seconds have passed. 

this should not be the case
either the interval gets used ehen send to controller is selected or it should be skipped(not looked at) when you are trying to do things in rules.

interval=0 would be my ideal setup(only send pulse when one is detected) but this does not work

TD-er, 2020-02-09:
The interval is not related to a controller.
What it does is calling `PLUGIN_READ` of the plugin.
This will (not always) yield a reading and call connected controllers.
It will also create an event which can be used in the rules.

What the `TaskRun` command does, is call the `PLUGIN_READ` function.

Some plugins, like the GPS plugin, do offer an extra feature.
The GPS plugin does allow to set another value to trigger a new `PLUGIN_READ` if the traveled distance since the last update is more than the set threshold.
If it sends an update, it will then reset the "interval" timer.
This means the interval between samples is at most the "interval" duration.

Is this the behavior you also like for the pulse counter?

GravityRZ, 2020-02-10:
Hi TD-er

explanation is clear(thought it was related to the controller)

if the threshold could be set from 1-60 then indeed this would be a good solution for the pulse counter as well

i however do not know if this is easy to do.
You would need to know the current counter total and if the total changes more then the treshhold it would send out a plugin_read

maybe it is easier to make it possible to select interval 0(no interval) and it only reacts on changes.

eg interval set to zero
call plugin_read only when a change is detected(a pulse)

this makes it more consistent with other plugins which also have the interval=0 option
Device[deviceCount].TimerOptional = true;

for now the 1 second interval works but somehow i have the idea that there is a chance that we can miss a pulse.
eg when a pulse is send and exactly on that moment the interval kicks in.

at the moment everything works stable but still once in a while i miss a pulse and i do not know why.
is it the interval, is it the 30 seconds timer in rules

TD-er, 2020-02-10:
Well the implementation of plugins differs among plugins.
Some do nothing until the `PLUGIN_READ` is called, so a change is then unknown until it is actually reading it.
Some other plugins (e.g. the GPS I mentioned) do give samples continuously and thus the call to `PLUGIN_READ` just yields a success if a value was updated since the last time and that threshold setting is then very easy to implement.

The pulse counter is triggered by an external source, so it already does resemble the way the GPS plugin operates.
A 'decimate' feature does sound like a useful feature for the pulse counter as it would otherwise send out lots and lots of events if the pulse frequency is too high.

tonhuisman, 2023-09-02:
This can be closed.

closed 2023-09-02 by TD-er