letscontrolit/ESPEasy #2884: P088_HeatpumpID does use way too many resources even when not used

TD-er, 2020-02-10:
See:
```c++

// Array with all supported heatpumps
HeatpumpIR *heatpumpIR[] = {new PanasonicCKPHeatpumpIR(), new PanasonicDKEHeatpumpIR(), new PanasonicJKEHeatpumpIR(),
                            new PanasonicNKEHeatpumpIR(), new PanasonicLKEHeatpumpIR(),
                            new CarrierNQVHeatpumpIR(), new CarrierMCAHeatpumpIR(),
                            new MideaHeatpumpIR(), new FujitsuHeatpumpIR(),
                            new MitsubishiFDHeatpumpIR(), new MitsubishiFEHeatpumpIR(), new MitsubishiMSYHeatpumpIR(), new MitsubishiFAHeatpumpIR(),
							new MitsubishiKJHeatpumpIR(),
                            new SamsungAQVHeatpumpIR(), new SamsungFJMHeatpumpIR(),new SharpHeatpumpIR(), new DaikinHeatpumpIR(),
                            new MitsubishiHeavyZJHeatpumpIR(), new MitsubishiHeavyZMHeatpumpIR(),
                            new MitsubishiSEZKDXXHeatpumpIR(),
                            new HyundaiHeatpumpIR(), new HisenseHeatpumpIR(),
                            new GreeGenericHeatpumpIR(), new GreeYANHeatpumpIR(), new GreeYAAHeatpumpIR(),
                            new FuegoHeatpumpIR(), new ToshibaHeatpumpIR(), new ToshibaDaiseikaiHeatpumpIR(),
                            new IVTHeatpumpIR(), new HitachiHeatpumpIR(),
                            new BalluHeatpumpIR(), new AUXHeatpumpIR(),
                            NULL};
```

This array is creating all (!!) supported models in an array.
This means all data that each class allocates as static or as member data will be taking resources on the heap.
This is even when P088 is not selected or enabled.

The best way for this is to use the "factory" design pattern to create the object when needed.
These classes all inherit from the `HeatpumpIR` class, so this factory can just return a pointer of this type and when using it, you can cast it to the right type.

Another bug in this is that the base class does not have a virtual destructor.
Right now, we don't destruct the objects, but if you make a factory for it, you should also be able to destruct it and thus you need to have a virtual destructor as you will call the delete on the base class pointer.

ToniA, 2020-02-10:
I'll take a look at this.

TD-er, 2020-02-21:
Should be fixed by the merged PR #2887

closed 2020-02-21 by TD-er