letscontrolit/ESPEasy #2888: Add some function callbacks to the source code

FanOfHue, 2020-02-15:
Since i've stopped working on ESPCoreRules, i moved back to native ESPEasy and started to work on some new plugins and customization. For some of my custom plugins to work properly, i need to have some more customized parsing of templates and events. (like parsing of string variables and JSON payloads in rules/events)

Similar to the way MQTT is handled in the code, i've customized my ESPEasy to use function callbacks for this.
It's only a very minor change to the code with no more overhead than checking for a zero-pointer when not activated.

Global section somewhere, add two global function pointers:

```
  void (*parseTemplate_CallBack_ptr)(String&)=0;
  void (*substitute_eventvalue_CallBack_ptr)(String&, String)=0;

```
Function ParseTemplate(), add two lines of code:

```
  if(parseTemplate_CallBack_ptr)
    parseTemplate_CallBack_ptr(line, event);
```

Function substitute_eventvalue():, add two lines of code:

```
  if(substitute_eventvalue_CallBack_ptr)
    substitute_eventvalue_CallBack_ptr(tmpString);
```

and remove this check:
```
  if (line.indexOf(F("%eventvalue")) == -1) {
    return; // Nothing to replace.
```

This way it's really easy to add custom code to ESPEasy without having to touch the stock source files.
You can add custom handlers to a custom file or a custom plugin.

Could this (or something similar) be implemented?

TD-er, 2020-02-15:
> Could this (or something similar) be implemented?

Right now, I don't see why not, but have to look at the code to see its impact.
The last one does look like an optimization I made, so it may have impact on performance?
Would it also be OK to have changes which may have an performance impact wrapped in #ifndef statements, so you could simply hook in your stuff when that define has been set?

Later if your changes will be ready to merge, we can look into more elegant way of handling those.

I do like the idea of having call back pointers in handling events or template parsing. It would make stuff way more extendable.

TD-er, 2020-02-15:
By the way, I am working very hard on restructuring some stuff in the controller handling, which is a merge which is very close to be finished, so will look to include this also.

FanOfHue, 2020-02-15:
Thanks for looking into this request!

I agree that adding something like #ifdef CALLBACKS around the new code will make sure that is has zero impact on existing compiles.

And then the other check like this:
```
#ifndef CALLBACKS
if (line.indexOf(F("%eventvalue")) == -1) {
    return; // Nothing to replace.
#endif 
```
Or have a limited check like this:
```
if (line.indexOf(F("%")) == -1) {
    return; // Nothing to replace.

```

TD-er, 2020-02-21:
I'm about to make these changes.
Maybe you can come up with a better (more descriptive) define for this code?
CALLBACKS is way to generic.

Should I call them `PARSE_TEMPLATE_PLUGIN` and `SUBSTITUTE_EVENTVALUE_PLUGIN` ?

TD-er, 2020-02-21:
Hmm fixed in in a more elegant way. See the PR I made.

closed 2020-02-21 by TD-er