letscontrolit/ESPEasy #2890: P036_FrameOLED: After framedoledcmd,n,text command the display is turned on

tonhuisman, 2020-02-16:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request
My setup is that values from my PV system are extracted from the inverter (ZeverSolar 3500S) and fed as numeric values using TaskValueSet to display on an OLED display connected to ESPEasy, updated every minute. This works as expected.
In addition to the numeric values, in my script (running on the RPi that also hosts Domoticz) I determine the initial start-time of the PV inverter, and the time the last update was retrieved. This is sent to ESPEasy using the oledframedcmd,<lineno>,'multi+word+text'[,<reserved>] command.
I have also set a display timeout of 120 sec. to extend the life expectancy of the OLED, and added a button to turn on the display when I want to read the latest values (I'm _not_ constantly watching the display...:P)
Side-effect of the oledframedcmd command, and reason for this request, is that the display is explicitly turned **on** after receiving a new text (see _P036_FrameOLED.ino around line 755
```
displayTimer = PCONFIG(4);
if (UserVar[event->BaseVarIndex] == 0) {
    // display was OFF, turn it ON
    display->displayOn();
    UserVar[event->BaseVarIndex] = 1;      //  Save the fact that the display is now ON
}
```
)
I would like to suggest to use the (reserved) 3rd argument of this command to optionally inhibit this turning on of the display, as currently there is no way to avoid waking up the display. The 8 bit value that is reserved for 'reserved' could be bit-mapped to facilitate 8 flags (1, 2, 4, 8, 16, 32, 64 and 128) to be passed.
Something in the lines of this:
```
// displayTimer = PCONFIG(4); // Remove from here
if (UserVar[event->BaseVarIndex] == 0 && !(P036_DisplayLinesV1[LineNo-1].reserved & 0x01)) {
    displayTimer = PCONFIG(4);
    // display was OFF, turn it ON
    display->displayOn();
    UserVar[event->BaseVarIndex] = 1;      //  Save the fact that the display is now ON
}
```
- (the 0x01 (and similar bit values) should probably be made a #define to avoid magic numbers, and give it a meaningful name)
- (the 'reserved' field should probably be renamed to OptionalFlags or something similar)

Resetting the display timer should also be moved to the point where the display is actually turned on, as it will never expire if any update is sent with a shorter interval than the display time-out. (And that would turn out to be a bug, after implementing this feature)

### Expected behavior
Optionally not enable the display after sending text to display on one of the lines.

### Actual behavior
The display is always turned on after sending some text.
This doesn't (and shouldn't!) happen when updating the displayed content by updating a variable or sensor value.

### Steps to reproduce

1. Send a text to the display using oledframedcmd,<lineno>,<text>[,<reserved>] command while the display is turned off after the timeout. It will turn on.

N/A

### System configuration
Hardware:
- Wemos D1 mini clone
- SSD1306 OLED display (128x64 & 128x32)

ESP Easy version: ESP_Easy_mega-20191208_test_beta_ESP8266_4M1M

ESP Easy settings/screenshots: N/A

### Rules or log data

TD-er, 2020-02-16:
As a programmer I am fond of bit mapped values, as they do nicely squish as much info as possible in memory.
But believe it or not, some people don't get it how to use bitmasks.

Maybe we should just make this a setting and then you should switch the display on/off in the rules to draw attention to specific notifications.
So the settings will then be something like "Turn on display on receiving text command" but then a bit shorter :)

tonhuisman, 2020-02-16:
I'm a programmer too, you might have guessed ;-), so I just wanted to keep the <reserved> argument as versatile as possible, hence the bitmap solution. Adding proper documentation should resolve any issue regarding the use of this argument. And most people using these commands are already a bit technical, as most unskilled/non-technical wouldn't know how to send commands this way.

Adding a setting for this would be a viable solution, thank you, it could even be disabled by default (though that wouldn't be backward compatible).
I just want to keep the display off when 'merely' receiving a new string to display, it doesn't turn on when a sensor/variable value is updated, AFAIK. If it would be an alert, I can turn on the display explicitly using the oledframedcmd,display,on command, and adjust the contrast using oledframedcmd,display,high if needed. (I have planned to add a button for that too, in my hardware-build, easily done using a rule.)

tonhuisman, 2020-03-04:
I've crafted the required changes myself, but now I'm stuck at the git process, committed to a branch, but now puzzled at how to start a pull request... (obviously not a git user here :-])

TD-er, 2020-03-04:
Well the process is quite simple (if you know it...)

Just as a precaution for inexperienced Git users. Copy your files first into a ZIP file or something to make sure you can mess up big time and not loosing anything :)

You probably have done most of it, but just the procedure:
- Fork the project to your own account.
- Use `git clone` to clone your own project to your hard drive.
- Create a branch at the HEAD of our master branch (called `mega` in our repository)  (e.g. `bugfix/2890_framed_oled` )
- Checkout this new branch
- Start hacking, compiling etc.
- Commit your changes to this branch you made
- push your branch to your repository on GitHub  (and thus also all the commits you made on it)
- Move to the ESPEasy repository on GitHub and you will probably see a yellow-ish bar on top when you click on the "code" or "Pull Requests" tab suggesting to create a new pull request of your just pushed branch.
- Create a pull request.

What you're essentially doing when creating a pull request, is the request to merge your branch with our repository's branch. (very likely the `mega` branch)
This means that if you push new commits to the same branch, they will be appended to the existing pull request.

So if you want to do some other work, you go back to your `mega` branch, create a new branch and repeat.

tonhuisman, 2020-03-05:
Thanks, just did all those steps, now to get used to the Github etiquette :-)

TD-er, 2020-03-05:
Better get used to it, as I would like it when more people become frequent contributors :)

PR #2925 (2020-04-10): [P036] Added setting for 'Wake display on receiving text'
Resolves #2890
closed 2020-04-10 by TD-er