letscontrolit/ESPEasy #2891: DS18B20 value when in Task #4 returns always 0

tparvais, 2020-02-16:
I observe that if I configure a DS18B20 in 4th row , the measure is always 0

I have this with different Wemos D1 Mini

Last Mega release be. of February. I only change the number of task from 12 to 24 in custom.h

Thomas

uzi18, 2020-02-16:
so if you changed 12->24 tasks you need to wipe out configuration - they are not compatible thats why you have problems.

tparvais, 2020-02-16:
Hi can you be morte explicit? I don’t understand what I have to,do practically

Domosapiens, 2020-02-16:
I have seen similar Task 4 problems:
https://github.com/letscontrolit/ESPEasy/issues/2786#issuecomment-559615017
I resigned myself to it and moved the DS18B20 to another task number.

uzi18, 2020-02-17:
first you need to flash blank.bin to your module and reflash new firmware, than reconfigure

@Domosapiens  I guess gpio-16 is not well suited for dallas, try other gpios

TD-er, 2020-02-17:
> Last Mega release be. of February. I only change the number of task from 12 to 24 in custom.h

Can you run `meminfodetail` from the command field on the tools page?
If you have executed that command, the next time you open the system info page, it will show a graphic layout of the settings file at the end of the page.
Just check there is no overlap in data segments in your settings file.

And like @uzi18 suggested, also make sure you don't use the wrong pins (probably nothing to do with this issue, just a generic remark)

If you have it in position 4, are there other instances of the Dallas plugin active?
If so, at what position?

I probably also replied something similar in the issue linked by @Domosapiens , but just as generic info about what happens in the ESP when processing tasks.
It will process the tasks in the order of the task ID.  (rows in the table on the "Devices" page)
So if the first 4 tasks are using Dallas sensors, then it may be the time to process them is too short and may reveal a bug in the timing of that plugin.
By moving it lower in the list (and having another tasks in between) it may just have enough time to process the other sensors data.

So what you could do to test, is try to make task nr 4 the first Dallas task and see if that makes the bug move "lower" on the list of tasks.

uzi18, 2020-02-17:
please paste here screenshot from your devices configuration

tparvais, 2020-02-17:
how to interpret this ?
[image]

tparvais, 2020-02-17:
[image]

TD-er, 2020-02-17:
The layout of the settings looks good. (no overlap)
You can see by the numbers in the last column that the sections do not overlap  eachother and the top line does show how the sections interleave in the settings file.
The bottom line of each type of section (represented with a color) does show how much of each section is used.

uzi18, 2020-02-18:
it is curious why exactly 4th task is not working properly, because other works well.
please try to set for task 4th dallas address from eg. 10th and vice versa.
maybe something is wrong with sensor itself (fake/clone? )

jimmys01, 2020-02-18:
Those are a LOT of sensors. Topology and wiring is also play also a big part in the success of this endeavour. Bravo for making it work at all!

tparvais, 2020-02-18:
I of course try to change the sensor that was working perfectly . Only task#4 gives issue

Domosapiens, 2020-02-19:
>  I guess gpio-16 is not well suited for dallas 

gpio-16 is excellent for DS18B20 (3 other sensors proved it on this unit and 14 other units proved it)

> Topology and wiring is also play also a big part

, yep, but not for this problem.

> First you need to flash blank.bin to your module

 ...and what function has update OTA ?

> So if the first 4 tasks are using Dallas sensors,

No ...never was the case

> So what you could do to test, is try to make task nr 4 the first Dallas task

No, initially task 4 was the first Dallas task, but I needed to move it
.
.
.

The remarkable and common observation is:

> Only task#4 gives issue

.....?

uzi18, 2020-02-19:
ok so for now please try to exchange tasks configuration

f4gqk, 2020-05-19:
Bonsoir tparvais

Je suis à la recherche du firmware qui permettrait d'avoir plus de 12 devices.

C'est pour un projet de chauffage où j'ai 8 zones. Chaque zone a besoin de deux sorties sur un MCP23017.

Pour pouvoir utiliser mon wemos et le MCP23017 complètement j'ai besoin de savoir où je peux télécharger le nouveau fichier compilé et modifié.

Merci d'avance

Sébastien

tonhuisman, 2020-05-19:
> Bonsoir tparvais

Please don't use Github issues for generic support requests, and please use English, as it is the common language here.
I have replied to your question in the forum.

f4gqk, 2020-05-19:
Good evening tparvais

I'm looking for the firmware that would have more than 12 devices.

It's for a heating project where I have 8 zones. Each zone needs two outputs on an MCP23017.

To be able to use my wemos and MCP23017 fully I need to know where I can download the new compiled and modified file.

thank you in advance

Sebastien

tonhuisman, 2023-09-01:
Dallas temperature sensors have got a big overhaul some time ago so this issue can be closed.

closed 2023-09-01 by TD-er