letscontrolit/ESPEasy #2893: Feature Improvement to connectionFailures

thomastech, 2020-02-17:
### Summarize of the feature request
### Current behavoir:
When a controller fails to publish a message the **connectionFailures** variable is incremented. And if the connectionFailure count exceeds the _Connection Failure Threshold_ setting the system reboots. Unfortunately the connectionFailures value is a persistant count that requires a reboot to reset back to zero.

This is an issue since random connections failures can occur in a stable system due to a busy MQTT controller, temporary WiFi signal issues, etc.  Over a period of days or weeks the count will eventually exceed  _Connection Failure Threshold_ setting. And a unwanted reboot will occur.

Of course the _Connection Failure Threshold_ setting_ can be disabled and not used at all. But in some situations it would be desirable to force a reboot after too many sequential failed publish attempts have occurred.

### Expected behavior
The connectionFailures count should be reset to zero after the system has successfully reconnected. The reset should be qualified by requiring several successful connections in a row before clearing the count. This new behavior could be implemented as a checkbox option on the Advanced settings page. Perhaps something like "Non-persistent Failure Threshold [ ]"

- Thomas

TD-er, 2020-02-17:
The counter does go down to 0, but not immediately after the first successful connection.
I have to check the code, but the general idea is this:
- +1 for a failed connection attempt.
- -2 for a successful connection attempt.

giig1967g, 2020-02-17:
> And if the connectionFailure count exceeds the _Connection Failure Threshold_ setting the system reboots.

Why does it reboot after the count exceed the Connection Failure Threshold?

TD-er, 2020-02-17:
> Why does it reboot after the count exceed the Connection Failure Threshold?

It is by design, although the functionality may need a more descriptive term.

thomastech, 2020-02-17:
> The counter does go down to 0, but not immediately after the first successful connection.
I have to check the code, but the general idea is this:
    +1 for a failed connection attempt.
    -2 for a successful connection attempt.

That might be what was intended, but in practice it is not working to our benefit.

Several hours ago I stopped the broker, waited a few seconds, then restarted it. ESPEasy's connectionFailures count had been zero, but incremented to 10. The count has remained unchanged since the brief shutdown, despite hundreds of successful published messages in the meantime. 

This unexpected behavior caused me a lot of hair pulling.

- Thomas

TD-er, 2020-02-17:
OK, will have a look at it, to see if it got broken somewhere.

FYI: the commit I reverted 2 weeks ago which was partly breaking the static IP stuff, also had changes to make handling the failed/success count in a single function. So then all controllers, etc. were using the same count function for the successful connection count.

So I will have a look at it again, to see where I may have messed up in that commit.
This means it is possible not all controllers use the same count logic right now as that was what I was trying to fix back then.

thomastech, 2020-02-17:
@TD-er, thanks for looking into it. 

I should have mentioned that the latest testing has been on ESP_Easy_mega-20200204_test_ESP8266_4M1M_VCC.bin. I am using the OpenHAB controller.

Sometime around Jan 2020 I enabled _Connection Failure Threshold_ (changed from 0 to 25). This was a Hail Mary attempt to stop the random reboots that have haunted my devices for months. As reward for enabling it I unknowingly made things worse. 

It's possible that the latest release is the only one that is broken. Or this bug may have been present for a much longer time span.

- Thomas

TD-er, 2020-02-17:
> It's possible that the latest release is the only one that is broken. Or this bug may have been present for a much longer time span.

I think the latter.

TD-er, 2020-02-17:
Just a short summary of my findings until now:
- Domoticz MQTT was using its own connection failure counter
- MQTTcheck was increasing +2 when it needed to reconnect and only -1 when the check found it was connected. (interval for MQTT check is not constant, but based on the number of messages)

So now I am moving all checks to the single function to keep track of the failed connection counts and it does now count on mqtt connect attempts. (as the counter name suggests)
For MQTT we still have the MQTT check function, which also decrements the counter.

It feels like we must just periodically lower this counter, but not sure how and when.
For MQTT it makes sense to have it in the MQTT check function as that one is called with an interval based on the use of that connection.
Not sure how we should decrement it for other controllers and maybe we should not worry about it as most do not keep a connection open at all.
I will run a few tests and then make a test build for you (which also is a good opportunity to test a few other changes I have pending :) )

thomastech, 2020-02-17:
No doubt there are many opinions on how the count should be reset. The decrement method seems biased towards short term publish problems. It wouldn't represent the conditions after a long period of MQTT inactivity (for example, long after WiFi interference has ended). 

Therefore, my preference is to zero the counter after two successful MQTT messages occur in a row. If there are scenarios that the decrement method is better then let's discuss them.

- Thomas

TD-er, 2020-02-18:
> Also, if multiple controllers are ever used then each would need there own connectionFailures counter. That is to say, they shouldn't all be lumped into one counter.

That suggests this parameter should also be part of the controller settings and not a general system wide parameter.

This leaves the question, what to do with other stuff that likes to connect to the outside, like MQTT-import plugin, publish command (which in a sense uses the controller for its settings) and other commands. Also we have NTP queries, p2p commands (which is also somewhat part of a controller nowadays) and DNS lookups.
What about DHCP attempts?

Well let's first see how your test build is working (is half way building right now) and for MQTT it is already programmed to decrease its value over time, so you may already have some of the functionality you suggested.

TD-er, 2020-02-18:
Here is a test build: [ESPEasy_mega-20200204-27-PR_2874.zip](https://www.dropbox.com/s/bd71sh0exikmj6n/ESPEasy_mega-20200204-27-PR_2874.zip?dl=0)

thomastech, 2020-02-18:
> Here is a test build:

Thanks for the test build. I installed ESP_Easy_mega-20200204-27-PR_2874_test_ESP8266_4M1M_VCC.bin  and checked the connectionFailures counter. It still is not working correctly for me.

Here is what I did.
1. Stopped MQTT broker.
2. Watched the ESPEasy weblog and waited until connectionFailures count got to 11. 
3. Restarted MQTT broker.
4. Continued to watch the log. After 70+ successful published messages (over a 15 minute period) the count remained at 11. It should have been reset much sooner.
5. Attached is the log.

[weblog.txt](https://github.com/letscontrolit/ESPEasy/files/4216990/weblog.txt)

> That suggests this parameter should also be part of the controller settings and not a general system wide parameter.

Agreed. The connectionFailure counter should be in the controller settings. It makes no sense to have all controllers contribute to the same counter.

> This leaves the question, what to do with other stuff that likes to connect to the outside, like MQTT-import plugin, publish command (which in a sense uses the controller for its settings) and other [...]

Those are good questions. I totally understand that a reboot is deserved if the existing WiFi STA connection is lost and re-connection fails after multiple attempts. That's a situation that may be caused by the ESP8266 core not behaving nicely.

But I wonder if there are any controller or NTP connection failures that deserve to cause a reboot? That is to say, if a controller is not responding, but WiFi is connected and working, will rebooting fix a non-responsive controller? Are there known cases that doing this is helpful?

- Thomas

giig1967g, 2020-02-18:
> > Why does it reboot after the count exceed the Connection Failure Threshold?
> 
> It is by design, although the functionality may need a more descriptive term.

It may explain my random reboots when the broker was powered off: see #2579 

When this reboot happens, what is the description in the "Reset Reason" and "Last Task" in the "System Info" page? In other words, is it possible to recognize this reboot from the reset reason?

Also, how does this Connection Failure Threshold Counter relates with the Controller Queue settings? I mean, my understanding is that the "Max Queue Depth" should limit the messages if there is no reply from the broker, but if in the "Max Queue Depth" is set to 20 and the "Connection Failure Threshold" is set to 10, the unit will reboot before reaching the queue limit. Is my understanding correct?

TD-er, 2020-02-18:
> Also, how does this Connection Failure Threshold Counter relates with the Controller Queue settings? I mean, my understanding is that the "Max Queue Depth" should limit the messages if there is no [...]

That's not correct.
The MQTT controllers do try to keep the connection open and thus are different from a HTTP controller.
So if there is no reconnect needed, it does push out as many MQTT messages as are needed, which can be for days/weeks for a single reconnect.

As Thomas showed, we need to clear this reconnect counter if it is working fine for N seconds or M messages of whatever metric we need to consider a network connection to be stable.

If there is a problem with the network, not a single connection attempt should succeed, so not a single message will be delivered to the broker from the controller buffer.
If a connection attempt is successful, but closed after just a few seconds, then some messages may have been delivered, so in that specific situation there is a possibility not all queued messages are delivered before a reboot happens.

But anyway, there is no relation between connection attempts and queue depth.

giig1967g, 2020-02-18:
I see. 
But I still couldn't understand from your expalnation if the reboot is triggered by the absence of WiFi connection of MQTT connection, and if it can end in a reboot loop.

Consider the following scenarios: Router + MQTT Broker + ESPEasy.
1) EspEasy unit cannot communicate with the wifi router (and therefore cannot communicate with MQTT broker too): after N times, it will reboot. 
Will it enter a reboot loop if the wifi router is down for any reason, or the reboot will happen only once?
2) EspEasy unit CAN communicate with the wifi router but cannot communicate with the MQTT broker. Will it reboot?

In my opinion, in scenario 1, it should reboot but should not enter a reboot loop.
In scenario 2, it should not reboot as the Wifi connectivity is working.

Thanks for the clarifications

TD-er, 2020-02-18:
If the connect failure threshold is set to something other than 0, then yes it will enter a reboot loop.
After all, there is a network connection, but connecting to the broker will fail every time.
So eventually this counter will reach the set threshold and thus reboot.

If you have not set this threshold, then it will retry to connect, but should not crash due to it.

If it does crash, then we have several possible issues why if crashes.
- a bug (nah, unlikely ;) )
- Running out of memory as the controller buffer does for some controllers buffer quite large strings.
- Pending connection requests may prevent other connections to fail which may cause watchdog resets or buffer overflows.

Especially the last one does seem like a problem which may be hard for me to fix.
I've seen several reported issues in the Arduino code which relate to the fact the set timeout may not be obeyed when a connection attempt was never really acknowledged by the other side.
So this then results in watchdog reboots if it is do a blocking wait for it. (default timeout is quite high and I always set it lower)

TD-er, 2020-02-18:
I now see why the counter is not decreased.
The call to MQTTcheck is only done if pubsubclient's loop() fails.
That one fails only if there is a problem.

So I will -for MQTT- decrease the connect failure counter for each successful publish. So that's one for each message.

This still isn't perfect as you may still have a few successful publish attempts as pubsubclient itself is also using a buffer which may hold one or more messages.
But at least it is occurring quite often and thus will bring the counter down quite fast.

I also noticed that some checks for "first enabled MQTT controller" do not use the new functions for checking if it is a valid controller index. Not sure if it was a bug, but better to have them all use the same logic to check for valid values.

giig1967g, 2020-02-18:
@TD-er 
Sorry to insist, but there is something I don't understand: why if wifi connection is up and mqtt broker is down the unit reboots?

TD-er, 2020-02-18:
@thomastech Can you test [this test build ESPEasy_mega-20200204-28-PR_2874.zip](https://www.dropbox.com/s/m8c32p5w23fgylb/ESPEasy_mega-20200204-28-PR_2874.zip?dl=0) ?

@giig1967g That's also something I cannot explain, if you have the "Connection Failure Threshold" set to 0.

giig1967g, 2020-02-18:
Hi @TD-er :
I have Connection Failure Threshold=10, but what I don't understand is why, by design, if WiFi is up and MQTT Broker is down the unit is  instructed to reboot...
For example, I have a unit that reads the power meter and sends the reading to a Google Sheet. The same unit uses MQTT to send the values to OpenHAB.
If MQTT broker is down for any reason (during an update of OpenHAB for example) I still want the unit to send the data to the Google Sheet; I don't want it to be rebooted.
Why should I loose readings just because the MQTT broker is down?

thomastech, 2020-02-18:
> Can you test this test build ESPEasy_mega-20200204-28-PR_2874.zip ?

Installed ESP_Easy_mega-20200204-28-PR_2874_test_ESP8266_4M1M_VCC.bin. The connectionFailures is reset to zero immediately after reconnection success. So it is now working!

> So I will -for MQTT- decrease the connect failure counter for each successful publish. So that's one for each message.

The log does not show evidence of a decrement action; It seems to be cleared upon connection success. See attached:
[weblog1.txt](https://github.com/letscontrolit/ESPEasy/files/4220500/weblog1.txt)

BTW, I also tried ESP_Easy_mega-20200204-28-PR_2874_test_beta_ESP8266_4M1M.bin and it would not run correctly. Although it was connected to the router and had the correct NTP time, the Web interface took several minutes to appear and is randomly sluggish loading pages. Also the Nextion plugin was missing some configuration data that needed to be re-entered.

- Thomas

TD-er, 2020-02-18:
> BTW, I also tried ESP_Easy_mega-20200204-28-PR_2874_test_beta_ESP8266_4M1M.bin and it would not run correctly. Although it was connected to the router and had the correct NTP time, the Web interface took several minutes (page timeouts) to appear and is randomly sluggish loading pages. Also the Nextion plugin was missing some configuration data that needed to be re-entered.

After restoring the Nextion plugin settings, was it fast again?
Still strange this data was mangled apparently. Will look into it.

@giig1967g 
Well by design, this parameter does exactly what you describe. 
Whether it is a good design, is debatable, as we also discussed here in this issue.

thomastech, 2020-02-18:
> After restoring the Nextion plugin settings, was it fast again?

No. The missing data was in the _Nextion Command Statements (Optional)_ area. These command strings are not mandatory for Nextion to operate.

> Whether it is a good design, is debatable, as we also discussed here in this issue.

Debatable for sure. Regardless, I suspect there are other users with "unexplained" reboots that would benefit by setting their _Connection Failure Threshold_ to zero. This should be mentioned in @Grovkillen's underutilized sticky: https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=5818

- Thomas

- Thomas

TD-er, 2020-02-18:
> No. The missing data was in the _Nextion Command Statements (Optional)_ area. These command strings are not mandatory for Nextion to operate.

Do you happen to have a backup of the old settings?
If so, can you send them + the changed settings (with proper names or in separate folders describing which is which) to my mail address?

giig1967g, 2020-02-18:
> Debatable for sure. Regardless, I suspect there are other users with "unexplained" reboots that would benefit by setting their _Connection Failure Threshold_ to zero. This should be mentioned in @Grovkillen's underutilized sticky: https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=5818

Agreed. 
I think that this parameter was created to resolve the issue of the units that were not responding to wifi but that were working "locally" (see #1640). A sort of wifi connection freeze, not recoverable without a physical reboot.
But the problem was a wifi connection issue not a MQTT broker issue. The difference is substantial.
I don't want my units to reboot if the MQTT broker is down, but I might want to reboot if the wifi is not reacheable, but without entering a reboot loop... 
I believe that we should think of an incremental mechanism that affects only wifi connectivity and not mqtt broker.

These are my 2 cents of course.

TD-er, 2020-02-18:
> I believe that we should think of an incremental mechanism that affects only wifi connectivity and not mqtt broker.

Well as Thomas also suggested, make it a setting per controller.
This also does remove controller outages from the "global" connection count.

thomastech, 2020-02-18:
> If so, can you send them + the changed settings (with proper names or in separate folders describing which is which) to my mail address?

Too late for that; I've already loaded the working ESP_Easy_mega-20200204-28-PR_2874_test_ESP8266_4M1M_VCC.bin and restored the missing config data. 

But the screenshot below shows the Nextion Device page. After loading ESP_Easy_mega-20200204-28-PR_2874_test_beta_ESP8266_4M1M.bin I found that "Line 1" in _Nextion Command Statements (Optional)_ area had been erased. I've since checked and if this line is missing the web interface still works fine on working firmware. 

[image]

> I believe that we should think of an incremental mechanism that affects only wifi connectivity and not mqtt broker.

I recall from past forum discussions that if the WiFi STA becomes disconnected, and reconnection fails after several attempts, the WiFi AP mode is turned back on. So if that is how it still works then you should be able to create some rules that causes a reboot when WiFi STA is offline for an extended time.

- Thomas

TD-er, 2020-02-18:
> But the screenshot below shows the Nextion Device page. After loading ESP_Easy_mega-20200204-28-PR_2874_test_beta_ESP8266_4M1M.bin I found that "Line 1" in _Nextion Command Statements (Optional)_ [...]

It may be a bit off topic, but I am also quite interested in what's going on here as it sounds like the firmware is looking at the wrong place to find the settings for that plugin.
So that's why I was interested in the settings file, to see if there is an offset in the data stored for this plugin.

I've had one other report of something similar, where MQTT user credentials suddenly appeared to be missing in some build, but were still working somehow.
After saving the settings again, they were lost.

So I'm afraid we (again) have some build issue, where one build may work just fine and a next one has all kinds of issues.

thomastech, 2020-02-18:
> So that's why I was interested in the settings file, to see if there is an offset in the data stored for this plugin.

I've reflashed with the suspect test_beta firmware and the page timeouts have reappeared. I'm unable to review the Nextion plugin to confirm a repeat of the data loss. However, I was able to save the broken settings file as well as a working version.

These files have my personal credentials in them. Please contact me by PM (letscontrolit forum user thomasb) and supply your private email address so I can safely forward the files to you.

- Thomas

TD-er, 2020-02-18:
OK, so it looks like "view" issue then.
For future reference, as I noticed my mail was not showing on my [profile page](https://github.com/TD-er) although I do get a number of emails from people seen me post on Github, so I thought it was open to the public.
The name is gijs noorlander and I think you can throw in a dot in between and gmail.com after it ;)
Or just anything at td-er.nl will also do.