letscontrolit/ESPEasy #2894: Can't use Publish command in rules

tsesakamoto, 2020-02-20:
When using mega-20191003
Publish mqqt doesn't work anymore in Rules.
When I submit this simple command: Publish domoticz/in,{"idx":8,"nvalue":0}
I get this message "Too many arguments, try using quotes!Command unknown: "Publish domoticz/in,{"idx":8,"nvalue":0}"

When using mega-20191003
And submit exactly the same command I get the result "OK'" and it is working.
Also it works flawlessly in Rules

Do I something wrong?

giig1967g, 2020-02-20:
try this:
Publish domoticz/in,'{"idx":8,"nvalue":0}'

TD-er, 2020-02-20:
There is also a setting to make parsing more relaxed:
[image]

See also one of the 2 sticky issues here: https://github.com/letscontrolit/ESPEasy/issues/2724

tsesakamoto, 2020-02-20:
Thanks that works.
Keep up the good work!

closed 2020-02-20 by TD-er