letscontrolit/ESPEasy #2896: After loading mega-20200222 it is not possible to update the rules

2bontb, 2020-02-23:
After loading mega-20200222 (minimal core 242 8285 1M OTA) it is not possible to update the rules. I did try this load on 1 Sonoff Basic and 1 Sonoff T1

When updating rules they will be corrupted:
e.g.
on ButtonTest#button do
 if [PowerTest#relay]=1
  gpio,12,0
 else
  gpio,12,1
 endif
endon

Becomes:
 �uttonTest#button do
 if [PowerTest#relay]=1
  gpio,12,0
 else
  gpio,12,1
 endif
endon

I tried this on a Sonoff Basic and a Sonoff T1.

Loading the rules from a file is possible. So this is how I restored the rules.

TD-er, 2020-02-23:
Oh not again....
Let me explain what is happening here (probably).
Some core versions suffer more from this than others.

The problem is that I don't have a way to determine whether a POST request is still in tact when processing it.
The ESP probably had another incoming HTTP request when you saved it.

Core 2.4.2 is rather old, but I know there are not many newer builds capable of 2-step OTA on 1M units.

I think the only thing you can do right now is indeed upload changed rules files.
Perhaps the only way we can probably proper fix it is by including JavaScript in the frontend that will generate a file upload instead of a POST request for the (rather large) rules.

2bontb, 2020-02-23:
TD-er, thanks for the explenation. 
It is no problem using the upload. As long as you know that this is the current way of working and if this "bug" is not screwing up other settings.
And I like to keep my Basics, T1, S20/S26 updateable via OTA. Else you always have to unscrew/open them.
Although, why should you update them if they are not broken. 
Currently I do this only when there are Wifi Improvements to get my connections more stable, I am struggling with stability in my house with many (~30)  wifi clients. But the latests releases do well!!!

TD-er, 2020-02-23:
> As long as you know that this is the current way of working and if this "bug" is not screwing up other settings.

Well it was messing also with other POST settings of for example task configs.
So there I added another (hidden) POST parameter which is added by JavaScript, to check if the POST was received complete. If not, the user will be notified and the settings should not be corrupted.

And good to know the WiFi stability is considered to be good these days.
With lots of clients, you may want to consider using multiple access points as sometimes these also run into issues when handling a few dozen clients.

PR #2921 (2020-03-23): Rules save fix
Now the rules are uploaded as a (virtual) file instead of form data. The previous way resulted in corrupted rules files if the unit didn't get the full POST body.

Should fix #2896
Should fix: #2819
Should fix: #2729
Should fix: #1296
closed 2020-03-23 by TD-er