letscontrolit/ESPEasy #2897: ESP32 Reboot on selecting GPIO on DS18b20 plugin

red55der, 2020-02-23:
Build en upload release mega-20200222. Boots fine.
When i add "Environment - DS18b20" --> selecting a GPIO (23) --> Submit, the ESP32 crashes.

12187 : Info  :  Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 1 3: 'TDN' length: 0 4: 'taskdevicepin1' length: 2 5: 'TDT' length: 1 6: 'TDVN1' length: 11 7: 'TDF1' length: 0 8: 'TDVD1' length: 1 9: 'edit' length: 1 10: 'page' length: 1
12198 : Info  : SaveToFile: free stack: 6920
12314 : Info  : FILE : Saved /config.dat
12316 : Info  : SaveToFile: free stack after: 6920
12400 : Info  : SaveToFile: free stack: 7192Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008c99a  PS      : 0x00060834  A0      : 0x8008bb57  A1      : 0x3ffb1a80
A2      : 0x3ffafec8  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060823  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060823  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x4008c99a:0x3ffb1a80 0x4008bb54:0x3ffb1aa0 0x40089daf:0x3ffb1ac0 0x4017c879:0x3ffb1b00 0x401a517a:0x3ffb1b20 0x4019a6a7:0x3ffb1b40 0x400d67f6:0x3ffb1b70 0x401081ae:0x3ffb1bf0 0x401083e7:0x3ffb1c20 0x401084ad:0x3ffb1c40 0x4010850a:0x3ffb1c70 0x4010bfc3:0x3ffb1c90 0x4010c283:0x3ffb1d00 0x4010c687:0x3ffb1d20 0x4010d36b:0x3ffb1d50 0x4011c8a1:0x3ffb1d90 0x401fbff9:0x3ffb1e30 0x400e0f52:0x3ffb1e50 0x400e0ffd:0x3ffb1e70 0x400e1076:0x3ffb1ea0 0x400e1203:0x3ffb1ef0 0x40150cf3:0x3ffb1f40 0x4016d95c:0x3ffb1f80 0x4017a065:0x3ffb1fb0 0x40089fa9:0x3ffb1fd0

Core 0 register dump:
PC      : 0x4008afde  PS      : 0x00060f34  A0      : 0x80089984  A1      : 0x3ffbbba0
A2      : 0x3ffafeec  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060f20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffbbba0
A10     : 0x00000003  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x3ffbc490  
A14     : 0x00000000  A15     : 0x3ffbaf20  SAR     : 0x00000010  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x4008afde:0x3ffbbba0 0x40089981:0x3ffbbbd0 0x4017c8fe:0x3ffbbc10 0x401a12e5:0x3ffbbc30 0x40090a2b:0x3ffbbc50 0x40089fa9:0x3ffbbc90

Rebooting...

TD-er, 2020-02-23:
And after the reboot, does it then work?
I've also seen strange reboots when tried to set some plugin using the serial port.
It shouldn't crash, but strangely it did work (with that plugin) after the reboot.

Just to know where to look for the issue, whether it is in the plugin itself or somewhere else.

red55der, 2020-02-23:
Nope. The submitted GPIO (13) don't get saved. 
If i set the GPIO pin again after reboot, it crashes directly.

Use a Lolin32 v1.0.0 board

TD-er, 2020-02-23:
Hmm not sure why, but for some reason is GPIO-13 not included in the [pins to use on ESP32](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32)
Can you try any of the noted input/output pins?

red55der, 2020-02-23:
GPIO14 and GPIO27 same result. Got it working. Placed a external pullup resistor. The internal pullups are to weak.

TD-er, 2020-02-23:
OK good to know, but still not so nice to have it crash on "not found" sensors.

aleksey-ru52, 2020-02-28:
double trouble? :-)
https://github.com/letscontrolit/ESPEasy/issues/1833

tonhuisman, 2023-09-02:
This has been resolved. Can be closed.

closed 2023-09-02 by TD-er