letscontrolit/ESPEasy #2901: OLEDFRAMEDCMD no longer working...  mega-20200222

mattlward, 2020-02-24:
### I have...
I have searched and have not found anything useful.  This command set was working in mega-20190301 and is not working in mega-20200222.  Both systems are running the dev 4M binary for SHT30 support.
- [ ] entered the full filename of the used version 
ESP_Easy_mega_20200222_dev_ESP8266)4M1M.bin
- [ ] given a list of active plugins or controllers when applicable.
Using this device: Display - OLED SSD1306/SH1106 Framed

### Steps already tried...
- [ x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x ] Tested on other node to make sure hardware isn't defective.
- [ x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

Under this version the log result indicates:
1614512 : Info  : SW  : GPIO=0 State=0 Output value=0
1614514 : Info  : EVENT: Switch#SwitchA=0.00
1614523 : Info  : ACT  : Oledframedcmd,off
1614524 : Info  : Command: Oledframedcmd
1614533 : Info  : Command unknown: "Oledframedcmd,off"

### Expected behavior
I did not expect to see Command unknown.

### Steps to reproduce

1. Install OLED module and configure switch entry
2. configure rules
3. press the A switch button on the display module

Yes

### System configuration

Hardware: Wemos D1, Wemos 64x48 2 button display

### Rules or log data

```
On Switch#SwitchA Do
  If [Switch#SwitchA]=1
  Oledframedcmd,low
 Else
  Oledframedcmd,off
 Endif
Endon
```

TD-er, 2020-02-24:
Please test with adding the sub command `display`
For example:
`Oledframedcmd,display,low`

mattlward, 2020-02-25:
That did it... I could not find that documented... did I just look in the wrong place?

TD-er, 2020-02-25:
[Documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P036.html#p036-page) is indeed absolutely not up to date.
I will make it as I'm already doing a lot of docs updates today...

TD-er, 2020-02-25:
It is now documented :)

PR #2904 (2020-02-25): [Commands] Update docs + add task/controller enable/disable commands
Also removed ProtocolIndex from the event struct as it is redundant information which can easily be forgotten to set.

Sorted the described commands in the documentation and added a few missing commands.

Also documented Cron plugin and Framed OLED plugin.

Fixes: #2901
closed 2020-02-25 by TD-er