letscontrolit/ESPEasy #2903: Feature request: M5StickC

s0170071, 2020-02-25:
Hi all, 

has anyone of you tried to run ESPEasy on a M5StickC or M5Stack? 
[https://www.youtube.com/results?search_query=m5stickc](https://www.youtube.com/results?search_query=m5stickc)

TD-er, 2020-02-25:
I've tested ESPEasy on M5Stack for over a year ago.
The display and some of the sensors included in them are not yet supported and I found the M5stack rather (hardware) unstable.
When not touching it, you can run it for a while, but the power supply unit in it is extremely unstable.

I needed to change quite a bit on it, but now the boards are well defined in the Arduino libs, so I guess it will be a lot less hassle to support them.

tonhuisman, 2022-04-17:
The TFT display ST7796 is supported by P116 via PR #3761, soon to be available in a release version of ESPEasy. Not all other sensors and devices available in M5Stack/M5Stick are supported by ESPEasy though.

tonhuisman, 2023-09-09:
This seems to be (mostly) solved, and the rest probably no longer relevant, so can be closed.

closed 2023-09-09 by TD-er