letscontrolit/ESPEasy #2914: remove "= valure" MQTT

bobdivx, 2020-03-02:
Hello I use esp easy mega with a Sonic probe.

I want to send a post to my mqtt broker:
{ "device_feature_external_id": "mqtt:fioul:1:level", "state": [fioul#Distance] }

the problem is that at the end of the message EASY automatically add the value at the end

{ "device_feature_external_id": "mqtt:fioul:1:level", "state": [fioul#Distance] } **= "valure"**

Thanks for your help

TD-er, 2020-03-02:
How do you post this?
Please give the exact way how you do this.

Is it a controller (if so, which one), or do you call publish from a rule, or something else?

bobdivx, 2020-03-02:
Hi,
broker on debian
I want to send the value in json format.

{ "device_feature_external_id": "mqtt:fioul:1:level", "state": [fioul#Distance] }

the Broker receives the message well but in the meantime espeasy is adding all alone = @@@

@@@=[fioul#Distance]
[fioul#Distance]= exemple 1234L

the broker receives
{ "device_feature_external_id": "mqtt:fioul:1:level", "state": 1234,20 } = 1234,20

[image]
[image]
[image]
[image]

TD-er, 2020-03-02:
Hmm that looks strange, a French ESPEasy :)  (Google translate I presume?)

Just to be sure, are you using "Generic HTTP" or "Generic HTTP Advanced" as controller?

bobdivx, 2020-03-18:
J'ai essayé même problème 😉

8none1, 2020-06-29:
I think this issue might be a duplicate of this: #2892

tonhuisman, 2022-05-07:
Not sure if this is still relevant, so can probably be closed

TD-er, 2022-05-07:
I don't think it is still relevant indeed, so will close it.
Please let me know if it is still relevant and if so, how to reproduce it.

closed 2022-05-07 by TD-er