letscontrolit/ESPEasy #2918: Append unit number to hostname doesn't work

dkadioglu, 2020-03-04:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

After flashing Release mega-2020-03-01 the function to "Append Unit Number to hostname:" doesn't work. With Release mega-20191127 this was working.

### Expected behavior

The unit number should be appended to the hostname, if set.

### Actual behavior

The unit number is not appended to the hostname, although set.

 yes

System:
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20200301
Plugin Count:⋄
46 [Normal]
Build Time:⋄
Mar  1 2020 19:12:53
Binary Filename:⋄
ESP_Easy_mega-20200301_normal_ESP8266_1M.bin

```

TD-er, 2020-03-04:
The hostname is used in several places, like:
- mDNS hostname
- MQTT client
- %sysname%

And some other places.
Which one specific do you mean?
And I will make it uniform for all places (I can find)

TD-er, 2020-03-04:
Hmm I am looking through the code and am not entirely sure what to do with it for controllers.
It looks like the hostname is also used in controllers, which are quite protocol specific.
This suggests the "append unit number to hostname" should also be a controller specific setting.

This then also implies the replace of `%sysname%` will not always be the same.

dkadioglu, 2020-03-04:
In my case I am using FHEM. I used a couple of releases and for me until mega-20191127 the FHEM controller seemed to use the hostname with unit number appended. Thank you for your investigation!

commit f251b6ad (2020-03-04): Fix Append unit number to hostname (#2918)

Fixes: #2918

Known issue:
Currently we don't use this in controllers, so we must add that as a setting for controllers.
closed 2020-03-04 by TD-er