letscontrolit/ESPEasy #2927: Bad formatted json returned in HTTP request response

MkLHX, 2020-03-06:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Hi,
I'm using ESPEasy on SonOff S20 EU flashed by using ESPEasyFlasher.
I have strange behavior when using command through HTTP protocol. I readed this doc: https://www.letscontrolit.com/wiki/index.php?title=ESPEasy_Command_Reference
And try to comand GPIO to 1 and 0 and getting his status.
the device GPIO is good setted to 0 but the json request response is bad formatted.

### Expected behavior

Expected json look like:
```
{
"log": "",
"plugin": 1,
"pin": 12,
"mode": "output",
"state": 0
}
```

### Actual behavior

Current json look like:
```
{
"log": "",
"plugin": 1,
"pin": 12,
"mode": "output",
"state": 0
}

Ok
```
### Steps to reproduce

1. when i hit http://esp_easy_address/control?cmd=GPIO,12,0
2. when i hit http://esp_easy_address/control?cmd=GPIO,12,1
3. when i hit http://esp_easy_address/control?cmd=status,GPIO,12

### System configuration

Hardware:
[image]

ESP Easy version: 

ESP Easy settings/screenshots: 
[image]
[image]

clipboard copied:
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core bc204a9b, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20191208
Plugins:⋄
46 [Normal]
Build Time:⋄
Dec  8 2019 18:01:48
Binary Filename:⋄
ESP_Easy_mega-20191208_normal_ESP8266_1M.bin

TD-er, 2020-03-06:
Looks like a duplicate of this one: https://github.com/letscontrolit/ESPEasy/issues/2892

MkLHX, 2020-03-06:
Ok so if i understand the #2892 this is not currently fixed but soon in Pre-release it will be?
Second question, the response will be a json string?
Can i do a pull request if i find where to correct it?

TD-er, 2020-03-06:
You can test this PR: https://github.com/letscontrolit/ESPEasy/pull/2778
The changes in the switch plugin are not yet OK, so that's why it has not been merged yet.

MkLHX, 2020-03-15:
Ok i’ll do the with a workaround.
Wait for new release.

closed 2020-03-15 by MkLHX