letscontrolit/ESPEasy #2928: All positions on a line behind character(s) written to LCD2004 are cleared using rules

urkelbundy, 2020-03-06:
### Checklist
- [X ] The title of this issue is "to the point" and descriptive.
- [X ] This issue describes what is happening.
- [X ] This issue describes what components are affected (e.g. name of plugin/controller)
- [X ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [X ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X ] Tested previous/other build (mention which one already tested)
- [X ] Tested on other node to make sure hardware isn't defective.
- [X ] Verified if the problem is limited to a single plugin/controller

###  Summarize of the problem/feature request
If I write a character to a line on a LCD2004 display the line behind the character written is cleared until the end of the line.

I don't use the 4 lines in the webinterface of the LCD2004 plugin but instead I use rules

### Expected behavior
Only the the positions that have characters written to it using rules should be changed and the characters behind until the end of the line should remain unaffected

On a LCD2004 display one line has 20 characters (positions)
E.g. line 1 shows:
12345678901234567890

When an A is written to e.g. position 5 on line 1 using rules "lcd,1,5,A" the should be:
1234A678901234567890

### Actual behavior
E.g. line 1 shows:
12345678901234567890

When an A is written to e.g. position 5 on line 1 using rules "lcd,1,5,A" the line becomes:
1234A
The rest of the line is cleared

As a workaround I rewrite the whole line everytime something changes anywhere on the line but this is relatively slow and you see the line being cleared first. 

### Steps to reproduce
Do what is explained above

The issue remains after a reboot

### System configuration
NodeMCU V3

ESP Easy version: ESP_Easy_mega-20200222_normal_ESP8266_4M1M.bin
I do not know when this started but I have seen this behaviour for a long time using various builds.
I only use prebuild binaries and do not compile myself.

Furthermore the %SP% variable should produce a space but seems to be ignored when used.
Also a space at the beginning of characters written is also ignored.
E.g. lcd,1,1,"Off" is identical to lcd,1,1," Off"
Watch the space before Off. The result is that the O of Off is always at position 1 using both commands above. It doesn't matter how many leading spaces are used, they are all ignored and the rest of the line is cleared.

TD-er, 2020-03-06:
Looks like the text is 'trimmed' before it is sent.

I don't think this is a recent change, as that plugin has not been changed a lot in the last 2 years.
The parsing of arguments in rules has changed, but I doubt that may cause the line to be cleared.

Without looking at the code, I think it somehow makes sense to clear the characters to the right of what is being printed.
For example most displays will list values like this:
```
Temp: 12.34
Hum: 45.67%
```
etc.
If values change in length, you want to make sure the last value isn't partly shown, so clearing them does make sense.

Not saying it is the best behavior, but I do understand why someone would chose to do so.

urkelbundy, 2020-03-06:
I also don't think that this is due to a recent change.

There are 20 positions in a line so it is reasonable to have on 1 line e.g.:
T: 12.34C H: 45.67%
When there is a new temperature value you cannot update only the temperature because that will clear the remainder of the line.

Or in my case for a heating status:
 Off | Nrml | Closed 
becomes
Auto | Nrml | Open

When Off changes into Auto I have to rewrite the rest of the line as well.
When Closed changes to Open, I only have to update the last part of the line.

Also I cannot start the line with a space e.g. for the word Off is ignored.

It seems to me that the LCD2004 plugin removes spaces at the front of characters being written and add spaces at the end of characters being written.
This doesn't sound right, does it?
If you have the ability to write on a specific part of the lcd using the rules command lcd,y,x,text then it doesn't make sence to me that you need the rewrite the line after x+length_of_text
If you want to clear the remainder of the line, do it yourself or maybe use an extra optional parameter for the command lcd,y,x,text,true
The last parameter "true" is for clearing the line.

Using rules I look at the temperature value to see how many characters needs to be written and change the starting position accordingly. If check for these temperatures:
<=10
>-10 and <0
>=0  and <10
>=10

It would be nicer though to have some more of the formatting rules as in sprintf as there are now using rules.
E.g. space padded on the left instead of padding with zeros and right/left justified text

uzi18, 2020-03-06:
@urkelbundy just check docs: https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#formatting-refered-values

urkelbundy, 2020-03-06:
I ave read those and using them.

Here are my rules for a heater/thermostat status on 1 line. I change the whole line at once using 3 statusses sent from domoticz using an thermostat event command that sends 3 eventvalues
on thermostat do
  let,4,%eventvalue1%
  let,5,%eventvalue2%
  let,6,%eventvalue3%

  if [VAR#4]=0 and [VAR#5]=10 and [VAR#6]=0
    lcd,4,1," Off | Nrml | Closed"
  endif

  if [VAR#4]=0 and [VAR#5]=20 and [VAR#6]=0
    lcd,4,1," Off | Eco  | Closed"
  endif

  if [VAR#4]=10 and [VAR#5]=10 and [VAR#6]=0
    lcd,4,1,"Auto | Nrml | Closed"
  endif

  if [VAR#4]=10 and [VAR#5]=10 and [VAR#6]=1
    lcd,4,1,"Auto | Nrml |  Open "
  endif

  if [VAR#4]=10 and [VAR#5]=20 and [VAR#6]=0
    lcd,4,1,"Auto | Eco  | Closed"
  endif

  if [VAR#4]=10 and [VAR#5]=20 and [VAR#6]=1
    lcd,4,1,"Auto | Eco  |  Open "
  endif

  if [VAR#4]=20 and [VAR#5]=10 and [VAR#6]=0
    lcd,4,1,"Force| Nrml | Closed"
  endif

  if [VAR#4]=20 and [VAR#5]=20 and [VAR#6]=0
    lcd,4,1,"Force| Eco  | Closed"
  endif

  if [VAR#4]=20 and [VAR#5]=10 and [VAR#6]=1
    lcd,4,1,"Force| Nrml | Open  "
  endif

  if [VAR#4]=20 and [VAR#5]=20 and [VAR#6]=1
    lcd,4,1,"Force| Eco  | Open  "
  endif
endon

On a 2nd line of the lcd display inside and outside temperatures are displayed that are also send from domoticz using an insideTemp and outsideTemp event command.

on insideTemp do
  let,1,%eventvalue%

  if [VAR#1]>=0 and [VAR#1]<10
    lcd,2,1,"I:  [var#1#D1.1]{D}C"
  endif

  if [VAR#1]>=10
    lcd,2,1,"I: [var#1#D1.1]{D}C"
  endif

  event,displayOutsideTemp
endon

on outsideTemp do
  let,2,%eventvalue%

  event,displayOutsideTemp
endon

on displayOutsideTemp do
  if [VAR#2]>=0 and [VAR#2]<10
    lcd,2,12,"O:  [var#2#D1.1]{D}C"
  endif

  if [VAR#2]>=10
    lcd,2,12,"O: [var#2#D1.1]{D}C"
  endif

  if [VAR#2]<0 and [VAR#2]>-10
    lcd,2,12,"O: [var#2#D1.1]{D}C"
  endif

  if [VAR#2]<=-10
    lcd,2,12,"O:[var#2#D1.1]{D}C"
  endif
endon

urkelbundy, 2020-03-06:
Something strange happens pasting my rules above.
All the lcd,2,12 lines seem identical but they should not be.
I vary the amount of spaces between O: and the temperature value so that the O: and the degree character are always on the same place on the screen and in between is the temperature.
The same applies to the I: temperatures rules above.

Why the spaces in the rules above have changed, I do not know.

I will try to make a picture of the LCD screen so that you understand what is on the display.

urkelbundy, 2020-03-06:
[image]

This is a picture not using Easy ESP but using other software.
As you can see, the word Off has one space in front. I was not able to do this with Easy ESP.

I: = insideTemperature
O: = outsideTemperature
S:= insideTemperature setpoint

PUI: = Open/Closed of a sliding door

Off/Auto = thermostat Off or Auto mode
Nrml/Eco = Normal or Economic thermostat setting
Open/Closed = the heater valve

urkelbundy, 2020-03-06:
@uzi18 
You reference a piece of code which says that it is zero filled.
That is not what I meant.

If I go to the Easy ESP tools menu and submit the lcd,1,1,A command on line 1 which is fully occupied with the date/time as shown in the screenshot above, the first position of that line shows the A, the rest of the line is cleared and that has nothing to do with zero filled.

Or did you respond on some other part I mentioned?

TD-er, 2020-03-06:
I edited the post with the rules by adding code block markers.
(3 times a backquote at begin and end on a new line)

urkelbundy, 2020-03-06:
It is this piece of code that clears the line in P012_LCD.ino:

          //clear line before writing new string
          if (Plugin_012_mode == 2){
              lcd->setCursor(colPos, rowPos);
              for (byte i = colPos; i < Plugin_012_cols; i++) {
                  lcd->print(" ");
              }
          }

This might be desirable in some cases but not in all, so an extra parameter for the lcd command would be nice

uzi18, 2020-03-07:
it was fast action for your quote 

> E.g. space padded on the left instead of padding with zeros and right/left justified text

will check your findings later, just thinking about some generic way of handling your situation

uzi18, 2020-03-07:
@urkelbundy here #2930

urkelbundy, 2020-03-08:
@uzi18 Padding with spaces works great. Saves a lot of rules. :-)

uzi18, 2020-03-08:
@urkelbundy so @TD-er will eventually merge this simple change, now need to look at your problem with cleaning rest of line

uzi18, 2020-03-08:
@urkelbundy did you tried to change setting "LCD command Mode" to truncate? 
What is your actual setting?

urkelbundy, 2020-03-09:
Yes the LCD command mode is truncate exceeding message.
I believe that is for when there are more characters to write than the line can hold.
This has nothing to do with the line being cleared or the unability to write leading spaces, or does it?
The latter is because the string to be written is trimmed.
I think I have found both causes and I assume you have found them too.

urkelbundy, 2020-03-11:
Let me rephrase my last comment.

I found the location in the source that causes leading spaces to be removed.
It is a trim command in StringConverter.ino in the function parseStringKeepCase.
I created an identical function parseStringKeepCaseLCD without the trim command and this works for leading spaces.
Using this function without the trim command enables characters to be written to the LCD screen with one or more spaces in front.

Still looking where the remainder of the line is cleared. It doesn't seem to be the source I listed earlier.

urkelbundy, 2020-03-11:
Found it in the LCD2004 plugin

There is this line: 
text = P012_parseTemplate(text, Plugin_012_cols);

This should be:
text = P012_parseTemplate(text, text.length());

In misc.ino padding takes place so that the line is padded with spaces.
  // padding spaces
  while (newString.length() < minimal_lineSize) {
    newString += ' ';

If someone wants the remainder of the line being cleared behind the last character written to the LCD display, then choose "Clear then truncate exceeding message" in the plugin.

My request to the devs, could you change the code of the LCD2004 plugin so that trimming does not occur anymore for the LCD2004 plugin as I explained in my previous message and the line that I describe above!?

TD-er, 2020-03-12:
I'm also working for another plugin to change how extra task settings are being saved, so that means I also have to check the display plugins to see if those will not be broken by it.
Then I will also have a look at these changes you suggested.

If I do not reply on this after Friday, please help me remember.

urkelbundy, 2020-03-14:
@TD-er 
This is your wake-up call, Friday has passed. ;-)

urkelbundy, 2020-03-21:
Is there any progress?
Many thanks.

TD-er, 2020-03-22:
I've been too busy with other things, but Github now has the feature of marking issues as unread, so that also helps me to track things to do.

TD-er, 2020-03-22:
[image]
I've looked at the code and we can make very simple changes:
- Change the padded length to the length of the string (which is useless if the text contains tokens to be replaced like `[task#var]`
- Add extra parameters to help formatting (row/col coordinates, min length)
- Write the texts via the write command from rules

The last option is already present.
The command `LCD,rownr,colnr,text` can be used to write texts on any position of the screen, but it does pad the string to append spaces to match the width of the screen. Not sure though what will happen if you write a string on the 1st line starting half-way the line. Will it overwrite the text on the next line?

So in short, do you only need the command changed (to be used in rules for example) or should the plugin config as shown in the screenshot also be extended?

urkelbundy, 2020-03-22:
The truncate exceeding message trancates lines longer that go passed the end of the line. This is my default setting but that doesn't solve the issues I reported.

The lcd command in rules is what I use mainly. I hardly use the 4 lines in the plugin config.

The static text and variables that I display on the LCD screen are carefully positioned and also the variable transformations are chosen so that no line goes beyond the last position of that line.

Currently I am using a self compiled build that has the changes I suggested earlier:

```
> Found it in the LCD2004 plugin
> 
> There is this line:
> text = P012_parseTemplate(text, Plugin_012_cols);
> 
> This should be:
> text = P012_parseTemplate(text, text.length());
> 
> In misc.ino padding takes place so that the line is padded with spaces.
> // padding spaces
> while (newString.length() < minimal_lineSize) {
> newString += ' ';
> }`
```
and

> It is a trim command in StringConverter.ino in the function parseStringKeepCase.
I created an identical function parseStringKeepCaseLCD without the trim command and this works for leading spaces.

These 3 small changes in _P012_LCD.ino, misc.ino and StringConverter.ino solve the issues I encountered. As far as I can see no other functionality is broken by these changes.
Any single character on one of the 4 lines can be changed without affecting other characters still present on the display with these 3 small changes to the code. Isn't this how it should be instead of the whole line being cleared for any number of characters being written to the display?!

If someone wants to clear the line before something is written to a line then the plugin config "Clear then truncate exceeding message" can be used. This option is still available after my suggested code changes.

TD-er, 2020-03-22:
> This should be:
> text = P012_parseTemplate(text, text.length());

Like I said, that's also not a fix and may result in behavior that's counter-intuitive.  (e.g. `[taskname#varname]` is often longer than the string replaced by it, like `12.34`)
So the best is to have the length value set to 0.  (no padding spaces)
But it will break existing installs.
So I think it is the best to add a new value to the combobox, to only use trimmed values and not append spaces.
If you need the chars after it be cleared, then you have to do it yourself.

urkelbundy, 2020-03-22:
> If you need the chars after it be cleared, then you have to do it yourself.

No I do not want characters to be cleared and if I want that it is fine that I do that myself.

It is not clear to me what you propose to change but if the result is the same as I need than it is fine.
Bear in mind that there were 3 issues:
- the whole line was cleared before characters written
- the remainder of the line was cleared behind characters written
- all leading spaces in front of characters to be written to the line are removed

Many thanks for your efforts to fix these issues.

tonhuisman, 2023-09-02:
This has been fixed quite some ago, so can be closed.

closed 2023-09-02 by TD-er