letscontrolit/ESPEasy #2931: flash sonoff basic r2 after connect to wifi net, is impossible to see online

megamarco833, 2020-03-07:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
i have some sonoff basic R2 V1 and i try to flash with new version of espeasy.
i used DOUT
and i succeeed. i see the ESP_Easy net, i connect to it ad i give my password and SSP of internal net.
when ESP reboot it's impossible to ping and open the sonoff on the internal net and ESP_Easy net desappear.
I try to use fw:
ESP_Easy_mega-20200204_normal_ESP8266_1M_VCC
ESP_Easy_mega-20200204_normal_ESP8266_1M

all same issue: after connecting ESP_Easy net and give internal SSP and password, is impossible to see on internal net the sonoff and ESP_Easy net desapper
the only way of became it working is using old version of EspEasy where there was PUYAversion:
ESP_Easy_mega-20181023_dev_ESP8266PUYA_1024_VCC.bin
in this case no problems

how it can be solved?
thanks

thomastech, 2020-03-10:
I installed ESP_Easy_mega-20200204_normal_ESP8266_1M on a couple Sonoff Basic R2's without issue. So more information would be helpful.

Flash again and connect using AP mode. After you enter the credentials, and before the reboot, go to the _Tools=>Advanced=>Serial Log Level_ and set the serial log to debug. Submit (save).

The serial log will provide messages that may explain the problem. So launch your favorite a serial terminal, reboot, and review all the messages from ESPEasy. 

BTW, whenever problems occur from a firmware update it is wise to flash with the _blank_1MB.bin_ file to clear out the Flash memory (removes old settings). Then flash the Mega Release.

- Thomas

TD-er, 2020-03-10:
The last nightly build has the `board_build.flash_mode` set to dout, where in some builds it may still have been set to dio.

There is a difference between flashing mode (dout/dio/etc) and the build settings.
Flashing mode is used to actually write the sketch and maybe also switching some flag in the bootloader (depending on the used flash tool).
The build flag is used at run time to determine the possible flash mode.

Do you see the node connecting to the WiFi?
Does the ESP react to network activity after a few minutes?

megamarco833, 2020-03-10:
> 
> 
> I installed ESP_Easy_mega-20200222_normal_ESP8285_1M.bin on a couple Sonoff Basic R2's without issue. So if this bin does not work for you then more information would be helpful.
> 
> Flash again and connect using AP mode. After you enter the credentials, and before the reboot, go to the _Tools=>Advanced=>Serial Log Level_ and set the serial log to debug. Submit (save).
> 
> The serial log will provide messages that may explain the problem. So launch your favorite serial terminal, reboot, and review all the messages from ESPEasy.
> 
> BTW, whenever problems occur from a firmware update it is wise to flash with the _blank_1MB.bin_ file to clear out the Flash memory (removes old settings). Then flash the Mega Release.
> 
>     * Thomas

hi, sure i always perfomr a blank_1MB.bin flash to eresa esp before flashing any fw
i used as tool ESP8266Flasher.exe under windows with DOUT and 1mb falsh size, budrate 115200
[image]

my sonoff is a sonoff basic and on pcb si written R2 v1.0 2017-10-11
i also a sonoff basic where is written on pcb sonoff TH (only one, the other three have on pcb the write R2)
al all sonoff is written on esp chip esp8266X
why you use as firmware: XXX_normall_ESP8285_1M.bin ? and not XXX_normall_ESP8266_1M.bin ?

anyway with  XXX_normall_ESP8266_1M.bin the issue was that after flash, when i connect using AP mode to my internal newtwork,, after reboot i'm not able to see the sonoff on my network. 
and when i'm in this situation the only solution is to flash again a new fw otherwise it's stuck because i can't find on my network and ESP_0 net desappear.
the only way to have sonoff visible on my network is only if i use the very old fw:
ESP_Easy_mega-20181023_dev_ESP8266PUYA_1024_VCC.bin

now i flash again with
Easy_mega-20200204_normal_ESP8266_4M1M.bin
but the problem is that when i enter on AP (http://192.168.4.1) and set my network, i can only click apply and then it reboot. so no way to activated the debug mode: Tools=>Advanced=>Serial Log Level
as i explain after the reboot i can not see on my internal network the new IP DHCP of the sonoff so i can't open web page Tools=>Advanced=>Serial Log Level

megamarco833, 2020-03-10:
> Do you see the node connecting to the WiFi?
no, unfortunatly, after connecting to AP mode: 192.168.4.1 selecting my network SSID and enterin the password, the sonoff reboots and i can't see on my network
> Does the ESP react to network activity after a few minutes?
no, i wait many minutes but i can't see it inside my network

megamarco833, 2020-03-10:
i have also try to flash:
ESP_Easy_mega-20200204_normal_ESP8285_1M.bin

in this case, after flashin appear ESP_Easy_0 network, i connect to that AP , i give credentianl of my network, it show the countdown, but after that sonoff remain connected at ESP_Easy_0 
if i reboot sonoff the ESP_Easy_0 net become visible again, i can connect to that AP again but if i try to load 192.168.4.1 at this time it can not load the webpage and i receive a connection error.

latest attempts was done using:
ESP_Easy_mega-20200310_normal_ESP8266_1M.bin
and
ESP_Easy_mega-20200310_normal_ESP8285_1M.bin

nothin difference on behaviour except name of SSID that is now become ESP_Easy

thomastech, 2020-03-10:
@megamarco833:
> why you use as firmware: XXX_normall_ESP8285_1M.bin ? and not XXX_normall_ESP8266_1M.bin ?

I made a bad assumption on which ESP chip was in your Sonoff Basic. Currently sold Sonoff Basics use ESP8285 (internal Flash). Older versions used ESP8266 (external Flash chip). 

> So no way to activated the debug mode: Tools=>Advanced=>Serial Log Level
> as i explain after the reboot

Unfortunately it does not appear that you can enable the log using serial commands. As a workaround, try this: reload the working firmware, set the log level to debug, save, and confirm the log is running. Then load the new firmware (without blank bin file). With a bit of luck the old Tool settings will still exists and provide a working serial log.

#TD-er,
The log would tell us a lot about what is going on. Especially if this problem is due to the WiFi core library bug I ran into last week. Is there a "_ALT-WIFI_" version in the latest mega release that the OP can try?

megamarco833, 2020-03-10:
i try also this: flash ESP_Easy_mega-20181023_dev_ESP8266PUYA_1024_VCC.bin
activate serial log as debug then flasg new fw: ESP_Easy_mega-20200204_normal_ESP8266_1M.bin
then i connect to AP ESP_Easy_0 but i can not see any log :(

TD-er, 2020-03-10:
Not sure if we have them for 1M builds.
These are the current settings for the used SDK 2.2.x builds:
https://github.com/letscontrolit/ESPEasy/blob/4e7cb1e32de521f9ae8ee1fabefa27c3790d107c/platformio_core_defs.ini#L113-L132

These "new" settings are now the default since you last tested them:
https://github.com/letscontrolit/ESPEasy/blob/cde4c93ef502005c4b9aeb4fb69e26461dfeae44/platformio_esp82xx_base.ini#L7-L11
So if the regular builds do not work and builds of about a week or a month ago also do not work, then we have another issue.

Also the settings do not seem to be saved, so I guess it has nothing to do with the WiFi but more with the way how flash is being saved.
The PUYA flash code should be in the core lib and active.

thomastech, 2020-03-10:
@megamarco833 
> then i connect to AP ESP_Easy_0 but i can not see any log :(

You won't see the serial log from the browser. You need to use a serial terminal communication program (puTTY, Arduino Terminal, etc.) and connect the serial port via USB.

First confirm the serial log is working when the "good" bin (ESP_Easy_mega-20181023_dev_ESP8266PUYA_1024_VCC) is running. After you see it working you can load the new firmware.

Also, the Expressif startup messages will appear on the serial port at reboot even with serial log disabled. Post the messages that appear. They will confirm that boot was successful.

With your serial terminal connected, enter this command in the terminal window:
**_settings_**
then press [enter].

It will report some useful setting info, like this example:
System Info
  IP Address    : 192.168.1.1
  Build         : 20104
  Name          : ESP_Easy_0
  Unit          : 0
  WifiSSID      : YOUR SSID HERE
  WifiKey       : YOUR WIFI PASSWORD HERE
  WifiSSID2     :
  WifiKey2      :
  Free mem      : 15592

@TD-er:
> Also the settings do not seem to be saved, so I guess it has nothing to do with the WiFi but more with the way how flash is being saved.

Let's not give up on that yet and wait for more feedback. BTW, too bad that the various log levels cannot be set via serial terminal.

- Thomas

dwarf2k, 2020-03-10:
Keep in mind this is a Sonoff Basic R2 so runs at mains voltage. Hooking it to USB while plugged in can be dangerous.

thomastech, 2020-03-10:
Good reminder about mains voltage. 

The serial terminal uses the USB port just like the flashing tool. The 3.3V power will come from USB. 

Never connect Mains power when USB is connected. Otherwise prepare an updated will for the surviving family members.

- Thomas

megamarco833, 2020-03-10:
> 
> 
> Good reminder about mains voltage.
> 
> The serial terminal uses the USB port just like the flashing tool. The 3.3V power will come from USB.
> 
> Never connect Mains power when USB is connected. Otherwise prepare an updated will for the surviving family members.
> 
>     * Thomas

i well know, but better underline :-)

TD-er, 2020-03-10:
Better to underline than to undertake....

megamarco833, 2020-03-10:
@thomastech
> Also, the Expressif startup messages will appear on the serial port at reboot even with serial log disabled. Post the messages that appear. They will confirm that boot was successful.
>
> With your [...]

i used putty with this is the log of fw with PUYA

```
266145 : WIFI : Set WiFi to AP+STA
336328 : WD   : Uptime 6 ConnectFailures 0 FreeMem 6336
336331 : LoopStats: shortestLoop: 60 longestLoop: 8498634 avgLoopDuration: 92.59 loopCounterMax: 500000 loopCounterLast: 343425 countFindPluginId: 0
336332 : Load File stats: Count: 2 Avg/min/max 3237.00/2267/4207 usec
336332 : Plugin call 50 p/s   stats: Count: 15634 Avg/min/max 120.05/77/327 usec
336333 : Plugin call 10 p/s   stats: Count: 3133 Avg/min/max 99.24/82/275 usec
336333 : Plugin call 10 p/s U stats: Count: 3133 Avg/min/max 1358.69/874/2807 usec
336334 : Plugin call  1 p/s   stats: Count: 318 Avg/min/max 392.87/184/35543 usec
336334 : setNewTimerAt()      stats: Count: 19709 Avg/min/max 118.82/51/329 usec
336335 : timeDiff()           stats: Count: 13574573 - CPU cycles per call: 8.37
336335 : Scheduler stats: (called/tasks/max_length/idle%) 4377321/19690/6/95.20
363799 : LoopStats: shortestLoop: 60 longestLoop: 8498634 avgLoopDuration: 78.16 loopCounterMax: 500000 loopCounterLast: 342260 countFindPluginId: 0
363800 : Plugin call 50 p/s   stats: Count: 1363 Avg/min/max 137.63/94/315 usec
363801 : Plugin call 10 p/s   stats: Count: 273 Avg/min/max 112.51/98/282 usec
363801 : Plugin call 10 p/s U stats: Count: 273 Avg/min/max 1719.42/1034/2764 usec
363802 : Plugin call  1 p/s   stats: Count: 27 Avg/min/max 475.22/442/531 usec
363802 : setNewTimerAt()      stats: Count: 1938 Avg/min/max 130.92/100/296 usec
363803 : timeDiff()           stats: Count: 1071406 - CPU cycles per call: 8.50
363803 : Scheduler stats: (called/tasks/max_length/idle%) 342260/1938/6/93.70
393799 : LoopStats: shortestLoop: 60 longestLoop: 8498634 avgLoopDuration: 76.62 loopCounterMax: 500000 loopCounterLast: 381230 countFindPluginId: 0
393800 : Plugin call 50 p/s   stats: Count: 1500 Avg/min/max 134.11/98/311 usec
393801 : Plugin call 10 p/s   stats: Count: 300 Avg/min/max 116.71/98/268 usec
393801 : Plugin call 10 p/s U stats: Count: 300 Avg/min/max 1643.16/1040/2049 usec
393802 : Plugin call  1 p/s   stats: Count: 30 Avg/min/max 463.80/391/566 usec
393802 : setNewTimerAt()      stats: Count: 2132 Avg/min/max 130.07/101/239 usec
393802 : timeDiff()           stats: Count: 1192814 - CPU cycles per call: 8.45
393803 : Scheduler stats: (called/tasks/max_length/idle%) 381230/2132/6/94.60
423799 : LoopStats: shortestLoop: 60 longestLoop: 8498634 avgLoopDuration: 76.62 loopCounterMax: 500000 loopCounterLast: 381252 countFindPluginId: 0
423800 : Plugin call 50 p/s   stats: Count: 1500 Avg/min/max 133.71/98/311 usec
423800 : Plugin call 10 p/s   stats: Count: 300 Avg/min/max 116.91/98/300 usec
423801 : Plugin call 10 p/s U stats: Count: 300 Avg/min/max 1648.86/1050/2063 usec
423801 : Plugin call  1 p/s   stats: Count: 30 Avg/min/max 465.87/413/552 usec
423802 : setNewTimerAt()      stats: Count: 2132 Avg/min/max 130.00/101/238 usec
423802 : timeDiff()           stats: Count: 1192879 - CPU cycles per call: 8.46
423803 : Scheduler stats: (called/tasks/max_length/idle%) 381252/2132/6/94.60
```
i can't write "settings" inside the command propt because it's not allow to write nothing.
this is the parameter that i used:
[image]

what's the next steps?

thomastech, 2020-03-10:
> i used putty with this is the log of fw with PUYA

So far so good.

> i can't write "settings" inside the command propt because it's not allow to write nothing.
> this is the parameter that i used:

Your PuTTY settings seem valid, but I suggest setting Flow Control to None. 

Maybe the "settings" command did not exist in your old bin version? Here are some other commands to try out to help confirm that you can fully communicate with the device (each command must be followed by [Enter]).
ip
dns
wifiscan
reboot

If you do not have success then try the _Serial Monitor_ feature that is provided in the ESP.Easy.Flasher.exe program (included in the ESPEasy firmware zip bundle).

After you are able to communicate, load the new firmware (do not use the blank bin). 
Launch the serial terminal and confirm you have log messages. 
Type the reboot command and capture all the messages (starting with the warm reboot). 
Collect messages for a couple minutes. Then post the log, but obscure any private information.

- Thomas

megamarco833, 2020-03-10:
```

2283801 : LoopStats: shortestLoop: 60 longestLoop: 8498634 avgLoopDuration: 78.44 loopCounterMax: 500000 loopCounterLast: 372502 countFindPluginId: 0
2283802 : Plugin call 50 p/s   stats: Count: 1492 Avg/min/max 136.84/97/312 usec
2283802 : Plugin call 10 p/s   stats: Count: 299 Avg/min/max 112.09/98/287 usec
2283803 : Plugin call 10 p/s U stats: Count: 299 Avg/min/max 1670.76/1059/2764 usec
2283803 : Plugin call  1 p/s   stats: Count: 30 Avg/min/max 420.67/332/553 usec
2283804 : setNewTimerAt()      stats: Count: 2122 Avg/min/max 133.37/103/300 usec
2283804 : timeDiff()           stats: Count: 1166407 - CPU cycles per call: 8.46
2283805 : Scheduler stats: (called/tasks/max_length/idle%) 372502/2122/6/94.60
>ip

IP:192.168.0.26(DHCP)

Ok
2288653 : Command: ip
>dns

DNS:192.168.0.8(DHCP)

Ok
2312910 : WD   : Uptime 39 ConnectFailures 0 FreeMem 9144
>wifiscan
WIFI : SSID Scan start
WIFI : 4 networks found
WIFI : 1: NETGEAR 80:3F:5D:6E:37:67 Ch:13 (-62dBm) WPA2/PSK
WIFI : 2: VodafoneASUS 74:36:6D:AD:F8:54 Ch:13 (-79dBm) WPA2/PSK
WIFI : 3: Vodafone-WiFi 62:36:6D:AD:F8:55 Ch:13 (-77dBm) open
WIFI : 4: DIRECT-a5M2070 Series 86:25:19:95:E0:73 Ch:13 (-87dBm) WPA2/PSK

Ok
2312998 : Command: wifiscan
2314661 : LoopStats: shortestLoop: 60 longestLoop: 8498634 avgLoopDuration: 82.25 loopCounterMax: 500000 loopCounterLast: 365811 countFindPluginId: 0
2314662 : Plugin call 50 p/s   stats: Count: 1461 Avg/min/max 133.77/97/304 usec
2314663 : Plugin call 10 p/s   stats: Count: 293 Avg/min/max 120.09/98/289 usec
2314663 : Plugin call 10 p/s U stats: Count: 293 Avg/min/max 1629.55/1040/2570 usec
2314664 : Plugin call  1 p/s   stats: Count: 30 Avg/min/max 421.37/338/500 usec
2314664 : setNewTimerAt()      stats: Count: 2079 Avg/min/max 133.49/103/306 usec
2314665 : timeDiff()           stats: Count: 1145359 - CPU cycles per call: 8.50
2314665 : Scheduler stats: (called/tasks/max_length/idle%) 365811/2079/6/94.90
```

these are commands for ESP with puya
now i will flash new FW directly without blank

megamarco833, 2020-03-10:
```
>reboot

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
   ▒U138 :

INIT : Booting version: mega-20181008 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
139 : INIT : Warm boot #2 - Restart Reason: Software/System restart
141 : FS   : Mounting...
148 : FS   : Mount successful, used 75802 bytes of 113201
589 : CRC  : program checksum       ...OK
603 : CRC  : SecuritySettings CRC   ...OK
625 : INIT : Free RAM:16488
626 : INIT : I2C
627 : INIT : SPI not enabled
644 : INFO : Plugins: 74 [Normal] [Testing] [Development] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
646 : WIFI : Set WiFi to STA
679 : WIFI : Connecting NETGEAR attempt #0
2016 : WD   : Uptime 0 ConnectFailures 0 FreeMem 13544
4582 : WIFI : Connected! AP: NETGEAR (80:3F:5D:6E:37:67) Ch: 13 Duration: 3902 ms
7178 : WIFI : DHCP IP: 192.168.0.26 (ESP-Easy-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 2595 ms
7184 : Webserver: start

```

megamarco833, 2020-03-10:
now i flashed directly: ESP_Easy_mega-20200204_normal_ESP8266_1M.bin
```

eset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
   ▒U106 : Info  :

INIT : Booting version: mega-20200204 (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
107 : Info  : INIT : Free RAM:32632
108 : Info  : INIT : Warm boot #1 Last Task: Background Task - Restart Reason: Software/System restart
110 : Info  : FS   : Mounting...
118 : Info  : FS   : Mount successful, used 75802 bytes of 113201
590 : Info  : CRC  : program checksum       ...OK
609 : Info  : CRC  : SecuritySettings CRC   ...OK
610 : Info  : CRC  : binary has changed since last save of Settings
612 : Info  : WIFI  : Start network scan
2829 : Info  : INIT : Free RAM:28664
2831 : Info  : INIT : I2C
2831 : Info  : INIT : SPI not enabled
2926 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
2940 : Info  : WIFI  : Scan finished, found: 4
2947 : Error : WIFI : No valid wifi settings
3049 : Info  : WIFI : Set WiFi to AP+STA
3975 : Info  : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
3977 : Error : WIFI : Could not prepare WiFi!
4273 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 22352 WiFiStatus 0
18849 : Info  : AP Mode: Client connected: 94:0E:6B:04:F7:33 Connected devices: 1
18853 : Info  : Webserver: start
33716 : Info  : WIFI : Credentials Changed, retry connection. SSID: NETGEAR
33751 : Info  : WIFI  : Start network scan
33754 : Info  : WIFI : Connecting NETGEAR attempt #0
34477 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 17072 WiFiStatus 6
37580 : Info  : WIFI : Connected! AP: NETGEAR (XX:XX:XX:XX:XX:XX) Ch: 13 Duration: 3614 ms
39718 : Info  : WIFI : DHCP IP: 192.168.0.26 (ESP-Easy-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 2304 ms
39722 : Info  : SaveToFile: free stack: 3584
40320 : Info  : FILE : Saved config.dat
40321 : Info  : SaveToFile: free stack after: 3584
40323 : Info  : SaveToFile: free stack: 3584
40657 : Info  : FILE : Saved security.dat
40658 : Info  : SaveToFile: free stack after: 3584
40718 : Info  : firstLoopConnectionsEstablished
64262 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 16352 WiFiStatus 3
94261 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 15152 WiFiStatus 3
124262 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 9000 WiFiStatus 3
127221 : Info  :  Webserver args: 0: 'oldrulesengine' length: 2 1: 'messagedelay' length: 3 2: 'ntphost' length: 0 3: 'dststartweek' length: 1 4:
127228 : Info  : SaveToFile: free stack: 3248
127813 : Info  : FILE : Saved config.dat
127814 : Info  : SaveToFile: free stack after: 3248
144954 : Info  :  Webserver args: 0: 'cmd' length: 6
144956 : Info  :      : Rebooting...

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
   ▒U107 : Info  :

INIT : Booting version: mega-20200204 (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
108 : Info  : INIT : Free RAM:32632
110 : Info  : INIT : Warm boot #2 Last Task: Const Interval timer, id: 3 - Restart Reason: Software/System restart
112 : Info  : FS   : Mounting...
119 : Info  : FS   : Mount successful, used 75802 bytes of 113201
587 : Info  : CRC  : program checksum       ...OK
605 : Info  : CRC  : SecuritySettings CRC   ...OK
630 : Info  : INIT : Free RAM:29456
632 : Info  : INIT : I2C
632 : Info  : INIT : SPI not enabled
728 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
835 : Info  : WIFI : Set WiFi to STA
870 : Info  : WIFI : Connecting NETGEAR attempt #0
2098 : Info  : WIFI : Connected! AP: NETGEAR (XX:XX:XX:XX:XX:XX) Ch: 13 Duration: 1125 ms
2749 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 24712 WiFiStatus 6
4648 : Info  : WIFI : DHCP IP: 192.168.0.26 (ESP-Easy-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 2559 ms
4653 : Info  : Webserver: start
4656 : Info  : firstLoopConnectionsEstablished
32209 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
62209 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
92209 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
122209 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 21032 WiFiStatus 3
152210 : Info  : WD   : Uptime 3 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
212210 : Info  : WD   : Uptime 4 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
242210 : Info  : WD   : Uptime 4 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
272209 : Info  : WD   : Uptime 5 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
302210 : Info  : WD   : Uptime 5 ConnectFailures 0 FreeMem 21680 WiFiStatus 3
```

the serial log was setted to "info"
i entered inside Easy_0 AP and set to debug, then reboot
it was connected for i while at NETGEAR network (my local network) with IP 192.168.0.26 but afterthat it disconnect and never become visible

thomastech, 2020-03-10:
From reviewing your logs I believe you have a WiFi library compatibility issue that is related to the hardware. I experienced that last week on a Sonoff Basic that was the exact same vintage as your hardware.

@TD-er:
If not too much trouble, it would be a useful experiment to try a build using the _ALT_WIFI_ library  (PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122). But if the latest Mega Release branch has this in it then please recommend the bin file name that should be used.

- Thomas

megamarco833, 2020-03-10:
```
32203 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 21824 WiFiStatus 3
62204 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 20576 WiFiStatus 3
92203 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 18672 WiFiStatus 3
>rebooyt
110890 : Info  : Command: rebooyt
110897 : Info  : Command unknown: "rebooyt"
Command unknown: "rebooyt"
>reboot
113605 : Info  : Command: reboot

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
   ▒U105 : Info  :

INIT : Booting version: mega-20200204 (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
107 : Info  : INIT : Free RAM:32632
108 : Info  : INIT : Warm boot #1 Last Task: Background Task - Restart Reason: Software/System restart
110 : Info  : FS   : Mounting...
118 : Info  : FS   : Mount successful, used 75802 bytes of 113201
585 : Info  : CRC  : program checksum       ...OK
603 : Info  : CRC  : SecuritySettings CRC   ...OK
628 : Info  : INIT : Free RAM:29472
630 : Info  : INIT : I2C
630 : Info  : INIT : SPI not enabled
727 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
833 : Info  : WIFI : Set WiFi to STA
869 : Info  : WIFI : Connecting NETGEAR attempt #0
2208 : Info  : WIFI : Connected! AP: NETGEAR  Ch: 13 Duration: 1126 ms
2753 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 24728 WiFiStatus 6
4968 : Info  : WIFI : DHCP IP: 192.168.0.26 (ESP-Easy-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 2956 ms
4974 : Info  : Webserver: start
4977 : Info  : firstLoopConnectionsEstablished
32208 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 17576 WiFiStatus 3
62207 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 17936 WiFiStatus 3
92208 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 17312 WiFiStatus 3
122208 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 18248 WiFiStatus 3
152208 : Info  : WD   : Uptime 3 ConnectFailures 0 FreeMem 18248 WiFiStatus 3
182207 : Info  : WD   : Uptime 3 ConnectFailures 0 FreeMem 19760 WiFiStatus 3
212207 : Info  : WD   : Uptime 4 ConnectFailures 0 FreeMem 19760 WiFiStatus 3
242208 : Info  : WD   : Uptime 4 ConnectFailures 0 FreeMem 19760 WiFiStatus 3
```
after many trial i succeed on having for a while the web page using a reboot inside serial monitor.
but as you can see there are many disconnecting, and in a while the sonoff will be not visible on the network

thomastech, 2020-03-10:
> after many trial i succeed on having for a while the web page using a reboot inside serial monitor.
but as you can see there are many disconnecting, and in a while the sonoff will be not visible on the network

Just more symptoms of the WiFi library issue I mentioned. So far it seems to involve Sonoff Basic V1 modules made 2-3 years ago. But the sample size is very small (one of mine and maybe this one of yours). So consider my conclusion to be a gut feeling.

I think you'll need @TD-er to assist with this. If the latest Mega release does not have the special WiFi library then a custom build will be needed to test things out.

- Thomas

TD-er, 2020-03-10:
I created 2 different 1M test builds, based on different SDK builds.
[1M_wifi_testbuilds_issue2931.zip](https://www.dropbox.com/s/ieinpq4om7jtgpr/1M_wifi_testbuilds_issue2931.zip?dl=0)

They both are based on the current HEAD of the mega branch.

thomastech, 2020-03-10:
> I created 2 different 1M test builds, based on different SDK builds.

Thanks. Fingers are crossed. Place your bets now.

- Thomas

TD-er, 2020-03-10:
I think his network may have gotten one of those very swiftly rolled out security patches which we have not yet implemented in ESPEasy.
Those patches do disable all TCP/IP traffic and switch over completely to UDP.
This is meant to prevent all handshakes, to protect their users.

megamarco833, 2020-03-10:
> 
> 
> I created 2 different 1M test builds, based on different SDK builds.
> [1M_wifi_testbuilds_issue2931.zip](https://www.dropbox.com/s/ieinpq4om7jtgpr/1M_wifi_testbuilds_issue2931.zip?dl=0)
> 
> They both are based on the current HEAD of the mega branch.

> 
> 
> I think his network may have gotten one of those very swiftly rolled out security patches which we have not yet implemented in ESPEasy.
> Those patches do disable all TCP/IP traffic and switch over completely to UDP.
> This is meant to prevent all handshakes, to protect their users.

hi TD-er i do not think that is related to my home network because i have for example nodemcu, wemoss and ESP12 breadbord where i use espeasy and i do not find issue.
anyway i'm testing the two versions
in a while a came back with feedbacks
thanks

megamarco833, 2020-03-10:
i flashed first this:
normal_ESP8266_1M_SDK22x_190703
i do not see any improvemts. it's very unstable
the connection was lost many times and the time respond on web pages are very long

then i tested normal_ESP8266_1M_SDK22x_191122
same situation, very slow and unstable with continue lost of connection where i need to power off and power on

what i notice is that with 2018 version of PUYA when i connected to AP Easy_0 and give my credentials after countdown the web page show me the internal IP address that was given to sonoff
with new version from github never. after the countdown the web page was unable to load and i have to manully search the ip that was assignet at sonoff
with first test version tested (normal_ESP8266_1M_SDK22x_190703) after the countdown it show all parameters of web page (main, config, controller, hardware etc..) but inside AP network (192.168.4.1) and inside main page i can see the internal IP given.
same also with second test version tested, but with a different graphics

thomastech, 2020-03-10:
@megamarco833:
That's a let down. Was hoping for better news.

@TD-er :
The special wifi build that is working for me has these details:
```
Build: 20104 - Mega
System Libraries: ESP82xx Core 2_6_3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support
Plugin Count: 80 [Normal] [Testing]
Build Time: Mar  4 2020 20:43:39
```
Note: This version was for 4MB Flash, not 1MB.

BTW, I didn't try out your two new 1MB test builds. The Sonoff node is in-use and not a good time to take it offline.

- Thomas

TD-er, 2020-03-11:
> what i notice is that with 2018 version of PUYA when i connected to AP Easy_0 and give my credentials after countdown the web page show me the internal IP address that was given to sonoff
> with new [...]

OK, that's something I must look into as that is a bad user experience.

I also got the feeling something has changed regarding WiFi (maybe in the core?) as it also takes a while on some specific nodes I have to get the first network traffic flowing after a cold boot.
It only happens on some nodes and I have no clue why.

@megamarco833 My remark [here](https://github.com/letscontrolit/ESPEasy/issues/2931#issuecomment-597342342) was a (failed?) attempt to make a nerdy joke referring to the Corona virus.

megamarco833, 2020-03-11:
> OK, that's something I must look into as that is a bad user experience.
>
> I also got the feeling something has changed regarding WiFi (maybe in the core?) as it also takes a while on some specific [...]

ahahah no no, the joke was understand, but behind that i was thinking if some settings of internal net could make this happened, but i have others ESP without issue.
anyway i think that the problem could be linked also for first early version in 2019
because the versions marked as PUYA works well, the versions that are coming after that not.
i tested some mega version of firts months of 2019 and i got issue with this hardware
just to make a double check using a different firmware, to avoid hardware problems, i tested tasmota, last version, on this sonoff and it works.
so i think that to see what's is affecting only this hardware we can have a look on very old versions, but i could imagine that in more than 1year a lot of works/changing was implemented!

Last question about sonoff or hardware that has only 1Mb, if i would like to update the fw from a sonoff like this, what i have to do, if i would like to use OTA?
i tried 2step and minimal OTA but both failed due to a not enouh free space
thanks

TD-er, 2020-03-11:
Tasmota does not use SPIFFS filesystem, so it may not be affected by PUYA chips.
I will look at the used flags to build the binaries, but as far as I know the current core libs should support PUYA chips (as I made the commit myself to the core libs)

thomastech, 2020-03-11:
> I will look at the used flags to build the binaries, but as far as I know the current core libs should support PUYA chips (as I made the commit myself to the core libs)

Assuming it is a valid tag, the _Firmware System Library_ line item on the /sysinfo page shows "PUYA support" on all my various ESPEasy nodes. So it appears the build flags are working.

I pulled my Sonoff Basic V1 offline and tested the two new 1MB bins. 
1. The normal_ESP8266_1M_SDK22x_190703 bin had slow refresh, some browser page hangs, and timeouts. Sometimes it would not respond for awhile, then randomly partially work again.
2. The normal_ESP8266_1M_SDK22x_191122 bin worked well, but occasionally a page would momentarily hang for a second or two during refresh. Also, web log had intermittent "resource" errors. But overall, I would say it is "working."

I don't experience any browser issues with the "secrete sauce" bin that was created last week for my Sonoff Basic V1. All page refresh is fast and reliable. But this file is a 4MB [TESTING] build since I upgraded all my Sonoff 1MB Flash chips to 4MB so I could do OTA. For reference, here's the file:
https://www.dropbox.com/s/2q0g9w6xbub6lfp/test_ESP8266_4M1M_VCC_secret_ingredient.bin?dl=0

Since I notice a subtle differences in the two "working" builds it seems that there is some important clue that is being overlooked. Maybe the issue is related to PUYA and/or the WiFi library that affects RF calibration, or maybe not. 

@TD-er: It would be ideal if you had a Sonoff Basic V1 mfg'd in 2017 - 2018. If by chance you have one then consider testing it out with the latest firmware.

- Thomas

TD-er, 2020-03-11:
I do have them.... somewhere.
I even should have a batch of 4M flash chips and a proper soldering iron.
So in theory I should be able to recreate your setup.

Only thing is, I have moved here very recently and my "lab" is now not the most well sorted one.
[image]

As I'm in this temporary house now, I only have a room of roughly 3x3 meter and as you can see it is a bit in "still sorting things" mode.

megamarco833, 2020-03-11:
> I don't experience any browser issues with the "secrete sauce" bin that was created last week for my Sonoff Basic V1. All page refresh is fast and reliable. But this file is a 4MB [TESTING] build [...]

without manually change the chip from 1mb to 4mb is not possible to use OTA upgrade?

thomastech, 2020-03-11:
> As I'm in this temporary house now, I only have a room of roughly 3x3 meter and as you can see it is a bit in "still sorting things" mode.

Your lab has a ways to go before it reaches the notoriety of Bob Pease's office. And by the time you get it the way you like you'll be packing up to move back into your rebuilt house.

> without manually change the chip from 1mb to 4mb is not possible to use OTA upgrade?

From what I understand this requires flashing with a recent "minimal_core" that is smaller than 599 kB (614384 bytes). After that, OTA would be possible with another minimal build that meets the size constraints. 

TD-er has been busy getting the size small enough to do this. I believe there are some bin files available that meet the size limitation.  For the record, I've never tried it.

- Thomas

megamarco833, 2020-03-23:
> 
> 
> I do have them.... somewhere.
> I even should have a batch of 4M flash chips and a proper soldering iron.
> So in theory I should be able to recreate your setup.
> 
> Only thing is, I have moved here very recently and my "lab" is now not the most well sorted one.
> [image]
> 
> As I'm in this temporary house now, I only have a room of roughly 3x3 meter and as you can see it is a bit in "still sorting things" mode.
> 
> At least one of the Basic nodes is hanging on the DIN rails under my Astron cap.

hi TD-er did you find time and a sonoff basic to verify the issue bug on flashing it?
thanks so much

TD-er, 2020-03-23:
Not yet as last week was a bit normal, but mostly chaotic.

Well today we learned people are forbidden to be with more than 2 persons within 1.5m (400 euro fine) for the next few months.
This also means I will have to take care of my daughter during the day when my better half is working in the hospital (for more hours than contracted).
So I hope we will get some more structure in the days here.

thomastech, 2020-03-24:
Please extend my warmest Thank You to your wife for being on the front lines. 

And it's nice to hear that the Noorlander coding team has doubled in size. Just be sure to allow for juice breaks so you don't violate any child labor laws. 

- Thomas

TD-er, 2020-03-24:
Well Lilly is now learning to read, write and to calculate, so she could indeed do the code review. (read: trying to break it)
At least she has a big interest of dad's projects (she calls them "pruts projectjes" in Dutch), so you will never know.
She is no longer accepting 3D prints without electronics in them.

megamarco833, 2020-04-01:
hi TD-er, first of all I hope that you and your family still safe!
secondary thing, did you get some time to spend over this bug on sonoff basic that's not possible to flash with latest buld of espeasy after end 2018 (ESP_Easy_mega-20181023_dev_ESP8266PUYA_1024_VCC.bin)
i would like to  update to improve wifi stability of early versiond and to avoid some wifi disconnection that i still have on this old version :-)
thanks so much

TD-er, 2020-04-01:
What is the reported sketch size of this build you're running?
If it is too big to allow for 2-step OTA, then I'm afraid only a manual update is possible.

megamarco833, 2020-04-01:
`

Storage
--
Flash Chip ID | Vendor: 0x85 Device: 0x6014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | Unknown
Flash Writes | 0 daily / 2 boot
Sketch Size | 814 kB (60 kB free)
SPIFFS Size | 110 kB (35 kB free)
`

i think there is no way to OTA right?

but anyway, no metter to flasj manually...the problem is that if i flash manually, after connecting to ESP_Easy net and select the internal SSID and password, is impossible to see on internal net the sonoff and ESP_Easy net desapper
the only way to have it working is manual flashin an old version of EspEasy where there was PUYAversion:
ESP_Easy_mega-20181023_dev_ESP8266PUYA_1024_VCC.bin
in this case no problems.

do you remember many trial that we did following this post here ? :-)

TD-er, 2020-04-01:
> do you remember many trial that we did following this post here ? :-)

I think those memories are put away somewhere very deep, locked behind very thick doors...

megamarco833, 2020-04-01:
ahahahah....
gooooood :-)
.....anyway, my short summary at the end of last post has descrived the issue?....or wake up some memories?

megamarco833, 2020-04-02:
hi tested also the latest build:
ESP_Easy_mega-20200328_normal_ESP8266_1M.bin
now is better, because after the reboot the new IP of sonoff is visible inside the net.
the problem is that i try to go at the sonoff web page it is very very very slow and almost impossible to configure it, because in a while it become not respondin

thomastech, 2020-04-03:
Just to sum things up from the previous forum discussion:

Like megamarco833, this strange WiFi connectivity problem only affects my old Sonoff Basic V1 Module (ESP8266). My three Sonoff Basic V2 (ESP8255) modules don't have the problem.

So there's something evil about the old Sonoff Basic. I believe it will require a special person with vast coding talents and a generous spirit to solve the problem. Cough, @TD-er, cough. 

- Thomas

TD-er, 2020-04-03:
Maybe this also has something to do with switching SDK builds (like we did on some other topic as well)

thomastech, 2020-04-03:
> Maybe this also has something to do with switching SDK builds

Affirmative. The problem goes away with a special [TESTING] build config you prepared for me 4 weeks ago. It's libraries are:
ESP82xx Core 2_6_3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support

See: https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=7474#p42804

- Thomas

megamarco833, 2020-04-04:
i tryed the version at post linked, but after flashing, i can't see the network SSID ESPeasy
:(
i flashed DOUT as usual for Sonoff basic R2

thomastech, 2020-04-04:
> i tryed the version at post linked, but after flashing, i can't see the network SSID ESPeasy

My Sonoff Basic has been upgraded to 4MB Flash, so the linked bin file is a 4MB release. It won't work in a stock Sonoff Basic.

- Thomas

TD-er, 2020-04-04:
I will prepare a test build using that SDK for a Sonoff Basic.

TD-er, 2020-04-04:
I checked the current platformio.ini config and the last buils should be using the `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122` that was working well on Thomas' units.
So the last nightly builds should have these for core 2.6.3 builds (default core)
The builds marked with "beta" have a different SDK build `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191105`

megamarco833, 2020-04-04:
sorry, where is the link for nightly build? i can't find it...

TD-er, 2020-04-04:
See: [nightly builds](https://github.com/letscontrolit/ESPEasy/releases)

megamarco833, 2020-04-04:
thanks i thought that there was another version :-)

i tested:
`20200328_normal_ESP8266_1M_VCC.bin` 
and
`20200328_normal_ESP8266_1M.bin`

in both case, after flash, i see the ESP_easy SSID, but when i try to connect to this SSID, the connection duration is just for a while and the i kicked out.
i try also using PC but when it connect, the IP 192.168.4.1 will display nothing and after some time i see web page not responding.
during tentative, one time, just for a while, i see the web page, i connect to my internal SSID and give password, but after the countdown, i can't see ESP on my internal network, and i continue to see ESP_easy network also after reboot. So i think that measns that the confiuration was not saved.

TD-er, 2020-04-05:
What brand of flash chips does this node have? PUYA?

megamarco833, 2020-04-05:
yes, i see that the chip labeled on pcb as U6 is marked PUYA

TD-er, 2020-04-05:
OK, that's strange.
The library we use should have PUYA support enabled.

Micha-he, 2020-04-06:
@TD-er : Herr, in my ESP01 with PUYA-flash, the actual Firmware works. Maybe another New ID?

megamarco833, 2020-04-09:
hi TD-er
i bougt a sonoff brand new, same hardware: sonoff basic R2 
tested with:

20200328_normal_ESP8266_1M_VCC.bin
and
20200328_normal_ESP8266_1M.bin

but also in this case, when it is flashed i connect to IP 192.168.4.1 and i set my internal SSID and give the password, after the countdown, i can't see the DHCP ip gived to ESP on my internal network,.

And after some minutes, i continue to see ESP_easy network, also after reboot.

So i think that it means that the confiuration was not saved.
if you have some test buld i can check.
having a brand new sonoff basic R2 i think that we can exclude issue from my own hardware.

TD-er, 2020-04-09:
That new one, does have PUYA flash?

[237 more replies]

PR #3148 (2020-07-20): [WiFi] Reduce wifi reset calls and call full wifi init at reset (#3147)
Fixes: #2931
Fixes: #1258
Fixes: #3155
Fixes: #3141
Fixes: #3111 

Connected state is now based on the state reported by the internal WiFi object.
Frequent calls to `reset()` of WiFi are now rate limited to make sure we don't end up in a loop.
PR #3179 (2020-07-31): [Alt WiFi] Use different SDK build for nodes with WiFi issues (#2931)
Fixes: #2931
commit ec8a0e20 (2020-07-31): [Alternate WiFi] Use different SDK build for nodes with WiFi issues (#2931)

Fixes: #2931
closed 2023-09-09 by TD-er