letscontrolit/ESPEasy #2934: generic dummy device output devices amount gets reset on cold boot

GravityRZ, 2020-03-08:
currently running on mega-20200204 but this bug has been in previous builds as well.

i use a generic dummy device and specified 2 output fields to use
[image]
[image]

so far so good but every time i do a cold boot or change things this dummy device resets its amount of fields and goes back 2 4 fields

[image]

[image]

not sure exactly when it happens but i have had it several times in the past and not only on cold/warm reboots bvbut also after making changes

everything keeps on working but when i select 2 fields instead of 4 i wants then to stay that way forever(unless i change the amount myself)

TD-er, 2020-03-08:
The output values of tasks are stored in RTC memory and restored at boot.
The RTC memory is cleared when power is removed, so a cold boot will indeed clear all values.
This is "by design".

It could be possible the values are also reset when you change settings of that task. Have to check the source for that to make sure.
Please note that when you have multiple instances of the Dummy plugin, that all must have the same output type.
The output type is shared among all instances, so if you have one set to 3 values and one using 4, then the last saved version may determine the number of active output values.

GravityRZ, 2020-03-10:
ok thanks. only have one dummy device so that rules out another instance

if this is by design(or to avoid problems with people having more values) then no problem, i can live with that.
it does not affect functionality

closed 2020-03-10 by GravityRZ