letscontrolit/ESPEasy #2936: 2Way Flash changes Hostname

fly74, 2020-03-08:
Hi there,

the last 2 releases changes in the 2WayFlashMode the hostname. So you have to look at the router to get the hostname for the second step. After the 2nd flash all is fine and the hostname is correct again.

TD-er, 2020-03-08:
I think it has to do with the more recent changes in how the hostname is generated from the settings (append unit number etc.) and the 2-way intermediate build has not been updated in years.

I'm afraid that making a new intermediate version may be larger.
But we will need to look into a new 2-step version, also because we really could use the GZIP compression that's possible in the very recent core version.

fly74, 2020-03-08:
The releases before are working perfekt. Now it is difficult to flash quickly the 1M OTA.

TD-er, 2020-03-08:
I've had some discussions with Martinius over the options for improving the 2-step OTA.
Newer core versions do make quite a bit larger builds, but the GZIP option does give some extra room.
So then we must skip some other features we don't need.
The idea is to get rid of SPIFFS and parsing the config file.
This means we must keep the settings to have it working during the OTA.

To be honest, I have not been thinking about the hostname, but that one should be kept also.

- IP config
- SSID/BSSID/pass
- admin password?
- hostname

Maybe something else is also needed?

fly74, 2020-03-08:
No, that's all. :)

jessie00, 2020-04-21:
Hi everyone,
I saw the same thing since the release mega-20200305.
All other next are the same issue.
After update when I go to:
 Tools > System Variables 
I see 
%sysname%  =  sonde02_2 =  sonde02_2
but the config hostname is just sonde02.

I try with ESP_Easy_mega-20200305_normal_ESP8266_4M1M.bin
With a Wemos D1 mini

When I back in release mega-2020-03-01 the hostname is correct again.

This issue affect sending to MQTT. I use the sensors information to graph temperture and humidity in all room in my house.

tonhuisman, 2020-04-21:
> %sysname% = sonde02_2 = sonde02_2
> but the config hostname is just sonde02.

On the Config page there is a setting "Append Unit Number to hostname:", I guess this option is on for your ESP...

jessie00, 2020-04-23:
Hi Ton,

You rigth !.
If y try with 0, I don't have issue.
If I have  a number not 0, this number append with "_" at the end of the
hostname.

Thank a lot.
Bye

fly74, 2020-04-26:
Can this plz have more priority, it is not funny to update many esp's and looking for the changed hostname for the 2nd step. Is it difficult to roll back to a version which worked?

TD-er, 2020-04-26:
Like I said, the 2-step OTA has not changed, it may be the way the normal ESPEasy is creating the hostname from the settings.

uzi18, 2020-04-26:
maybe default setting for hostname ending is incompatible with older releases