letscontrolit/ESPEasy #2938: Custom Web Page Broken After Mega-20200222

thomastech, 2020-03-09:
### Summarize of the problem/feature request
Mega-20200301 and Mega-20200305 have broken my custom web page's formatting. From reviewing the posted html code it seems that styles and scripts are being truncated. This problem has not been observed on prior releases.

### Expected behavior
The custom page normally looks like this:
[image]
The html output is:
[html_ok.txt](https://github.com/letscontrolit/ESPEasy/files/4304235/html_ok.txt)

### Actual behavior
Beginning with Mega-20200301 the page is presented like this, which has missing buttons and Pump Status:
[image]
The html output is:
[html_bad.txt](https://github.com/letscontrolit/ESPEasy/files/4304236/html_bad.txt)

### Steps to reproduce

1. Install my custom .esp file (see below).
2. Visit custom page.
3. Compare Mega-20200222 with Mega-20200305.

Hardware:
NodeMCU V2, 4MB

ESP Easy version: 
Problem has been observed on Mega-20200301 and Mega-20200305

### Custom web page, webcontrol.esp
```
<!-- ESPEasy Web Control for Hot Water Circulation -->
<meta name="viewport" content="width=width=device-width, initial-scale=1">
<STYLE>
 * {font-family:sans-serif; font-size:xx-large;}
 .button {margin:4px; padding:4px 10px; background-color:#07D; color:#FFF; text-decoration:none; border-radius:4px}
 .pwron {background-color:green; border-radius:2px; color:white; font-weight: bold; margin:1px}
 .pwroff {background-color:white; border-radius:2px; color:black; font-weight: normal; margin:3.5px}
 .pumpon {background-color:red; border-radius:2px; color:yellow; font-weight: bold; margin:1px}
 .pumpoff {background-color:white; border-radius:2px; color:black; font-weight: normal; margin:3.5px}
</STYLE>
<HTML>
<HEAD>
  <meta charset="utf-8">
  <meta http-equiv="refresh" content='2; URL=/webcontrol.esp'> <!-- Reload every 2 secs -->
  <title>Circulation Pump Control</title>
</HEAD>
<body>
<h1><font color="blue"><strong>Hot Water Circulation Pump Control</strong></font></h1>
<p></p>

<table>
 <tr>
  <td><p style="font-size:xx-large;font-weight: bold;">Temperature: [DS18B20#temperature]°F</p></td>
 </tr>
</table>
<p></p>

<table>
 <tr>
  <td><p id="acpwr"></p></td>
  <td><a class='button link' href="/webcontrol.esp?cmd=event,RelayOn">On</a></td>
  <td><a class='button link' href="/webcontrol.esp?cmd=event,RelayOff">Off</a></td>
 </tr>
</table>
<p></p>

<table>
 <tr>
  <td><p id="pump"></p></td>
<!-- <td><a class='button link' href="/webcontrol.esp?cmd=event,RelayOff">Stop Pump</a></td> -->
 </tr>
</table>

<script>
 var pwrstate
 if ([OPTO#input]>=1) {
  pwrstate = "<span class='pwron'>Start Pump:</span>";
 }
 else {
  pwrstate = "<span class='pwroff'>Start Pump:</span>";
 }
 document.getElementById("acpwr").innerHTML = pwrstate;
</script>

<script>
 var pumptate
 if ([STATEVAR#rlyval]>=1) {
  pumpstate = "<span class='pumpon'>Pump Status: Requesting Circulation</span>";
 }
 else {
  pumpstate = "<span class='pumpoff'>Pump Status: Idle</span>";
 }
 document.getElementById("pump").innerHTML = pumpstate;
</script>
</body>
</HTML>
```

uzi18, 2020-03-09:
this should be fixed with latest build

TD-er, 2020-03-09:
Can you test with this test build: https://github.com/letscontrolit/ESPEasy/issues/2932#issuecomment-596200103 ?

ironataerial, 2020-03-09:
I do not mean to hijack the thread nor to be a troll, but this has yet to fix the gpio,x,y JSON return formatting. Can only test the new releases but can not use them on deployed units.

TD-er, 2020-03-09:
> but this has yet to fix the gpio,x,y JSON return formatting. Can only test the new releases but can not use them on deployed units.

That's a different issue, as this was caused by the replacement of all {...} occurrences in the parser.

thomastech, 2020-03-09:
> Can you test with this test build: #2932 (comment) ?

Thanks, ESP_Easy_mega-20200305-18-PR_2933_normal_ESP8266_4M1M.bin is working.

- Thomas

TD-er, 2020-03-10:
Thanks for testing.
I just started a new nightly, so this issue can be closed.

closed 2020-03-10 by TD-er