letscontrolit/ESPEasy #2940: ESPEasy Status LED is not working on ESP32 with buildin LED

mister-gefahrensucher, 2020-03-13:
### Expected behavior
i have installed the latest Version ESPEasy mega-20200305 on an ESP32.
Tried to use buildin LED (GPIO2) as WIFI Traffic indicator (like it is working on ESP8266 devices)

### Actual behavior
The LED stays always drak.

### Steps to reproduce
1.  Use buildin LED (GPIO2) as a normal device and switsch it on and off-> working. So GPIO2 is the right one.
2. Now assing it as WIFI Traffic indicator. Nothing happend.
3. Reboot the device. Nothing happend.
4. Sorry i have not tried an external LED because i don't have enyone.

### System configuration

Hardware:
ESP32-WROOM-32 (ESP32S) (Hardware like nodemcu)
ESP32 NodeMCU IoT Development Board mit Espressif ESP-WROOM-32
[image]

ESP Easy version: 
mega-20200305

ESP Easy settings/screenshots: 
[image]
[image]

TD-er, 2020-03-13:
The notification LED is using the PWM feature, so maybe that's the problem with ESP32?

mister-gefahrensucher, 2020-03-14:
I've tested with mqtt command PWM,2,0,2000 or PWM,2,1000,2000
The builin LED changes the brightness. So PWM is working fine.
I fear there is somthing wrong pin assignment to WIFI or the WIFI module don't deliver data for dimming.

mister-gefahrensucher, 2020-03-14:
I've tested also external LED with mqtt command PWM,2,23,2000 or PWM,23,1000,2000
The builin LED changes the brightness. So PWM is working fine.
But GPIO23 as WIFI Traffic indicator, no chance.

thomastech, 2020-03-16:
In the statusLED() function (see Misc.ino) the Status LED's analogwrite() is only available to the ESP8266. See the code here: https://github.com/letscontrolit/ESPEasy/blob/mega/src/Misc.ino#L757

Maybe something like this (untested) is the solution?
```
    #if defined(ESP8266)
      analogWrite(Settings.Pin_status_led, pwm);
    #endif
    #if defined(ESP32)
       analogWriteESP32(Settings.Pin_status_led, pwm);
    #endif // if defined(ESP32)
```
- Thomas

TD-er, 2020-03-16:
@thomastech That looks like a valid fix indeed.

mister-gefahrensucher, 2020-03-18:
@thomastech 
@TD-er 
i can confirm the fix work's. Now my led is blinking like in esp8266 devices

TD-er, 2020-03-19:
Fixed it along with another PR I am about to merge.

commit 023c97d5 (2020-03-21): ESPEasy Status LED is not working on ESP32 with buildin LED (#2940)

Fixes: #2940

Copy-paste fix from @thomastech https://github.com/letscontrolit/ESPEasy/issues/2940#issuecomment-599627656
closed 2020-03-21 by TD-er