letscontrolit/ESPEasy #2941: Flash instruction for ESP32 Devices

mister-gefahrensucher, 2020-03-13:
**1.** only if you have a fresh one with factory firmware installed, erase the flash with esptool.
**2.** one time i have to flash the bootloader and partition2 and boot_app0.bin
esptool.exe --chip esp32 --port COM6 --baud 256000 --before default_reset --after hard_reset erase_flash
esptool.exe --chip esp32 --port COM6 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x8000 partitions2.bin 0xe000 boot_app0.bin 0x1000 bootloader.bin
**3.** then flash the firmware
esptool.exe --chip esp32 --port COM6 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 ESP_Easy_mega-20191208_test_ESP32_4M316k.bin

later i was able to update the firmware only with
esptool.exe --chip esp32 --port COM6 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 ESP_Easy_mega-20200305_test_ESP32_4M316k.bin

one hint: start the flashtool then hit the boot button on esp32. The esptool is not the one came with ESP-Easy this does not work with ESP32 (i don't kow why). I found esptool.exe version of the esptool.py. I use this because i don't have installed python. But i think esptool.py should also work.

here is the esptool.exe i use:
[esptool.zip](https://github.com/letscontrolit/ESPEasy/files/4330642/esptool.zip)

TD-er, 2020-03-13:
Thanks for making it into an issue.

haplm, 2020-03-15:
Thanks for the info, @mister-gefahrensucher, you have saved my day. Normally I'm flashing from platformio, but I needed to flash official release to check on something weird happening to my build, and this was the only way how to do it (using esptool.py, as I'm on mac). 

@TD-er - it would be also great to document where to get the standard *.bin  files (I have downloaded mine from some post somewhere, but that's for sure not the best way)

megamarco833, 2020-03-23:
hi, we have talked some months ago on forum:
https://www.letscontrolit.com/forum/viewtopic.php?p=42557#p42557
is it possible also falsh using esp32 download tool 3.6.6

in this post 
https://www.letscontrolit.com/forum/viewtopic.php?p=42559#p42559
we also add the files:
1) bootloader.bin
2)boot_app0.bin
3)partition2.bin

i do not know if these are the new one or best one to flash, but maybe, due to the fact that they are needed for ESP32, maybe is better to include inside the build.
what do you think @TD-er ?
we talked about it.
thanks

TD-er, 2020-03-23:
Yep we certainly talked about it and I do feel we should include them files in the nightly build.

TD-er, 2020-03-23:
I added an initial PR for this: https://github.com/letscontrolit/ESPEasy/pull/2964
I still have to check if I can collect newer versions of these files and still have to add documentation for it.

uzi18, 2020-03-25:
@mister-gefahrensucher @haplm @megamarco833 could you please attach here flash dumps of your esp32 ?

haplm, 2020-03-25:
@uzi18 - could you help me with how to determine the addresses for esptool.py's 'read_flash' command? Start is I guess 0x0, but I'm not sure how to set the end address.

uzi18, 2020-03-25:
from 0 to max to 0x10000: 
https://github.com/espressif/esptool#read-flash-contents-read_flash

megamarco833, 2020-03-25:
> 
> 
> @mister-gefahrensucher @haplm @megamarco833 could you please attach here flash dumps of your esp32 ?

@uzi18 
https://www.letscontrolit.com/forum/download/file.php?id=4405&sid=450c3b40caeb3cd557e9881a5f328696

uzi18, 2020-03-25:
on mymodule this command works like charm:
```
esptool.py --chip esp32 --port "/dev/ttyUSB0" --baud 115200 read_flash 0 0x10000 dump.bin
```

it creates 64k file, zip it and attach here by drag&drop please

haplm, 2020-03-25:
I've just realised that after having these difficulties, I have switched to flashing using MacOS and PlatformIO, where it is very straightforward. I guess that you are not interested in such dump, correct? But if you absolutely need it, I can re-create the situation using one of my spare boards...

uzi18, 2020-03-25:
@haplm you can still do it in platformio terminal :)

haplm, 2020-03-25:
Ah, I see. Can you send me your email, so I can send it to you? Just don't want to post it here (WiFi password)

uzi18, 2020-03-25:
@haplm already sent you an email.

uzi18, 2020-03-25:
@haplm you can find me on gitter.im open private room there :)

uzi18, 2020-03-25:
your email does not work ;/
find me here: https://gitter.im/jeelabs/esp-link

uzi18, 2020-03-25:
@haplm your login is password https://onetimesecret.com/secret/8ip7z16t5mzqv42b0ysw7dh3gevk0uk

uzi18, 2020-03-26:
@haplm #2967 is ready :)

you can try to flash factory firmware at 0x0 address from this zip:
[test_ESP32_4M316k_firmware-factory.bin.zip](https://github.com/letscontrolit/ESPEasy/files/4384331/test_ESP32_4M316k_firmware-factory.bin.zip)

megamarco833, 2020-03-26:
Sorry, but now i'm a bit confused.
1. What's the correct bin to flash this esp32 esp-wroom-32?
Pictures attached. 
[image]
[image]
[image]

2. What are the corrects bootloader.bin boot_app.bin partition2.bin
? 
And at what addresses should loaded?

3. Is it ok to use esp32 download tool 3.6.6 to load firmwares?

Thanks!

uzi18, 2020-03-26:
just try factory.bin attached to my post with esp32 download tool flash it at address 0x0

megamarco833, 2020-03-26:
hi, i tested 
it flash without issue, i find the AP wifi and no problems to connect

but i see that after config the ESP32 setting for example MQTT and NTP server
i see always 1970 as year (on others ESP it works)
and many disconnection of MQTT

```
227023: WIFI : Static IP: 192.168.0.123 (ESP32-123-123) GW: 192.168.0.1 SN: 255.255.255.0
227138: WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4 ms
227150: EVENT: WiFi#Connected
227311: WIFI : Set WiFi to AP+STA
227413: IP : Static IP : 192.168.0.123 GW: 192.168.0.1 SN: 255.255.255.0 DNS: 8.8.8.8
227418: WIFI : Connecting NETGEAR attempt #22
227771: IMPT : MQTT 037 Intentional reconnect
228224: IMPT : MQTT 037 Intentional reconnect
228616: WIFI : Static IP: 192.168.0.123 (ESP32-123-123) GW: 192.168.0.1 SN: 255.255.255.0
228640: EVENT: WiFi#Disconnected
228826: IMPT : MQTT 037 Intentional reconnect
229278: IMPT : MQTT 037 Intentional reconnect
229656: EVENT: WiFi#Connected
229839: IMPT : MQTT 037 Intentional reconnect
230292: IMPT : MQTT 037 Intentional reconnect
230803: IMPT : MQTT 037 Intentional reconnect
231255: IMPT : MQTT 037 Intentional reconnect
231775: IMPT : MQTT 037 Intentional reconnect
232227: IMPT : MQTT 037 Intentional reconnect
232773: IMPT : MQTT 037 Intentional reconnect
233225: IMPT : MQTT 037 Intentional reconnect
233772: IMPT : MQTT 037 Intentional reconnect
234225: IMPT : MQTT 037 Intentional reconnect
234773: IMPT : MQTT 037 Intentional reconnect
235226: IMPT : MQTT 037 Intentional reconnect
235773: IMPT : MQTT 037 Intentional reconnect
```

TD-er, 2020-03-26:
JUst to be sure, can you test with DHCP also?

megamarco833, 2020-03-27:
yes, i did it, and with DHCP is working perfectly!
thanks

TD-er, 2020-03-27:
OK, so there is still some issue with the wifi event handling for static IP.

megamarco833, 2020-04-09:
there are some news on this: usingf static IP ? :-)
will be added inside next build of esp32 the files: bootloader.bin ; )boot_app0.bin  ;  3)partition2.bin  ??

uzi18, 2020-04-09:
factory bins are created now but needs to be added to release

megamarco833, 2020-04-10:
So maybe in the next build?
And in this case we just flash the bun file at 0x000 address like esp8266 and nothing more?
Last question also the issue with static ip vs dhcp is solved?
Thanks so much

TD-er, 2020-04-10:
Is there a specific issue for static IP on ESP32?

megamarco833, 2020-04-10:
Yes, look some post above. You and uzi give me a build to test, it worked but only with dhpc

https://github.com/letscontrolit/ESPEasy/issues/2941#issuecomment-605083896

TD-er, 2020-04-10:
I just built and flashed one from my IDE, to see what was the output here.

```
Building .pio\build\test_ESP32_4M316k\firmware.bin
Retrieving maximum program size .pio\build\test_ESP32_4M316k\firmware.elf
Checking size .pio\build\test_ESP32_4M316k\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.6% (used 57720 bytes from 327680 bytes)
Flash: [========= ]  93.6% (used 1778818 bytes from 1900544 bytes)
esptool.py v2.6
after_build([".pio\build\test_ESP32_4M316k\firmware.bin"], [".pio\build\test_ESP32_4M316k\firmware.elf"])
Generating factory bin for genuine esp units
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Uploading .pio\build\test_ESP32_4M316k\firmware.bin
esptool.py v2.6
Serial port COM6
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:e9:50:89
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15856 bytes to 10277...

Writing at 0x00001000... (100 %)
Wrote 15856 bytes (10277 compressed) at 0x00001000 in 0.9 seconds (effective 139.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1069.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5795.9 kbit/s)...
Hash of data verified.
Compressed 1779056 bytes to 1009410...

Writing at 0x00010000... (1 %)
Writing at 0x00014000... (3 %)
...
Writing at 0x00100000... (98 %)
Writing at 0x00104000... (100 %)
Wrote 1779056 bytes (1009410 compressed) at 0x00010000 in 89.2 seconds (effective 159.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

This results in 2 bin files in the build dir:
- `firmware.bin`
- `firmware-factory.bin`

I will change the `before_deploy` file to also include the `-factory`  file in the ZIP and document it.

megamarco833, 2020-04-10:
Yes, also for me it works, as reported days ago, only with dhcp :)
Thanks so much for all your effort and anwers! If you need other help, just ask, unfortunately it's only what i can do with my knowledge: test

TD-er, 2020-04-10:
> Yes, also for me it works, as reported days ago, only with dhcp :)

Working on that one as we speak

uzi18, 2020-04-10:
@TD-er have prepared changes for before_deploy script but ... not sure if crc2.py will handle factory bin - need to check

uzi18, 2020-04-10:
@TD-er please look at PR #2998

TD-er, 2020-04-10:
> md5 module is depracted (from >=2.5) and unavailabe in python3 - we should move to hashlib

Any idea how big that's going to be?

uzi18, 2020-04-10:
@TD-er hashlib is in standard python lib, but first need to understand how crc2.py works, shouldn't be big change

uzi18, 2020-04-10:
@TD-er and crc2.py does not work for esp32 bins.

TD-er, 2020-04-10:
> @TD-er hashlib is in standard python lib, but first need to understand how crc2.py works, shouldn't be big change

I meant if we must check it in the ESP code, then we also should implement those in the ESP code and that may add to the binary size.

uzi18, 2020-04-10:
it is not implemented for esp32 file structure is also different.
about transition to hashlib - this is trivial -> https://github.com/uzi18/ESPEasy/commit/c23426f26ab76edc84c185ee33f1e33b3a11b049
more work is needed to move from python2 to python3 in crc2.py

TD-er, 2020-04-10:
Maybe we should include the binary name at compile time instead of patching it?
The CRC is something we can also move a bit for a future patch.

uzi18, 2020-04-10:
PR is ready #3003
about crc for esp32 we need to rethink it

PR #2967 (2020-04-06): ESP32: add subtarget for genuine units firmware-factory.bin
Fixes: #2964
Fixes: #2941
Fixes: #2486
commit 1127fdaa (2020-04-12): Merge pull request #3000 from uzi18/before_deploy

Before deploy - update release script to also include factory bins for esp32 (with bootloader etc.) fix #2941
closed 2020-04-12 by TD-er