letscontrolit/ESPEasy #2942: Multiple AM2320 --> CRC FAILED

airvb, 2020-03-14:
Hello .
Tested with :
ESP_Easy_mega-20200310_dev_ESP8266_4M1M
ESP_Easy_mega-20200310_test_ESP8266_4M1M_VCC

What is expecting : 
Having 2 AM2320 sensors on I2C line.

Each sensor has been tested individually
They are working properly

When I connect the second sensors, log message shows : AM2320: CRC failed
If i disconnect one sensors, values Temperature & Humidity goes with the same values to both device

Trying sensors connected with 10K pull-up resistors on SDA and SCL et +5V  : same snag.

I have a LCD connected on I2C and it's working well.

I can help. I'm in the trial phase.
Thanks

TD-er, 2020-03-14:
It is probably the same issue as for some other plugins when running multiple instances of the same plugin.

They probably share temp values which should not be shared.

tonhuisman, 2020-03-14:
Is it even possible to have more than 1 of these sensors on an I2C bus? It's address is fixed (0x5C), so it would need a separate I2C bus for each sensor, like done in this Arduino library https://github.com/lazyscheduler/AM2320 (comments in Russian...)
The CRC error makes sense in this context, IMHO...

TD-er, 2020-03-14:
If you cannot change the I2C address, then it is indeed not possible to use multiple instances as we don't support multiple I2C buses.

uzi18, 2020-03-14:
But it is possible to use 1wire legacy mode, see datasheet

TD-er, 2020-03-14:
> But it is possible to use 1wire legacy mode, see datasheet

Is that supported in our plugin?

uzi18, 2020-03-14:
It should work, but we can add description/hint in plugin configuration

uzi18, 2020-03-14:
connect SCL-GND and SDA to gpio

airvb, 2020-03-14:
> If you cannot change the I2C address, then it is indeed not possible to use multiple instances as we don't support multiple I2C buses.

What is the point of having I2C sensors if it is not possible to use them on the data bus?

airvb, 2020-03-14:
> connect SCL-GND and SDA to gpio

Will be ok only if you have only few sensors,

TD-er, 2020-03-14:
> > If you cannot change the I2C address, then it is indeed not possible to use multiple instances as we don't support multiple I2C buses.
> 
> What is the point of having I2C sensors if it is not possible to use them on the data bus?

Well you should ask the manufacturer that question, why it makes sensors that cannot have their address set.
There are lots of I2C sensors out there and it is quite common to have a "collision" in default addresses, so just for that reason alone you should have the option to set another I2C address.
Also if you need to have several on the same bus, you need to have a unique address on that bus or else you will run into issues like you're now experiencing.

airvb, 2020-03-15:
Is there  good I2C temp & humidity sensors compatible with espeasy ?

tonhuisman, 2020-03-15:
A couple of BME280's work very good over here, and you can use 2 on the same bus, as the address is configurable (check out the images on Banggood or AliExpress to see if they have the address selection connections available on the board, not all have that!).

Check also that you receive actual BME280's, as the BMP280 has a lot of similarities but no humidity sensor, read this: https://goughlui.com/2018/08/05/note-bosch-sensortec-bmp280-vs-bme280-sensor-confusion/

uzi18, 2020-03-15:
> > connect SCL-GND and SDA to gpio
> 
> Will be ok only if you have only few sensors,

@airvb but please try it, for other users.

how. many sensors you need to connect?

TD-er, 2020-03-15:
> Is there good I2C temp & humidity sensors compatible with espeasy ?

See [supported "environment" sensors](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#environment)

uzi18, 2020-03-16:
@airvb did you tried to force 1wire mode?

airvb, 2020-03-17:
> @airvb did you tried to force 1wire mode?

sorry for the response delay.
Yes i tried the other sensor as legacy one , works well. Recognize as DHT11.

i"ll buy  BME280's in my next ali 's command to test it .

Thanks all for yr response.

TD-er, 2020-03-17:
Just make sure you buy the correct ones.
There are units sold that have a regulator and only work (stable) on 5V.

uzi18, 2020-03-17:
@TD-er so we can add AM2320 to be supported in 1wire mode (DHT plugin)

TD-er, 2020-03-17:
What is needed for that? Any change in the plugin code?

uzi18, 2020-03-17:
nothing, just set it as DHT11, and hardware: connect SCL to GND and SDA to gpio with 10k pullup as usual with DHT sensors

TD-er, 2020-03-17:
So only documentation?

airvb, 2020-03-18:
If you have to make a modification , it could be a good thing to inform than I2C sensors MUST have different addresses for multiple sensors ( thing that isn't possible on AM2380 and maybe other )
But it's not a ESPEASY failure !

uzi18, 2020-03-19:
@airvb @TD-er another idea is to use I2C multiplexer module: Adafruit TCA9548A

TD-er, 2020-03-19:
@uzi18 Yep that's for sure a nice feature to have.
But it does require some added code to switch that multiplexer in the right mode before starting to query that plugin.
Shouldn't be too hard though, as it is known which plugin communicates via I2C.

ghtester, 2020-06-02:
> 
> 
> nothing, just set it as DHT11, and hardware: connect SCL to GND and SDA to gpio with 10k pullup as usual with DHT sensors

It's strange in my case this does not work as expected.  All three sensors are at the same place, first one is connected to I2C bus and gives the correct data, remaining two connected as legacy in accordance the instructions above. I wonder if it's due to China's vendor or if there's another issue. The ESP firmware is recent:
Build:⋄	20106 - Mega
System Libraries:⋄	ESP82xx Core 9c56ed1f, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄	
Plugin Count:⋄	84 [Normal] [Testing] [Development]
Build Origin:	Travis
Build Time:⋄	May 16 2020 00:49:40
Binary Filename:⋄	ESP_Easy_mega_20200516_dev_ESP8266_4M1M
Build Platform:⋄	Linux-4.15.0-91-generic-x86_64-with-glibc2.27
Git HEAD:⋄	mega

[image]

airvb, 2020-06-03:
Hello , 
I didn't try with two legacy sensors, but personally it works with one aslegacy mode, the other one in I2C , however I didn't put any pullup resistance.

uzi18, 2020-06-03:
check if they work correctly in i2c mode

ghtester, 2020-06-03:
@airvb Thanks for the hint, I have already tested but it's all the same, doesn't matter if there's one or two sensors in legacy wiring.
@uzi18 i2c mode works fine on the same sensor which returns nonsense values in legacy mode. Tried to decrease pullup resistor up to 5k but it's still the same.

airvb, 2020-06-04:
[image]

Just in case : Legacy mode you have to short scl and gnd .
You can try also with 5v VCC 
[image]

ghtester, 2020-06-04:
Yeah but there's also recommendation in datasheet to use VCC 3,3V in 1wire mode if the cable length is shorter than 1m (this is my case ) otherwise there may be troubles with communication.

airvb, 2020-06-04:
[image]

tonhuisman, 2023-09-02:
This seems to have been fixed some time ago. Can be closed.

closed 2023-09-02 by TD-er