letscontrolit/ESPEasy #2944: Cannot connect to public MQTT brokers with latest releases

denisfrench, 2020-03-15:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.
`Tried several earlier versions to determine that the latest version that would connect to a public broker is ESP_Easy_mega-20181003_normal_ESP8266_4096.bin`

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
```
Most relevant issues are:
#1873; tried MQTT message interval 100, 300, 500 and 800.
#1918; tried MQTT change ClientId at reconnect setting (but not connecting initially anyway).
#1931; not sure how to swap between core versions, but I did try a custom build with custom_sdk3_ESP8266_4M1M set, to no avail.
```
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
```
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20200310
Plugin Count:⋄
46 [Normal]
Build Time:⋄
Mar 10 2020 01:23:44
Binary Filename:⋄
ESP_Easy_mega-20200310_normal_ESP8266_4M1M.bin
```
- [x] entered the full filename of the used version.
`ESP_Easy_mega-20200310_normal_ESP8266_4M1M.bin`
- [x] given a list of active plugins or controllers when applicable.
`Fresh 'normal' build, with only Home Assistant (openHAB) MQTT controller (C005) enabled`
- [x] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
`Latest release build that connects to public brokers is mega-20181003`
- [ ] Tested on other node to make sure hardware isn't defective.
`I dont have a spare ESP8266 that is not in-production, however the test Wemos D1 R2 in question connects immediately with earlier versions of ESPEasy, and also the latest version of Tasmota.`
- [x] Verified if the problem is limited to a single plugin/controller
`Only Home Assistant (openHAB) MQTT controller (C005) enabled.`

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I am unable to connect to broker.hivemq.com (no credentials required or entered), or io.adafruit.com (with credentialls) using any of the more recent release builds.

### Expected behavior

ESPEasy should connect with the configured broker immediately after WiFi is connected.

### Actual behavior

With correct credentialls (or none as applicable) later release versions are unable to connect with io.adafruit.com or broker.hivemq.com, responding with `Error : MQTT : Broker C005 connection failed (1/0)`.

Reconnection is attempted at regular intervals, without success.

### Steps to reproduce

1. Flash blank_4MB.bin, then ESP_Easy_mega-20200310_normal_ESP8266_4M1M.bin, and configure WiFi.
2. Configure controller 1 as Home Assistant (openHAB) MQTT, broker.hivemq.com:1883, no user/password, enable, and submit.
3. Connect serial terminal to view log, press reset, and observe output.

Alturnately, try with io.adafruit.com (using appropriate credentials). If there is any MQTT connection at all, it will be evident on the io.adafruit webapp if you are logged in.

Power off/on or factory reset and reconfigure has no effect.

As stated above, I have tried; MQTT message interval 100, 300, 500 and 800, MQTT change ClientId at reconnect setting (but not connecting initially anyway), and also a custom build with custom_sdk3_ESP8266_4M1M set.

Rolling back to `ESP_Easy_mega-20181003_normal_ESP8266_4096.bin` results in immediate connection with either broker (neither broker is fussy with the ClientID). No later versions that I tried (14 were tested) worked.

### System configuration

Hardware:
Wemos D1 R2 - https://wiki.wemos.cc/products:d1:d1

ESP Easy version: 
Git Build: mega-20200310

ESP Easy settings/screenshots: 
[image]

### Rules or log data

```
INIT : Booting version: mega-20200310 (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
72 : Info  : INIT : Free RAM:33360
73 : Info  : INIT : Warm boot #2 Last Task: Background Task Last systime: 949 - Restart Reason: External System
74 : Info  : FS   : Mounting...
99 : Info  : FS   : Mount successful, used 75802 bytes of 957314
499 : Info  : CRC  : program checksum       ...OK
533 : Info  : CRC  : SecuritySettings CRC   ...OK
641 : Info  : INIT : Free RAM:30680
642 : Info  : INIT : I2C
643 : Info  : INIT : SPI not enabled
723 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
825 : Info  : WIFI : Set WiFi to STA
928 : Info  : WIFI : Connecting WitsEnd attempt #0
932 : Info  : Time set to 949.000 Time adjusted by 948069.00 msec. Wander: 263.35 msec/second
934 : Info  : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
936 : Info  : Local time: 1970-01-01 00:15:49
2138 : Info  : WIFI : Connected! AP: WitsEnd (10:DA:43:1A:BC:94) Ch: 6 Duration: 1031 ms
2140 : Info  : WIFI : DHCP IP: 192.168.1.15 (ESP-Easy) GW: 192.168.1.1 SN: 255.255.255.0   duration: 13 ms
2143 : Info  : Webserver: start
2227 : Info  : firstLoopConnectionsEstablished
2240 : Error : MQTT : Intentional reconnect
2380 : Error : MQTT : Broker C005 connection failed (1/0)
2384 : Info  : WD   : Uptime 0 ConnectFailures 1 FreeMem 23632 WiFiStatus 3
17482 : Error : MQTT : Intentional reconnect
17653 : Error : MQTT : Broker C005 connection failed (2/0)
32270 : Info  : WD   : Uptime 1 ConnectFailures 2 FreeMem 23120 WiFiStatus 3
47754 : Error : MQTT : Intentional reconnect
47888 : Error : MQTT : Broker C005 connection failed (3/0)
62271 : Info  : WD   : Uptime 1 ConnectFailures 3 FreeMem 23120 WiFiStatus 3
77989 : Error : MQTT : Intentional reconnect
78157 : Error : MQTT : Broker C005 connection failed (4/0)
92270 : Info  : WD   : Uptime 2 ConnectFailures 4 FreeMem 23120 WiFiStatus 3
108258 : Error : MQTT : Intentional reconnect
108416 : Error : MQTT : Broker C005 connection failed (5/0)

```

TD-er, 2020-03-15:
Have you tried to increase the timeout settings in the controller parameters?
For online use, the default timeout of 100 msec may be too short.

ironataerial, 2020-03-15:
You had success connecting to hiveMQ with a prior ESPEasy version ?

-D

denisfrench, 2020-03-16:
Couldn't see the timeout setting for looking!

I had to set timeout up to 500ms before getting a half decent connection to the broker, and even then it is lost every few minutes. Going higher (600, 800) didn't help, and was possibly worse.

So I again reverted to ESP_Easy_mega-20181003_normal_ESP8266_4096.bin, factory reset to be sure, then reconfigured.

With the older release, connection to broker.hivemq.com is instant and rock solid. As a bonus the webif is snappy and never fails to respond, unlike the most recent release.

So something seems amiss. Are there any other settings I could be tweaking to get a more reliable MQTT connection with the current release?

TD-er, 2020-03-16:
Maybe the MQTT "client ID" has changed since the 15 month older build?
You can also try to activate the "no sleep" for WiFi?

Do you know what the timeout is on that broker?

denisfrench, 2020-03-17:
The broker is not fussy about the Client ID, so long as it is unique. I think their timeout is set to 10000ms (https://www.hivemq.com/docs/hivemq/4.3/user-guide/restrictions.html#connection-timeouts).

Force WiFi No Sleep seems to help with the webif responsiveness, but doesn't really make the MQTT connection any more stable.

I tried Periodical send Gratuitous ARP too; possibly this extended the MQTT connection time from 30 or 60s to 3 or 5 minutes. Is there a way to set the interval at runtime, or do I have to recompile to try different settings?

TD-er, 2020-03-17:
The connection timeout of 10 seconds is also used in ESPEasy (as far as I know, will check just to be sure).
The interval of Gratuitous ARP is dynamic in ESPEasy.
If a connection timeout happens the send interval for Gratuitous ARP is set to a low value (< 1 sec) and on each send it will increase this interval to a max. of 5 seconds I think.
What you can do is keep sending a ping from just any node to the ESP.
This helps to fix a number of issues:
- Force the node's WiFi to stay on (no sleep)
- Keep ARP tables active.
- It looks like it also does keep the DTIM value low (number of wifi beacon intervals to be listening)

That last one is something I cannot prove, but I base it on what I'm seeing when looking at responsiveness and power consumption.

So try to send a continuous ping to the ESP node (e.g. `ping <ip> -n 10000` ) and check if it does stay connected while sending pings.

denisfrench, 2020-03-18:
Nope; continuous pings to the ESP node makes no difference to the reliability of the connection, but it is interesting to see that the ping request times-out at the same time that the MQTT connection is lost.

To my mind the key question is; what changed between the mega-20181003 release and mega-20181004 to cause this?

TD-er, 2020-03-18:
> To my mind the key question is; what changed between the mega-20181003 release and mega-20181004 to cause this?

See the [changes between 20181003 and 20181004](https://github.com/letscontrolit/ESPEasy/compare/mega-20181003...mega-20181004)

denisfrench, 2020-03-18:
So maybe I have a WiFi  connectivity issue, not an MQTT issue. WiFi isn't actually disconnecting so far as I can see; just a few dropped ping packets coinciding with the MQTT disconnect.

Could it be from the reversion of core version? But we're back on 2.4.2 now, right?

I might try building with different core versions and see how that goes...

TD-er, 2020-03-18:
Well in the 20181003 build you're running core 2.4.x and the latest builds use core 2.6.3.
A lot has changed in roughly 18 months, so it is hard to say what is the real issue here.
The only thing we can say for sure is that the differences between those 2 builds you mentioned are mainly about using a timeout which was not used before.

rka-germany, 2020-03-18:
Hello, I have similar problems with mqtt with release mega-20200310, a new mqtt object entry with wrong name is being created.
correct entry: mqtt.0.esp8266-1
new, wrong entry: mqtt.0.esp8266-1-1
With version release mega-20200222 everything still works correctly.

My error is a little different than the one described here, but maybe my advice will help with the current troubleshooting.

TD-er, 2020-03-18:
You can enable/disable appending the unit ID.
Or do you mean the unit ID is applied twice?

rka-germany, 2020-03-18:
Hello, yes I know about the enable / disable function, the entry is indeed double in the last (and also in the penultimate) software version.

rka-germany, 2020-03-18:
So, I just tested again with the current version, after restarting the ESP a new wrong mqtt entry is created.
However, I did not correctly describe the error beforehand.
old correct entry: esp8266-4
new wrong entry: esp8266-4_4
So not a hyphen but an underscore!

[image]

TD-er, 2020-03-18:
OK, that's more clear what may be wrong here.

rka-germany, 2020-03-18:
Thank you for your great project support and good luck troubleshooting! Greetings from Germany.

denisfrench, 2020-03-19:
Progress on a new line of investigation on the original issue...

Clues (on mega-20200310, with MQTT client timeout set to 500 for public brokers):
* Initially good MQTT connection for several minutes, decreasing to a few minutes, then less and less time until disconnects every few seconds.
* The above correlates with no missed pings, then missed pings with increasing frequency.
* Initially snappy webif, with decreasing responsiveness over the same timeframe.
* Issue clearly arose with release mega-20181004.

The above sounds a little like a memory leak, but looking for a cause within the commits in question, I thought this one might be suspicious; https://github.com/letscontrolit/ESPEasy/commit/68f6e913a2784ed72482b4881274ed84c9ff2c48

So I reverted line 385 (in that commit) of _CPlugin_Helper.h in the current mega-20200310 release (which is line 35 there), and my issue is solved!

With that one reversion I am back to a rock solid MQTT connection to a public broker, and a responsive webif :-)

I propose that with a low initial timeout (10ms), incremented by 10ms on every occurance, eventually this timeout became ineffective and that loop was therefore vulnerable to delaying everything for an ever increasing amount of time.

Questions:
* Likely 100ms might be a bit long in the first instance; what would be a good compromise? (I haven't tested options yet)
* This timeout value doesn't seem to be prevented from growing until the same problem occurs eventually anyway; should there be a specified limit or reset over time?
* Your thoughts?

TD-er, 2020-03-19:
Well I will certainly look for this fix you propose here.

Just to make sure we're talking about the same here...
Line 62 in this code?

https://github.com/letscontrolit/ESPEasy/blob/4e32d98fbc2625bfc7f1c62ef9d4b610b0f4b6eb/src/_CPlugin_Helper.cpp#L60-L70

It is in the function `safeReadStringUntil` ?

I have been thinking about a more dynamic approach for the timeout, as a single set value will never be optimal.
So what I was thinking about is to keep track of the average response time of the last N successful transfers and keep a timeout of 3x that average.

But what I don't understand is that function is only used in HTTP communication, not in MQTT.
So maybe you're referring to some other part of the code?

denisfrench, 2020-03-19:
No; line 35, where the initial timeout is set to 10ms, I reverted it to 100ms.

Whatever that loop does, I figure it's sometimes holding up the works altogether for an ever increasing time as the timeout increments upwards by 10ms on each occurrence.

TD-er, 2020-03-19:
But still I don't understand why _that_ function is affecting the MQTT performance.
Are you also trying to do some HTTP communications?

denisfrench, 2020-03-20:
I was probably sitting there with a page of the webif open, if that's what you mean. MQTT disconnects would occur regardless of whether I was moving between pages or not.

Other than that, I'm working with clean installs with just a DS18B20 connected to generate some sensor data.  No controllers other than C005 running, no rules, and no incoming HTTP commands.

TD-er, 2020-03-20:
Hmm then it is quite strange to see that change you made was helping.
Could you make a clean build with the same source and try again?
I think you're running into the same issue as I've seen before, where one build is working fine and another one isn't. Seemingly random. :(

denisfrench, 2020-03-21:
So just the fact of building myself rather than using the release .bin might be fixing it then; weird!

I'll do some testing comparing release .bins with unmodified self-builds over then weekend then, and get back to you...

TD-er, 2020-03-21:
> So just the fact of building myself rather than using the release .bin might be fixing it then; weird!

Nope, even worse.
If my theory on this is correct, then it may be somewhat random whether it is working and seems to depend on the order in which things are linked.
So a rebuild (with minor code changes) without performing a clean first may act different compared to a clean build.

denisfrench, 2020-03-21:
So I compared:
1. `ESP_Easy_mega-20200310_normal_ESP8266_4M1M.bin` from the mega-20200310 release .zip.
2. An unmodified self-build from the same commit (https://github.com/letscontrolit/ESPEasy/commit/1bbfb456f7fc777f93befb41fe9ee52d8b467c20).
3. A build with line 35 of _CPlugin_Helper.h changed from +10 to +100 (`unsigned long backgroundtasks_timer = start + 100;`) **without a clean first**.
4. A build with line 35 of _CPlugin_Helper.h changed from +10 to +100 **after a clean**.

I'm using platformio/VSCode to build, and the latest ESP Easy Flasher to upload. I'm building 'normal_ESP8266_4M1M' without using custom.h or modifying platformio.ini, and flashing blank_4M.bin before every new firmware upload.

In all cases I didn't need to enable WiFi no sleep or Gratuitous ARP, but did need to increase the MQTT timeout to 500 before a connection to broker.hivemq.com succeeded.

The pattern of MQTT disconnects was, respectively:
1. 10 mins, 3 mins, 1 min, 10+ mins.
2. 16 mins, 1 min, 8 mins, 1 min, 15+ mins.
3. 25+ mins (ie no disconnect).
4. 45+ mins (ie no disconnect).

My conclusions are, with respect to mega-20200310 MQTT connections to broker.hivemq.com:
* Neither WiFi no sleep nor Gratuitous ARP are required.
* MQTT client timeout needs to be increased to 500 (by the user).
* The pattern of decreasing time to MQTT disconnect was not replicated (and maybe I am going crazy, because I was certain this was a thing).
* For whatever reason, my tinkering with _CPlugin_Helper.h results in a much more stable MQTT connection regardless of whether a clean is performed before the build.

Going back to my previous testing when I could not connect to broker.hivemq.com at all, I narrowed the change to between mega-20181003 and mega-20181004. That probably related to https://github.com/letscontrolit/ESPEasy/commit/f9e2a211e1c20ca888e6d97f3ce8fa6128e36a5b , which is fixed by increasing the MQTT timeout to 500 (in the webif).

So my original issue is solved, leaving me with a flakey MQTT connection. If you are certain the code change to _CPlugin_Helper.h can't really be the actual fix, I'm not sure where-to from here.

Perhaps we need a ridgidly standardised built environment so we are all debugging apples with apples? I was thinking a Gitpod (https://gitpod.io/) setup file might be a good alturnate option to resolve #2943 as well?

(Tasmotas is here; https://github.com/arendst/Tasmota/blob/development/.gitpod.yml ; am I allowed to say that here!?!)

TD-er, 2020-03-21:
First of all, you may certainly refer to Tasmota, as we're having same goals in mind.
I do email with Theo about issues sometimes, have been at his place for coffee and we have met a number of times on Meetups.

I don't know GitPod, but will have a look at it for sure. Looks interesting.

About your changes to `safeReadStringUntil`.
That function is not used in MQTT related code.
But that doesn't mean the change will not have an effect on MQTT stability.
Like I said, I do suspect some bug somewhere (can also be in core lib code) where the order in which objects are linked may affect WiFi stability.
Changing that  initial offset can result in different sized object file during compile, which may just fix this issue we're seeing with WiFi instability.
But for a new build, with changed source code it may be some other block is linked next to the "buggy part" and thus yield a new unstable build.

I've been trying to pinpoint this for about 2 years now and the commit you link to may fit that time frame, but I just cannot explain why it should fix these issues you're seeing.
Others have reported similar symptoms and I've also seen similar issues which are really hard to reproduce. So that's why I still believe there is some error somewhere with a string not terminated well as that's the only thing I can come up with to explain these issues which seem build dependent.

denisfrench, 2020-03-22:
Well I'll close this issue then, if there is no more progress to be made. Hopefully the random WiFi/MQTT issues will be resolved eventually.

In the meantime I'll let you get back to #2271, so that I can stop sending my MQTT credentials over the internet without encryption ;-)

closed 2020-03-22 by denisfrench