letscontrolit/ESPEasy #2945: P090 CCS811 how does it work?

henkiejan1, 2020-03-15:
Hi,

This plugin is for me not really clear to use. I can choose readings 1, 10 or 60 seconds. But a few lines lower is a ´Data acquistion´. Where i can put also in some seconds. When i add the CCS811 as new it shows 0 seconds. But after adding the plugin its default on 60 seconds. When i change it to 0 seconds it comes back with 60 seconds.

I got also NaN as result from eC02 TVOC. About 1 minute i got indeed results. When i change data acquisition to 1 second the values goes randomly to values. 1 second 0TVOC and 400 eCO2. 2 second 87 TVOC and 900 eCO2 etc...

What are the correct settings for a good measuring? :) Thank you.

TD-er, 2020-03-15:
The interval settings are standard for all plugins (using the interval setting)
The value of "0" means use the default value (60 seconds)

This interval is the number of seconds between calling the `PLUGIN_READ` call.
Typically this call is the moment when a task may output new values and thus call their connected controllers to send out new values.

The CCS811 does have a few extra options in the sensor itself for taking readings.
These are 1, 10 and 60 seconds.
I don't know the exact differences of these reading periods.
Maybe the sensor does actually sample  only at those intervals and compute an average over all internal samples until the values are read. (maybe good to search for it in the datasheet)

Now the readings of the sensor itself.
I have one of those sensors here in my breadboard and I also don't really know how to interpret the readings.
What it does is not reading the CO2 values, but instead it is reading H2 and some other VOC (ethanol?) 
Its response time is incredibly fast, but this also means it is hard to interpret the values.
For example breading on the sensor (to see if the eCO2 value makes any sense) will peak and almost immediately go back to normal again.

To be honest, I don't really know how this sensor should be used as the single values are indeed "all over the place", so it really depends on the use case how to interpret these values.
I guess you could need some post processing to have some useful values for your specific use case.
For example filtering to only follow the trend (only add a fraction of the diff between new and current value) so you don't see those short spikes.
But maybe those spikes are what interests you and then you should have a peak detection algorithm.

I would say that such a "trend filter" is the most useful for features like controlling ventilation.

To summarize:
I don't know :)
I think having the reading and the interval to somewhat similar values would make sense (e.g. reading interval of 60 seconds and interval to 60)
It does not make sense to have the interval set to a lower value than the reading period.

NorbertRoller, 2020-09-23:
@TD-er 
You do not have a binary that includes the CCS811 Plug-In in a current release availabe to share ?

I have not setup the toolchain as there wasn't any need for me so far.

Many thanks,
Norbert

TD-er, 2020-09-23:
It should be included in the "testing" build and there's also one for ESP32.
But since there's been a lot of changes in all I2C plugins, which have not yet been included in the latest nightly build, I would be glad to hear if you find anything not working with this [test build for ESP32](https://www.dropbox.com/s/k183w03846szcrc/ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip?dl=0) (and [ESP8266](https://www.dropbox.com/s/iaea3k4zxe98964/ESPEasy_ESP82xx_mega-20200913-72-g023f38a5.zip?dl=0) )

It is a build made on the 20th with the last changes on the Mega branch at that moment.

dorinvlas, 2020-12-28:
P090 is not included in the latest ESP_Easy_mega_20201227_dev_ESP8266_4M1M.bin release.

Anybody know why? I have to use ESP_Easy_mega_20201130_dev_ESP8266_4M1M.bin so I have TESTING P090.

Thank you.

TD-er, 2020-12-28:
Hmm it looks like it is commented out in the `define_plugin_sets.h` file.
No idea why.

tonhuisman, 2020-12-28:
@TD-er Oops 😊 I did, want me to fix that with a new PR, or do you re-enable it with the current work in progress?

tonhuisman, 2020-12-28:
Made PR #3442 to re-enable this plugin

Verydarksheep, 2021-01-18:
Thanks for (re-)enable the plugin!

I have looked at the manual. My understanding of those three types is, that all of them just measure once within this periodes. It seems that the sensor needs some "heating" right before the measurement as many other environment sensors. The difference seems to be the "cool down" time between each cycle and more or less important as you should wait at least 10 minutes when changing from a higher samlpe rate to a lower one for valif values (1sec -> 10sec-> 60sec)

This theorie has no valid arguments as of the following two observations:

Select "1 second" for the CSS811 and 10 sec for the "Data acquistion"
-> You get a new value every 10sec in the web ui, but will see a new value every 1 sec in the log

Select "60 seconds" for the CSS811 and 1 sec for the "Data acquistion"
-> You get a new value every second

Also there is some information about a baseline which CAN be set, as there is also an auto-baseline-update. And it seems very important to wait 20 minutes, before accurate readings are generated.

I will make some testings with the sensor via arduino IDE.

TD-er, 2021-01-18:
@Verydarksheep Please keep us posted for updates and suggestions for changes of the plugin and/or documentation.

Verydarksheep, 2021-01-28:
Still testing, but what to share my experience of the last days:

This Sensor is a MOS Sensor and made for Air Quality, not for CO2. Should be clear, but just want to mention. This affects most things you can smell, i tested parfume, nappies/diapers, sanitizer and of course breath. This "tests" where very usefull as the senor startes "recalibrating" itself if the upper Limit is reached. The baseline starts going wild up and down until the values of TVOC and eCO2 reach a 4 digit value. However the value of TVOC before that "smell-attack" is never reached again. At boot up the sensor always starts at 0/400 and considers this air quality as clean. If you move the sensor to fresh(er) air the, same roller coaster starts to the baseline. This seams to be one part of the automatic calibration.

Based on that observations i could finde several posts about exact this behaviour and also some information in the spec sheets. The sensor needs several Minutes (20min - src datasheet) to reach valid results. Also the sensor needs to be startet in fresh air or moved to fresh air within this period, as the sensor doesen't "know" what fresh air feels like.

Theoretically it is possible to get the sensor-individual baseline and set this after each powerup, however you have to wait until the conditioning-time is over (src datasheet). Also there are severals posts that mention, that the baseline value is not human readable and i had trouble to set this to a fix value.

Also i had a lot frustration with the plugin itself until i noticed, that the "Data Acquisition" must be higher (not even equal) then the operating mode. Otherwise the plugin is not working if you use the "temp/humid compensation". Also you have to reboot the ESP every time you change a setting within the plugin, as there are only "CCS811: Error: MsgInvalid" in the debug-Log. Also the plugin sw-resets the sensor on each PLUGIN_INIT (reboot of the ESP) as the .begin() includes that. That starts a new conditioning (also resetting to 0/400) and you have to wait again.

I will give it a try to modify the plugin, but as i'm not a developer it will take a bit longer ;)

That's it so far, i keep you posted!

TD-er, 2021-01-28:
If memory serves me well, some of the things you mention are already present/done in the plugin:
- Measurement is taken more often than it is reporting a value.
- Baseline is kept in RTC and restored at (warm) reboot. So as long as the unit is kept powered, the baseline should be restored.

Maybe it is a good idea to store the "fresh air" base line and store it in the settings?
If it useful to keep track of how often the unit performs some kind of re-calibration to help interpret the data?

The need for a reboot does sound like a bug, which should be dealt with IMHO.

Verydarksheep, 2021-01-28:
I double checked the  _P090_CCS811.ino  and there are a getBaseline and a setBaseline in the P090_data_struct.cpp but never used? Sry if i miss something, first time with ESPEasy code.

TD-er, 2021-01-28:
Hmm you're correct.
Looks like I messed up when I turned the plugin into supporting multiple instances.