letscontrolit/ESPEasy #2947: TaskEnable/Disable LCD Interval

piano-thomas, 2020-03-16:
I have an issue using the commands TaskEnable/TaskDisable that were recently added. 

When using these commands to enable an LCD device/task, the interval option does not work. I want to create two devices, assigned to the same LCD, but with different data on the lines.
Using the rules feature with commands TaskEnable, and TaskDisable. If I would run that rule every minute, I can enable one task, while disabling the other one at the same time.

My problem however:
If a task/device is enabled by the command TaskEnable,<taskno>, the interval function of the LCD device does not work. Even when I add the command RunTask,<taskno>, the 1 second refresh rate does not work. Does anyone know a workaround, apart from rebooting the entire NodeMCU?

TD-er, 2020-03-16:
> Does anyone know a workaround, apart from rebooting the entire NodeMCU?

You already took the first step, by adding an issue for it :)

TD-er, 2020-03-16:
Can you test this test build [ESPEasy_mega-20200310-52-PR_2948.zip](https://www.dropbox.com/s/jk17klubqj4mr31/ESPEasy_mega-20200310-52-PR_2948.zip?dl=0)

piano-thomas, 2020-03-17:
> Can you test this test build [ESPEasy_mega-20200310-52-PR_2948.zip](https://www.dropbox.com/s/jk17klubqj4mr31/ESPEasy_mega-20200310-52-PR_2948.zip?dl=0)

Works like a charm!! Thanks a lot for your very very quick fix!

PR #2948 (2020-03-17): [Task] Start scheduled runs again after (re)enableTask command
Fixes: #2947
commit b1d95fb8 (2020-03-17): [Task] Start scheduled runs again after (re)enableTask command

Fixes: #2947
closed 2020-03-17 by TD-er