letscontrolit/ESPEasy #2951: can not build test_ESP32_4M316k

mister-gefahrensucher, 2020-03-18:
I've triede to make a new build because of ISSUE #2940
Can someone point me in the right direction ?

Building in release mode
Compiling .pio\build\test_ESP32_4M316k\lib8cb\ESPeasySerial\ESPEasySC16IS752_Serial.cpp.o
Compiling .pio\build\test_ESP32_4M316k\lib8cb\ESPeasySerial\ESPEasySerial_ESP8266_noSWserial.cpp.o
.pio\libdeps\test_ESP32_4M316k\ESPeasySerial\ESPEasySC16IS752_Serial.cpp: In member function 'void ESPEasySC16IS752_Serial::b

egin(long int)':
.pio\libdeps\test_ESP32_4M316k\ESPeasySerial\ESPEasySC16IS752_Serial.cpp:66:17: error: 'class SC16IS752' has no member named 
'beginA'
       _i2cuart->beginA(speed);
                 ^
.pio\libdeps\test_ESP32_4M316k\ESPeasySerial\ESPEasySC16IS752_Serial.cpp:69:17: error: 'class SC16IS752' has no member named 
'beginB'
       _i2cuart->beginB(speed);
                 ^
Compiling .pio\build\test_ESP32_4M316k\lib8cb\ESPeasySerial\ESPEasySoftwareSerial.cpp.o
*** [.pio\build\test_ESP32_4M316k\lib8cb\ESPeasySerial\ESPEasySC16IS752_Serial.cpp.o] Error 1

TD-er, 2020-03-18:
That's something I changed this morning on the ESPEasySerial library.

But what I don't understand is that your PlatformIO is trying to download the latest version of that library as I have a version tag set in the library dependencies.

I will merge that PR, and then you should clear your libdeps folder in your .pio folder and try building again.

TD-er, 2020-03-18:
Ah found that for ESP32 I had no version of the ESPEasySerial library set so it is then using the master branch of that repository.

TD-er, 2020-03-18:
It has been closed (as I marked my latest commit to close this one), but let me know if it is still not fixed by cleaning the libdeps folder and building again using the latest HEAD of the Mega branch.

closed 2020-03-18 by TD-er

IgorTupichenko, 2020-06-14:
Hi!
I've same error when built archive source ESPEasy-mega-20190903:
Compiling .pio/build/memanalyze_ESP8266/lib726/ESPeasySerial/ESPEasySC16IS752_Serial.cpp.o
.pio/libdeps/memanalyze_ESP8266/ESPeasySerial/ESPEasySC16IS752_Serial.cpp: In member function 'void ESPEasySC16IS752_Serial::begin(long int)':
.pio/libdeps/memanalyze_ESP8266/ESPeasySerial/ESPEasySC16IS752_Serial.cpp:66:17: error: 'class SC16IS752' has no member named 'beginA'                                                                            
       _i2cuart->beginA(speed);                                                                                                                                                                                   
                 ^                                                                                                                                                                                                
.pio/libdeps/memanalyze_ESP8266/ESPeasySerial/ESPEasySC16IS752_Serial.cpp:69:17: error: 'class SC16IS752' has no member named 'beginB'                                                                            
       _i2cuart->beginB(speed);                                                                                                                                                                                   
                 ^                                                                                                                                                                                                
What I must to do?
thanks!)

TD-er, 2020-06-15:
You could set the version of the ESPEasySerial lib in platformio.ini in the libdeps.
I think the functions called from the older builds still use the same signature, so you could select the latest ESPEasySerial.
Or you download the ESPEasySerial from the same date and copy it in the libdeps directory where you now see the error reported.  See: https://github.com/TD-er/ESPEasySerial/tree/v1.0.14   (tag from before I implemented I2C to UART support.)

commit 7ee5cd2c (2020-03-18): [ESP32] Make sure to include the version tag in ESPEasySerial libdeb

Fixes: #2951