letscontrolit/ESPEasy #2953: restore configuration from a backup not working

megamarco833, 2020-03-19:
hi, with the latest build: ESP_Easy_mega-20200204_normal_ESP8266_4M1M.bin
if i save a configuration of one node and i want to restore on another one, it not working.
on webgui i see that it perform the command and suggest to reboot, but if i reboot nothing is restored.
for example IP adrress will remain the same, the device list is not updated, the controls are not updated and so on....
if i go at file browser on webgui and i uplod for example an config.dat, in this case it works and IP configuration, devices, controls will be loaded.

TD-er, 2020-03-19:
If you restore the settings, you have to reboot almost immediately and don't make any changes before rebooting.
The active settings are mostly kept in memory, so if you save something you will also save parts from memory and thus overwrite those parts from the just restored settings file.

megamarco833, 2020-03-19:
yes, but after restoring the settings i reboot, and after reboot the setting is not restored

TD-er, 2020-03-19:
Can you try to 'clone' the node using the (new) Settings Archive option? (Tools page)
See: [Settings Archive documentation](https://espeasy.readthedocs.io/en/latest/Tools/Tools.html#settings-archive)

megamarco833, 2020-03-19:
i do not see this option inside my build
i have this core:
ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
and this build:
ESP_Easy_mega-20200204_normal_ESP8266_4M1M.bin

uzi18, 2020-03-19:
> hi, with the latest build: ESP_Easy_mega-20200204_normal_ESP8266_4M1M.bin
> if i save a configuration of one node and i want to restore on another one, it not working.
> on webgui i see that it perform [...]

Can confirm, only work from file browser.

thomastech, 2020-03-20:
> on webgui i see that it perform the command and suggest to reboot, but if i reboot nothing is restored.

Before selecting the settings file for upload you **MUST** rename it to **config.dat**. Otherwise the file upload is ignored without warning.  The upload will look successful, and suggest the reboot, but it's all an illusion.

Unfortunately ESPEasy does not post an error message if the proper file naming is not used. Maybe one day ESPEasy will allow any *.dat file in the uploads, or at least post an error/warning if the name is not acceptable. In the meantime, this dirty work is up to the user. 

- Thomas

TD-er, 2020-03-20:
Maybe it is also a good idea to have the file uploaded using a different filename and let it rename at the next boot.
Right now it is somewhat dangerous to continue working on the current session after uploading a config.dat file.

thomastech, 2020-03-20:
> Maybe it is also a good idea to have the file uploaded using a different filename and let it rename at the next boot.

Nice idea for 4MB devices. But is there sufficient file space on the 1MB devices to temporarily hold the extra config file?

- Thomas

TD-er, 2020-03-20:
Nope, unless we change the config.dat file to 36k (the rest is reserved with 0)

tonhuisman, 2023-09-01:
This seems to be fixed, so can be closed.

closed 2023-09-01 by TD-er