letscontrolit/ESPEasy #2954: mqtt domoticz

megamarco833, 2020-03-19:
hi, if i connect one relay at GPIO of nodemcu and i set as controller domoticz mqtt
i create a task with relay and give the IDX of device on domoticz.
now if i clidk on domoticz to toggle the device idx espeasy not switch on the relay
it seams that there is not comunication mqtt between espeasy and domoticz 
that's possible?
if i manual publish for example:
`Publish domoticz/in,'{"command": "switchlight", "idx": 175, "switchcmd": "Toggle" }'`
it works perfectly

TD-er, 2020-03-19:
Can you try rules like these:

```
on Button1#switch do
  if [Button1#switch]=1
    gpio,12,1
  else
    gpio,12,0
  endif
endon
```

The button is the one you assign to the Domoticz controller, GPIO12 in this example is the relay here.

megamarco833, 2020-03-19:
[image]
[image]

```
on pulsante_mcp1#State do
  if [pulsante_mcp1#State]=1
     mcpgpio,9,0
  else
     mcpgpio,9,1
  endif
endon
```
in this way is it working....

but i was using monitor to save task and in this case is not working:

```
on System#Boot do
monitor,mcp,9
monitor,gpio,0
endon

on gpio#0 do
  if [gpio#0]=1
     mcpgpio,9,0
  else
     mcpgpio,9,1
  endif
endon
```

the log is:

```
319308: EVENT: Clock#Time=Thu,23:53
320689: EVENT: GPIO#0=0
320740: ACT : mcpgpio,9,1
320741: Command: mcpgpio
320745: MCP : GPIO OUTPUT 9 Set to 1
320751: EVENT: MCP#9=1
321389: EVENT: GPIO#0=1
328989: EVENT: GPIO#0=1
329041: ACT : mcpgpio,9,1
329042: Command: mcpgpio
329047: MCP : GPIO OUTPUT 9 Set to 1
329052: EVENT: MCP#9=1
332863: WD : Uptime 6 ConnectFailures 0 FreeMem 16392 WiFiStatus 3
351989: EVENT: GPIO#0=0
352040: ACT : mcpgpio,9,1
352041: Command: mcpgpio
352046: MCP : GPIO OUTPUT 9 Set to 1
352052: EVENT: MCP#9=1
352489: EVENT: GPIO#0=1
```

TD-er, 2020-03-19:
There are some changes coming for the switch plugins. See: https://github.com/letscontrolit/ESPEasy/pull/2778

In short, I am not sure this will work with Domoticz if you only have 1 switch plugin.
But I'm looking into it.

megamarco833, 2020-03-19:
but the problems, looking at log, if i'm not wrong, is related not only to domoticz, but also to output
please look at mcp output it's always: 
`mcpgpio,9,1`
never
`mcpgpio,9,0`

in realty my relays never turn on (i have inverted relay so 1=off 0=on)

megamarco833, 2020-03-19:
that's the log using input device task:
and rule like:
```
on pulsante_mcp1#State do
  if [pulsante_mcp1#State]=1
     mcpgpio,9,0
  else
     mcpgpio,9,1
  endif
endon
```
```
1042698: EVENT: GPIO#0=0
1042798: SW : GPIO=0 State=0 Output value=1
1042819: EVENT: pulsante_mcp1#State=1.00
1042865: ACT : mcpgpio,9,0
1042866: Command: mcpgpio
1042871: MCP : GPIO OUTPUT 9 Set to 0
1042902: EVENT: MCP#9=0
1043098: EVENT: GPIO#0=1
1044500: EVENT: GPIO#0=0
1044598: SW : GPIO=0 State=0 Doubleclick=3
1044620: EVENT: pulsante_mcp1#State=3.00
1044666: ACT : mcpgpio,9,1
1044667: Command: mcpgpio
1044672: MCP : GPIO OUTPUT 9 Set to 1
1044702: EVENT: MCP#9=1
1044798: EVENT: GPIO#0=1
```
you can see that here the output go with the first press of pushbutton at `mcpgpio,9,0` and with second press of button go to `mcpgpio,9,1`

now with deactivating the task for pushbutton and using monitor with this rule:
```
on System#Boot do
monitor,mcp,9
monitor,gpio,0
endon

on gpio#0 do
  if [gpio#0]=1
     mcpgpio,9,0
  else
     mcpgpio,9,1
  endif
endon
```

the log is:

```
319308: EVENT: Clock#Time=Thu,23:53
320689: EVENT: GPIO#0=0
320740: ACT : mcpgpio,9,1
320741: Command: mcpgpio
320745: MCP : GPIO OUTPUT 9 Set to 1
320751: EVENT: MCP#9=1
321389: EVENT: GPIO#0=1
328989: EVENT: GPIO#0=1
329041: ACT : mcpgpio,9,1
329042: Command: mcpgpio
329047: MCP : GPIO OUTPUT 9 Set to 1
329052: EVENT: MCP#9=1
332863: WD : Uptime 6 ConnectFailures 0 FreeMem 16392 WiFiStatus 3
351989: EVENT: GPIO#0=0
352040: ACT : mcpgpio,9,1
352041: Command: mcpgpio
352046: MCP : GPIO OUTPUT 9 Set to 1
352052: EVENT: MCP#9=1
```
here you can see that output never go at `mcpgpio,9,0` 
and phisically here i never see the relay tuon on (0=on 1=off i have relay inverted) and domoticz never show the virtual device on

megamarco833, 2020-03-20:
hi, have any idea on this, that i can test?

TD-er, 2020-03-20:
> hi, have any idea on this, that i can test?

Not sure what you mean by this.
What idea do you have to test?

megamarco833, 2020-03-20:
just put on hold for a while mqtt and domoticz.
it seams that if i use monitor for GPIO it give a double status change with only one press of push button....or better, that's what i'm understand if i look at the log posted below.
what i'm doing wrong?
the rule that i write, should triger OFF mpc pin9 when GPIO0=1 and trigger ON mcp pin9 when GPIO0=0
but if i press one time gpio0 to GND i receive that log.

wdonker, 2020-03-23:
> now if i clidk on domoticz to toggle the device idx espeasy not switch on the relay
> it seams that there is not comunication mqtt between espeasy and domoticz
> that's possible?

For communication from Domoticz to ESP you need a P029 Output - Domoticz MQTT Helper device. The MQTT-controller only performs the other way round.

TD-er, 2022-04-01:
This is no longer an issue, as using the Domoticz MQTT helper works fine (as far as I know)

closed 2022-04-01 by TD-er