letscontrolit/ESPEasy #2955: ESP8266MOD and I2C doesn't work

mobrembski, 2020-03-20:
Hi All!

I'm a fresh user of ESPEasy, and i love it on my Olimex PoE board. 
But now, i was trying to run ESPEasy on my ESP8266MOD based board:
[gerdmuller.de easyeda](https://easyeda.com/gerdmuller.de/BME280_Battery_Sensor_rev_B_10_16-toUc60iAS
)

It contains only one single BME280 breadboard connected to GPIO12(SDA) and GPIO13(SCL).
So i flashed with ESP_Easy with build on my own dev_ESP8266_4M1M (tried also other versions, also i;ve tried to move back in git history and test official build), and configured I2C with GPIOs 12 and 13. And it doesn't find any devices. Of course BME280 plugin also couldn't read any data showing:
`239490 : Info  : BMx280 : Unable to detect chip ID`

On this board i used a NodeMCU custom firmware. I've configured bme280 in a following way:
```
i2c.setup(0, 7, 6, i2c.SLOW)
bme280.setup() 
```
And it works.
It looks like bug for me now, or anybody know what's special in ESP8266MOD chip that it need to be done with I2C?

Below is system info:
```
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 2_6_3, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄

Plugin Count:⋄
81 [Normal] [Testing] [Development]
Build Time:⋄
Mar 19 2020 20:05:09
Binary Filename:⋄
Self built!
```

TD-er, 2020-03-20:
I don't know what you changed where.
Can you please give the exact lines in the code where you made changes to make it work?

mobrembski, 2020-03-20:
No sorry i think you misunderstood me.
The problem still exists in ESPeasy, it cannot find BME280 on ESP8266MOD.

I used following code in LUA in NodeMCU to configure BME280, and BME works in NodeMCU:
```
i2c.setup(0, 7, 6, i2c.SLOW)
bme280.setup() 
```
But on the same board in ESPEasy configuring i2c as GPIO12(SDA) and GPIO13(SCL) i cannot get BME280 to get it work:
```
239490 : Info : BMx280 : Unable to detect chip ID
```

TD-er, 2020-03-20:
What do the parameters of this call mean?
Do you have a link to the parameter description?

TD-er, 2020-03-20:
According to [this documentation](https://nodemcu.readthedocs.io/en/master/modules/i2c/#i2csetup) it is first SDA, then SCL.
Your given command has them reversed as GPIO12 is D6 and GPIO13 = D7.

mobrembski, 2020-03-20:
I've tried both combinations, but none of them works

mobrembski, 2020-03-20:
Hmm, i've added some debug messages inside I2C.ino:
```
uint8_t I2C_read8_reg(uint8_t i2caddr, byte reg, bool *is_ok) {
  uint8_t value;

  Wire.beginTransmission(i2caddr);
  Wire.write((uint8_t)reg);
  byte wireTrans = Wire.endTransmission(END_TRANSMISSION_FLAG) != 0;

  if (wireTrans != 0) {
    /*
       0:success
       1:data too long to fit in transmit buffer
       2:received NACK on transmit of address
       3:received NACK on transmit of data
       4:other error
       See https://www.arduino.cc/en/Reference/WireEndTransmission
     */
        String log = F("I2C : I2C_read8_reg i2caddr ");
        log += i2caddr;
        log += " wireTrans ";
        log += wireTrans;
        addLog(LOG_LEVEL_INFO, log);
    if (is_ok != NULL) {
        addLog(LOG_LEVEL_INFO, F("I2C : I2C_read8_reg is_ok is null"));
      *is_ok = false;
    }
  }
  byte count = Wire.requestFrom(i2caddr, (byte)1);

  if (is_ok != NULL) {
    addLog(LOG_LEVEL_INFO, F("I2C : I2C_read8_reg count not equal to 1"));
    *is_ok = (count == 1);
  }
  value = Wire.read();

  return value;
}
```
wireTrans (Wire.endTransmission from Arduino()) return 1, which is
```
data too long to fit in transmit buffer
```

The problem is that i cannot find related .cpp file for Wire for ESP O_o

mobrembski, 2020-03-20:
I've compared NodeMCU I2C driver, and they use I2C stretch value 100000 us.
I've set the same value in Settings->Advanced and now it's working fine :)

closed 2020-03-20 by mobrembski

TD-er, 2020-03-20:
Hmm a stretch limit of 100msec is ridiculously long.
What other I2C device is there on that board?

mobrembski, 2020-03-20:
Only BME280.
Indeed, 100ms is long, i''ve shrinked it o 100us and it's still working, hmm...Interesting

mobrembski, 2020-03-20:
By the way, i've got a question, why there's no dew point on WEB?
I can see in code it is being calculated anyway, but not presented

TD-er, 2020-03-20:
Because some controllers like Domoticz need a very specific set of values.
You can use standard conversions, like this one:

Dew point(T,H): `%c_dew_th%(18.6,67)`
See the system variables page.