letscontrolit/ESPEasy #2968: Sysinfo-Vcc reading with extreme delay at boot/start

Micha-he, 2020-03-26:
Firmware:
Own build from source 03/24/2020 but same with many other pre-build Vcc-firmware since ~ 1 year.

Issue:
After boot my ESP via RST or power-on, the value of system-info/Vcc takes about 30-40s to get a valid value.
Wifi- and MQTT-connect needs only max. 5-10s.

Older firmware from 2018 on the same ESP needs only 5-10s for all, incl. a valid Vcc value.
The difference is extreme !

TD-er, 2020-03-26:
Less than a year ago I did "fix" the reading by keeping track of the read value before starting WiFi and not reading the Vcc during WiFi setup.
The problem is that the A/D converter is also used during WiFi RF calibration, so reading during WiFi setup may give strange values and perhaps also affect WiFi stability. (not entirely sure about that though, as I don't know if reading it may affect a timed read of the same register)

What interval setting do you use for this task?

Micha-he, 2020-03-26:
You mean the Sysinfo-Vcc device/task ?

Micha-he, 2020-03-26:
Yes you're right. I've reduced the default interval from 60s to 5s and now it fast enough.

I assumed, that the initial read is immediately executed after the wifi is connected and stabilized !?

TD-er, 2020-03-26:
The plugins only output a value + event when the `PLUGIN_READ` is called.
This may already be called for the first time before most other items are initialized, so you may miss the very first read. It then takes <Interval> time to read the next.

I guess you can also read the value in the rules by referring to the `[taskname#varname]`. (or call taskrun to trigger a call to `PLUGIN_READ`)
Although I am not 100% sure it will give a valid reading immediately.
There is some logic to determine whether it is a valid value.

So the quickest way to get a valid reading is probably to trigger a rule based on the wifi connected event and then call a taskread on that task.

Micha-he, 2020-03-26:
> I guess you can also read the value in the rules by referring to the `[taskname#varname]
No, that didn't work.

Ok. Now I'm used the workaround with taskrun and if the value is >=0 after MQTT is connected.
The Vcc-value is now available again, in less than 5s.

This used to be easier...

closed 2020-03-26 by Micha-he

TD-er, 2020-03-26:
> This used to be easier...

Well the WiFi connection time also used to be 30+ seconds after boot. (if the first entered SSID was unreachable)

I can have a look at how the initial read of a plugin must be dealt with.
This was also suggested when I changed it to not read when the WiFi starts.
The ADC is the first one that really is affected by the WiFi initialization process, so maybe we can make it into a special case, or maybe we should add a new state to the plugin where a first "post boot" or "post wifi setup" call must be implemented.
For example the MQTT import is probably also one that only can be useful after wifi setup is complete.