letscontrolit/ESPEasy #2973: Feature request. make pulsecounter act on changes instead of interval(interval set to 0)

GravityRZ, 2020-03-29:
I noticed that the pulsecounter can only update itself by using the interval setting.
It would be nice if we can add functionality that when interval is set to 0 it will only update(start PLUGIN_READ) when receiving a pulse/trigger.
It is suspected that this will up the workload and resources might run out if there are a lot of pulses per second.
This functionality is particularly usefull for anybody using the pulscounter for watermeter readouts.
if the maximum water consumption is 30 liters a minute this means roughly 1 pulse every 2 seconds
Can we make an option to set the interval to zero so that the plugin only reads/updates when receiving a pulse
If that option is enabled make sure the debounce time is forced to set to at least 2000ms(or 1000ms for people using a lot of water)
if 1000ms is a workable value(without flooding the esp).

i do not think this will put a lot of extra load/resources on the esp because at the moment i am polling it every second so things might even get better when the plugin only react to activity instead of intervals.

TD-er, 2020-03-29:
Just a copy/paste of my comment on the forum about this:

Make it an option, with a "minimum time between events"

So if there is an update, it will schedule a read at last time + minimum interval and set an internal bit to make sure no new read is scheduled until that bit is cleared again (e.g. when a read is performed)

denisfrench, 2020-04-02:
@GravityRZ ; have you tried setting the pulse counter to type Delta/Total/Time?

In this mode the third value (Time) is the interval in ms between pulses, which I have found is updated regardless of the device reporting interval. I think total pulses is also updated, but I'm not using this value.

I've used the time formula field to convert the pulse interval from an electricity meter to watts. You should similarly be able to convert the pulse interval from a flow meter to LpM, if that is what you are after.

[image]

GravityRZ, 2020-04-02:
i have set it to Delta/Total/Time.

i use the time to calculate the flow/sec.

sure i can try to use the time value change as a trigger(if that one is updated regardless of the interval) but that would not be logical because that is what the delta is designed for
in your case it is logical
the Delta and the Total are only updated on an interval base 
this is why i hav sent the interval to a minimum of 1 second.
since the max amount of pulses is 1 per 2 seconds i almost always receive the pulse but there is a grey area where a pulse gets received and at that time the interval is reset.

This can be avoided by letting the delta/total/time only update when there is a pulse received

in my case the esp load will get less because now it is polling every second and when it reacts on a pulse read it will only do somethingg every 2 seconds.

i understand how you use it but i must say you specify Watts and there is no time component in Watts.
i think you are calculating kWh (watts per hour)