letscontrolit/ESPEasy #2974: bug or new feature ?

Sasch600xt, 2020-03-29:
i found some strange changes in:
ESP_Easy_mega-20200328_normal_ESP8266_4M1M.bin

first, in rules i can noch change the page to rules 2,3 or 4.
second, it feels like rules is working a lot slower and sometimes trigger are ignored if they are too close together.

in
ESP_Easy_mega-20200310_normal_ESP8266_4M1M.bin

all works fine for me.

hope that helps :)

b0kar, 2020-03-30:
Same problem here with 20200328 build.
Trying to change ruleset form results in a javascript error
`Uncaught ReferenceError: wide is not defined
    at HTMLSelectElement.onchange (rules:4)`

ironataerial, 2020-03-30:
Possible to also fix the html layout of the "Rules Set" pulldown when viewing full screen / desktop.
[image]

TD-er, 2020-03-30:
I will try to reproduce it and fix asap.
Just to be sure, what browser are you using? (although that doesn't sound like a browser issue given the error)

ironataerial, 2020-03-30:
> I will try to reproduce it and fix asap.
> Just to be sure, what browser are you using? (although that doesn't sound like a browser issue given the error)

Both FF and Chrome (latest)

TD-er, 2020-03-30:
As a work-around, it is possible to download the rules, edit them locally and upload back.
So you're not stuck until I fix it.
Also it is possible to append all rules into 1 file, as the recent patch does allow longer uploads.
But rules parsing may become a bit slower when working on large files.

TD-er, 2020-03-30:
N.B. it appears all comboboxes which should trigger a reload are affected by this, so adding/changing controllers for example also do not work as they should.

TD-er, 2020-03-30:
Found it, class and name were swapped...

waspie, 2020-03-30:
only webserver markup file affected? got the file and recompiled and now the rules page doesn't load at all for me :\

TD-er, 2020-03-30:
Hmm that's strange. It was indeed only that file and I checked it to be working here.
Please check by downloading the rules file to see it isn't damaged.

PR #2975 (2020-03-30): [Frontend] Fix reload on change for selectors (#2974 )
Fixes: #2974
commit 9a6e027f (2020-03-30): [Frontend] Fix reload on change for selectors (#2974 )

Fixes: #2974
closed 2020-03-30 by TD-er