letscontrolit/ESPEasy #2976: ESPEasy_2step_UploaderMega_1024.bin no longer build using flash mode DOUT

fmuntean, 2020-03-30:
I am testing the new build Release 20200328
It seems that the ESPEasy_2step_UploaderMega_1024.bin is not build using the flash mode DOUT
My ESP07 modules only support DOUT
Older version of this firmware (unfortunately I just override it) were using dout

Where is the source code for this firmware so I can build it myself ?

fmuntean, 2020-03-30:
Found a build from 20181211 that is having flash mode set to Dout.
So something must have happened after December 2018

TD-er, 2020-03-30:
Well the intermediate bin was not rebuilt for a while, so I guess there must have been something else.

I did change the mode to the slower mode, for other builds, but apparently it does make a difference for some units.

I will revert that change.

fmuntean, 2020-03-30:
Thank you,
Where can I find the source code for this special build ?

TD-er, 2020-03-30:
This intermediate build was not rebuilt because you need to use an old core lib or else it will be bigger.
I plan to make a new one, which supports GZIP compression, so we can have a bit more room.

fmuntean, 2020-03-30:
Ok.

closed 2020-03-30 by fmuntean

TD-er, 2020-03-30:
I was looking for the changes (to revert them) but I noticed they are all set from dio to dout.
Are you sure your module needs dout?
If so, then these changes could not be an issue.

fmuntean, 2020-03-31:
I am talking about the special bin ESPEasy_2step_UploaderMega_1024.bin in the release zip file.
Each time I flash it the board does not start then I used the esptool.py and only changed the flash format to DOUT when flashing and is working fine. 
So this is the only issue with this bin.

I could not find a way to build it myself and I think this special bin should be using dout in order to work on any board that requires it.(especially that the lower memory boards tend to have the crappy flash chips)

TD-er, 2020-03-31:
That specific file has not changed so I guess it may be a change in the default mode for the flash tool then?

fmuntean, 2020-04-07:
I also used the python tool and the same it is inside the 2 step firmware not the tool. This is how I detect the issue by forcing the esptool.py tool to write it using DOUT vs what was inside the firmware.