letscontrolit/ESPEasy #2977: Internal variables are erased on reboot

GravityRZ, 2020-03-30:
using latest build mega-20200328

i noticed that when using internal variables[VAR#1] the values are not saved when doing a reboot

let 1, 1234
[VAR#1]=1234

reboot
[VAR#1]=0

variables on dummy devices are saved but internal variables are not.

TD-er, 2020-03-30:
This is by design.
The internal variables were introduced to not be forced to use a dummy for all as we only have a limited number of task slots.
So if you need values to survive a (warm) reboot, you should use dummy tasks and not variables.

GravityRZ, 2020-03-31:
ok. i thought reading the wiki that these survive warm reboots. ok closing this one

closed 2020-03-31 by GravityRZ

uzi18, 2020-03-31:
@TD-er this should be clarified in docs

TD-er, 2020-03-31:
> @TD-er this should be clarified in docs

Yep, agree